marazmista/radeon-profile #10: Critical build error on OBS prevents radeon-profile from being packaged (and so does lack of releases)

v-fox, 2015-06-28:
Recently I tried to make [radeon-profile package for openSUSE](https://build.opensuse.org/package/show/home:X0F:HSF/radeon-profile) but couldn't because OBS automatically checks the code for potential security vulnerabilities and plain weirdness. And when it finds something, it deletes the package before letting it be downloaded. Its errors in [the log](https://build.opensuse.org/package/live_build_log/home:X0F:HSF/radeon-profile/openSUSE_13.2/x86_64) are like this:
```
[  186s] I: Program returns random data in a function
[  186s] E: radeon-profile no-return-in-nonvoid-function ../radeon-profile/dxorg.cpp:248, 362, 445, 554
[  186s] E: radeon-profile no-return-in-nonvoid-function ../radeon-profile/gpu.cpp:52, 176
```

It also doesn't like that {C,CXX}FLAGS are ignored by your build scripts:
```
[  186s] I: File is compiled without RPM_OPT_FLAGS
[  186s] W: radeon-profile no-rpm-opt-flags <cmdline>:../radeon-profile/main.cpp, ../radeon-profile/radeon_profile.cpp, ../radeon-profile/qcustomplot.cpp, ../radeon-profile/uiElements.cpp, ../radeon-profile/uiEvents.cpp, ../radeon-profile/gpu.cpp, ../radeon-profile/dxorg.cpp, ../radeon-profile/dfglrx.cpp, ../radeon-profile/settings.cpp, ../radeon-profile/daemonComm.cpp, ../radeon-profile/execTab.cpp, ../radeon-profile/execbin.cpp, qrc_radeon-resource.cpp, moc_radeon_profile.cpp, moc_qcustomplot.cpp, moc_daemonComm.cpp, moc_execbin.cpp
```

marazmista, 2015-06-29:
I've modified some code in commit b9fc7a18429265f9084d9ddd2c6ba6a730e42121, could you try build it now?

v-fox, 2015-06-29:
That did it, builds fine now, thanks !

closed 2015-06-29 by v-fox