marazmista/radeon-profile #11: Add  amdgpu support

pontostroy, 2015-07-26:
amdgpu - kernel module
amdgpu  - ddx driver
pm and fan control It is the same as with radeon

marazmista, 2015-07-28:
I created amdgpu branch although I can't test it because kconfig says that new driver can by applied to CIK cards. Can you try compile rp-amdgpu branch and see if it finds the card?

pontostroy, 2015-07-29:
does not work, here is some information about the module
```
ls /sys/kernel/debug/dri/0 -l
total 0
-r--r--r-- 1 root root          0 Jul 29 12:43 amdgpu_fence_info
-r--r--r-- 1 root root          0 Jul 29 12:43 amdgpu_gem_info
-r--r--r-- 1 root root 2147483648 Jul 29 12:43 amdgpu_gtt
-r--r--r-- 1 root root          0 Jul 29 12:43 amdgpu_gtt_mm
-r--r--r-- 1 root root          0 Jul 29 12:43 amdgpu_pm_info
-r--r--r-- 1 root root     262144 Jul 29 12:43 amdgpu_regs
-r--r--r-- 1 root root          0 Jul 29 12:43 amdgpu_ring_cp1
-r--r--r-- 1 root root          0 Jul 29 12:43 amdgpu_ring_cp2
-r--r--r-- 1 root root          0 Jul 29 12:43 amdgpu_ring_dma1
-r--r--r-- 1 root root          0 Jul 29 12:43 amdgpu_ring_dma2
-r--r--r-- 1 root root          0 Jul 29 12:43 amdgpu_ring_gfx                                                                                                                                                                                    
-r--r--r-- 1 root root          0 Jul 29 12:43 amdgpu_ring_uvd                                                                                                                                                                                    
-r--r--r-- 1 root root          0 Jul 29 12:43 amdgpu_ring_vce1                                                                                                                                                                                   
-r--r--r-- 1 root root          0 Jul 29 12:43 amdgpu_ring_vce2                                                                                                                                                                                   
-r--r--r-- 1 root root          0 Jul 29 12:43 amdgpu_sa_info                                                                                                                                                                                     
-r--r--r-- 1 root root 2147483648 Jul 29 12:43 amdgpu_vram                                                                                                                                                                                        
-r--r--r-- 1 root root          0 Jul 29 12:43 amdgpu_vram_mm                                                                                                                                                                                     
-r--r--r-- 1 root root          0 Jul 29 12:43 bufs                                                                                                                                                                                               
-r--r--r-- 1 root root          0 Jul 29 12:43 clients                                                                                                                                                                                            
drwxr-xr-x 2 root root          0 Jul 29 12:43 DP-1                                                                                                                                                                                               
drwxr-xr-x 2 root root          0 Jul 29 12:43 DVI-D-1                                                                                                                                                                                            
drwxr-xr-x 2 root root          0 Jul 29 12:43 DVI-I-1                                                                                                                                                                                            
-r--r--r-- 1 root root          0 Jul 29 12:43 gem_names                                                                                                                                                                                          
drwxr-xr-x 2 root root          0 Jul 29 12:43 HDMI-A-1                                                                                                                                                                                           
-r--r--r-- 1 root root          0 Jul 29 12:43 name                                                                                                                                                                                               
-r--r--r-- 1 root root          0 Jul 29 12:43 ttm_page_pool                                                                                                                                                                                      
-r--r--r-- 1 root root          0 Jul 29 12:43 vm                                                                                                                                                                                                 
-r--r--r-- 1 root root          0 Jul 29 12:43 vma                       
```

```
modinfo -p amdgpu
vramlimit:Restrict VRAM for testing, in megabytes (int)
gartsize:Size of PCIE/IGP gart to setup in megabytes (32, 64, etc., -1 = auto) (int)
benchmark:Run benchmark (int)
test:Run tests (int)
audio:Audio enable (-1 = auto, 0 = disable, 1 = enable) (int)
disp_priority:Display Priority (0 = auto, 1 = normal, 2 = high) (int)
hw_i2c:hw i2c engine enable (0 = disable) (int)
pcie_gen2:PCIE Gen2 mode (-1 = auto, 0 = disable, 1 = enable) (int)
msi:MSI support (1 = enable, 0 = disable, -1 = auto) (int)
[... cut ...]
-r--r--r-- 1 root root 4096 Jul 29 13:22 coresize
drwxr-xr-x 2 root root    0 Jul 29 12:51 holders
-r--r--r-- 1 root root 4096 Jul 29 13:22 initsize
-r--r--r-- 1 root root 4096 Jul 29 13:22 initstate
drwxr-xr-x 2 root root    0 Jul 29 13:21 notes
drwxr-xr-x 2 root root    0 Jul 29 13:21 parameters
-r--r--r-- 1 root root 4096 Jul 29 13:22 refcnt
drwxr-xr-x 2 root root    0 Jul 29 13:21 sections
-r--r--r-- 1 root root 4096 Jul 29 13:22 srcversion
-r--r--r-- 1 root root 4096 Jul 29 13:22 taint
--w------- 1 root root 4096 Jul 29 13:22 uevent

```

marazmista, 2015-08-09:
Thanks for all these info. Maybe now, after commit 477d3702d32bd9367959fabd02f504f4c5a9aab2 ?

sharique, 2016-05-27:
Is it still in separate branch or merged in master?
Consider make a release. Having a release make it easier to mention version while reporting issue.
Thanks for writing such a useful utility.

marazmista, 2016-05-27:
No, it is not merged because I didn't get response whether this modification worked as expected and I don't have a proper hardware to test amdgpu module myself.
As release, do you mean something like 1.0? There is a number based on date and I treat it like version number. I bump it when there is something added or modified in code and you can refer to it without ambiguity in report.
Thanks.

retuxx, 2016-05-27:
> No, it is not merged because I didn't get response whether this modification worked as expected and I don't have a proper hardware to test amdgpu module myself.

I got a R9 380 and will test your application later this day.

marazmista, 2016-05-27:
That's great, thanks :)
If you build and test this branch, you should run program as root, because there was a lot of changes in daemon communication stuff since this bug and it almost definitely won't work with latest daemon build.

retuxx, 2016-05-27:
>  it almost definitely won't work with latest daemon build

Which build should I use then?

retuxx, 2016-05-27:
I compiled and run the application (without deamon) and it crashes immediately in `gnu.cpp` line 42 (seems to not always be the answer)

```c++
dXorg::configure(gpuList[currentGpuIndex]);
```

Danysan1, 2016-05-27:
I just did a git rebase of rp-amdgpu on my Development branch, you could try that: https://github.com/Danysan1/radeon-profile/tree/rp-amdgpu

marazmista, 2016-05-28:
@retuxx The only difference seems to be the file name in debugfs and module info path. Sysfs looks the same. I updated the rp-amdgpu branch so you can build it and try, without using daemon.

pontostroy, 2016-05-29:
@marazmista  rp-amdgpu does not work
i made a strace log
https://gist.github.com/pontostroy/1e33655b6dbeb1675ab61d99fb08f59c

pontostroy, 2016-05-29:
with this it works but without power profile control
```
diff -ruN ./radeon-profile-rp-amdgpu/radeon-profile/dxorg.cpp ./radeon-profile/radeon-profile/dxorg.cpp
--- ./radeon-profile-rp-amdgpu/radeon-profile/dxorg.cpp	2016-05-28 22:39:17.000000000 +0300
+++ ./radeon-profile/radeon-profile/dxorg.cpp	2016-05-29 09:38:47.096155693 +0300
@@ -84,20 +84,20 @@
 
 void dXorg::figureOutGpuDataFilePaths(QString gpuName) {
     gpuSysIndex = gpuName.at(gpuName.length()-1);
-    QString devicePath = "/sys/class/drm/"+gpuName+"/device/";
+    QString devicePath = "/sys/class/drm/card1/device/";
 
     filePaths.powerMethodFilePath = devicePath + file_PowerMethod;
     filePaths.profilePath = devicePath + file_powerProfile;
     filePaths.dpmStateFilePath = devicePath + file_powerDpmState;
     filePaths.forcePowerLevelFilePath = devicePath + file_powerDpmForcePerformanceLevel;
     filePaths.moduleParamsPath = devicePath + "driver/module/holders/amdgpu/parameters/";
-    filePaths.clocksPath = "/sys/kernel/debug/dri/"+QString(gpuSysIndex)+"/amdgpu_pm_info"; // this path contains only index
+    filePaths.clocksPath = "/sys/kernel/debug/dri/1/amdgpu_pm_info"; // this path contains only index
     //  filePaths.clocksPath = "/tmp/radeon_pm_info"; // testing
     filePaths.overDrivePath = devicePath + file_overclockLevel;
 
 
     QString hwmonDevicePath = globalStuff::grabSystemInfo("ls "+ devicePath+ "hwmon/")[0]; // look for hwmon devices in card dir
-    hwmonDevicePath =  devicePath + "hwmon/" + ((hwmonDevicePath.isEmpty() ? "hwmon0" : hwmonDevicePath));
+    hwmonDevicePath =  devicePath + "hwmon/" + ((hwmonDevicePath.isEmpty() ? "hwmon1" : hwmonDevicePath));
 
     filePaths.sysfsHwmonTempPath = hwmonDevicePath  + "/temp1_input";
 
@@ -308,8 +308,8 @@
 
         // if above fails, use lm_sensors
         QStringList out = globalStuff::grabSystemInfo("sensors");
-        if (out.indexOf(QRegExp("radeon-pci.+")) != -1) {
-            sensorsGPUtempIndex = out.indexOf(QRegExp("radeon-pci.+"));  // in order to not search for it again in timer loop
+        if (out.indexOf(QRegExp("amdgpu-pci.+")) != -1) {
+            sensorsGPUtempIndex = out.indexOf(QRegExp("amdgpu-pci.+"));  // in order to not search for it again in timer loop
             return PCI_SENSOR;
         }
         else if (out.indexOf(QRegExp("VGA_TEMP.+")) != -1) {
@@ -323,10 +323,10 @@
 QString dXorg::findSysfsHwmonForGPU() {
     QStringList hwmonDev = globalStuff::grabSystemInfo("ls /sys/class/hwmon/");
     for (int i = 0; i < hwmonDev.count(); i++) {
-        QStringList temp = globalStuff::grabSystemInfo("ls /sys/class/hwmon/"+hwmonDev[i]+"/device/").filter("label");
+        QStringList temp = globalStuff::grabSystemInfo("ls /sys/class/hwmon/hwmon1/device/").filter("label");
 
         for (int o = 0; o < temp.count(); o++) {
-            QFile f("/sys/class/hwmon/"+hwmonDev[i]+"/device/"+temp[o]);
+            QFile f("/sys/class/hwmon/hwmon1/device/"+temp[o]);
             if (f.open(QIODevice::ReadOnly))
                 if (f.readLine(20).contains("VGA_TEMP")) {
                     f.close();
@@ -343,7 +343,7 @@
 
 QList<QTreeWidgetItem *> dXorg::getModuleInfo() {
     QList<QTreeWidgetItem *> data;
-    QStringList modInfo = globalStuff::grabSystemInfo("modinfo -p radeon");
+    QStringList modInfo = globalStuff::grabSystemInfo("modinfo -p amdgpu");
     modInfo.sort();
 
     for (int i =0; i < modInfo.count(); i++) {
@@ -373,9 +373,9 @@
     QStringList data;
     QStringList out = globalStuff::grabSystemInfo("ls /sys/class/drm/").filter("card");
     for (char i = 0; i < out.count(); i++) {
-        QFile f("/sys/class/drm/"+out[i]+"/device/uevent");
+        QFile f("/sys/class/drm/card1/device/uevent");
         if (f.open(QIODevice::ReadOnly)) {
-            if (f.readLine(50).contains("DRIVER=radeon"))
[... 66 lines cut ...]
-rw-r--r-- 1 root root      4096 May 29 09:00 pp_sclk_od
--w--w---- 1 root root      4096 May 29 09:00 remove
--w--w---- 1 root root      4096 May 29 09:00 rescan                                                                                                                                                     
-r--r--r-- 1 root root      4096 May 29 08:18 resource                                                                                                                                                   
-rw------- 1 root root 268435456 May 29 09:00 resource0                                                                                                                                                  
-rw------- 1 root root 268435456 May 29 09:00 resource0_wc                                                                                                                                               
-rw------- 1 root root   8388608 May 29 09:00 resource2                                                                                                                                                  
-rw------- 1 root root   8388608 May 29 09:00 resource2_wc                                                                                                                                               
-rw------- 1 root root       256 May 29 09:00 resource4                                                                                                                                                  
[... cut ...]
-r--r--r-- 1 root root      4096 May 29 09:00 subsystem_vendor                                                                                                                                           
-rw-r--r-- 1 root root      4096 May 29 08:18 uevent                                                                                                                                                     
-r--r--r-- 1 root root      4096 May 29 08:18 vendor             
```

also you cad add OverDrive support 
For Memory
echo [0-20] > /sys/class/drm/card[0-1]/device/pp_mclk_od
For Core
echo [0-20] > /sys/class/drm/card[0-1]/device/pp_sclk_od

https://www.youtube.com/watch?v=96yg2zWdhSs

pontostroy, 2016-05-29:
```
BOSS:/sys/class/drm/card1/device # cat power_dpm_force_performance_level                                                                              
auto                                                                                                                                                  
BOSS:/sys/class/drm/card1/device # cat power_dpm_state 
balanced                                                                                                                                              
BOSS:/sys/class/drm/card1/device # cat pp_dpm_mclk 
0: 150Mhz *
1: 1500Mhz 
BOSS:/sys/class/drm/card1/device # cat pp_dpm_pcie 
0: 8.0GB, x16 *
1: 8.0GB, x16 
2: 8.0GB, x16 
3: 8.0GB, x16 
4: 8.0GB, x16 
5: 8.0GB, x16 
BOSS:/sys/class/drm/card1/device # cat pp_dpm_sclk 
0: 300Mhz *
1: 550Mhz 
2: 725Mhz 
3: 830Mhz 
4: 914Mhz 
5: 977Mhz 
6: 1029Mhz 
7: 1050Mhz 

cat /sys/kernel/debug/dri/1/amdgpu_pm_info
uvd disabled
vce disabled
power level avg    sclk: 30000 mclk: 15000
GPU load: 0 %

```

marazmista, 2016-05-29:
Ok thanks, I will look into it.

marazmista, 2016-05-29:
Thanks for code, looks like card was not detected because different driver name in uevent. I pushed some changes, you may try it now.

pontostroy, 2016-05-30:
yes it wotks but without power profile and gpu clock monitoring

marazmista, 2016-05-30:
With or without using daemon? It should work in root mode, without using daemon, because daemon needs modifications as well.

deri82, 2016-08-04:
I dont remember anymore where I added the ppa for this program and it doesnt work with radeon 480, but I found this ticket and tried the new program. And it seems to work a lot better. So I do have the daemon installed...

When I launch the program it prints this message: Unknown power method, can't get clocks

Profiles and dpm are disabled, also overclocking. Graphs page draws only the top graph.

Setting the fan speed seems to work just fine, which is great.

marazmista, 2016-08-04:
Thanks for feedback! Can you check whether file `/tmp/amdgpu_pm_info` exists while running with daemon? Also, can you run radeon-profile in root mode? That could help find out if there is something wrong with daemon or with code that finds which driver is used. I'm yet to find where the problem is, since it looks like just a different module name but apparently it is not just this. 
Also, you can try buld from this: https://github.com/Danysan1/radeon-profile/tree/Development and see if it changes something in that matter. 
I will look into it next week because  I'm traveling now. Thanks!

deri82, 2016-08-04:
There seems to be build error.

/usr/include/qt4/QtCore/qstringlist.h:162:16: note:   initializing argument 1 of ‘QString QStringList::join(const QString&) const’
 inline QString QStringList::join(const QString &sep) const
                ^
Makefile:388: recipe for target 'execbin.o' failed
make: *** [execbin.o] Error 1

The program did launch even it failed to build, but I dont see any major changes (fan setting didnt work this time...)

I am not even sure the daemon is running, but I think the ppa set it up. And i have the ppa still enabled got few days a ago some update via it.

I dont see such directory currently...

deri82, 2016-08-04:
Actually I think it was the global version I started...anyway  radeon-profile-rp-amdgpu/
seems to build and also runs from the directory ./radeon-profile...it shows gpu celsius. Fan speed and it's changeable. Basically tested only fixed version of fan tuning. 

Also these work from command line

cat /sys/class/drm/card0/device/pp_dpm_sclk 
cat /sys/class/drm/card0/device/pp_dpm_mclk

deri82, 2016-08-09:
About fan control

https://lists.freedesktop.org/archives/dri-devel/2016-August/115281.html

deri82, 2016-08-15:
Dev build

execbin.cpp: In member function ‘void execBin::runBin(const QString&)’:
execbin.cpp:37:84: error: invalid user-defined conversion from ‘char’ to ‘const QString&’ [-fpermissive]
 text_enviroinment->setPlainText(processEnvironment().toStringList().join(' '));
                                                                             ^
In file included from /usr/include/qt4/QtCore/qobject.h:48:0,
                 from /usr/include/qt4/QtCore/qiodevice.h:46,
                 from /usr/include/qt4/QtCore/qdatastream.h:46,
                 from /usr/include/qt4/QtCore/qmetatype.h:49,
                 from /usr/include/qt4/QtCore/qvariant.h:48,
                 from /usr/include/qt4/QtCore/QVariant:1,
                 from ui_exec_bin.h:12,
                 from execbin.h:10,
                 from execbin.cpp:4:
/usr/include/qt4/QtCore/qstring.h:419:43: note: candidate is: QString::QString(const char*) <near match>
     inline QT_ASCII_CAST_WARN_CONSTRUCTOR QString(const char *ch) : d(fromAscii
                                           ^
/usr/include/qt4/QtCore/qstring.h:419:43: note:   conversion of argument 1 would be ill-formed:
execbin.cpp:37:84: error: invalid conversion from ‘char’ to ‘const char*’ [-fpermissive]
 text_enviroinment->setPlainText(processEnvironment().toStringList().join(' '));
                                                                             ^
execbin.cpp:37:84: error: invalid conversion from ‘char’ to ‘const char*’ [-fpermissive]
In file included from /usr/include/qt4/QtCore/qobject.h:48:0,
                 from /usr/include/qt4/QtCore/qiodevice.h:46,
                 from /usr/include/qt4/QtCore/qdatastream.h:46,
                 from /usr/include/qt4/QtCore/qmetatype.h:49,
                 from /usr/include/qt4/QtCore/qvariant.h:48,
                 from /usr/include/qt4/QtCore/QVariant:1,
                 from ui_exec_bin.h:12,
                 from execbin.h:10,
                 from execbin.cpp:4:
/usr/include/qt4/QtCore/qstring.h:419:43: note:   initializing argument 1 of ‘QString::QString(const char*)’
     inline QT_ASCII_CAST_WARN_CONSTRUCTOR QString(const char *ch) : d(fromAscii
                                           ^
In file included from /usr/include/qt4/QtGui/qcolor.h:47:0,
                 from /usr/include/qt4/QtGui/qpalette.h:46,
                 from /usr/include/qt4/QtGui/qwidget.h:50,
                 from /usr/include/qt4/QtGui/qaction.h:47,
                 from /usr/include/qt4/QtGui/QAction:1,
                 from ui_exec_bin.h:13,
                 from execbin.h:10,
                 from execbin.cpp:4:
/usr/include/qt4/QtCore/qstringlist.h:162:16: note:   initializing argument 1 of ‘QString QStringList::join(const QString&) const’
 inline QString QStringList::join(const QString &sep) const
                ^
Makefile:388: recipe for target 'execbin.o' failed
make: *** [execbin.o] Error 1

marazmista, 2016-08-15:
Are you trying to build it with Qt4? If you have Qt5 installed, try build using `qmake-qt5`.

> I dont see such directory currently...

Does that mean that file `/tmp/amdgpu_pm_info` not exists?

You can check if daemon is running `ps aux | grep radeon-profile-daemon`. Old daemon from ppa may not work with radeon-profile compiled from amdgpu branch. You can compile it same way as radeon-profile and then execute it from root terminal.
Have you tried run radeon-profile as root, without using daemon to see if it works that way?

deri82, 2016-08-15:
ps aux | grep radeon-profile-daemon
root      2688  0.0  0.0 151672  9972 ?        Ssl  03:44   0:00 /usr/bin/radeon-profile-daemon
user     12603  0.0  0.0  15444  1028 pts/2    S+   07:27   0:00 grep --color=auto radeon-profile-daemon

I think this is from the ppa install. 

I have tried to run as root. 

qmake && make as the quide says doesnt compile either and qmake-qt5 doesnt work at all. I installed qt5.

Interesting...

cat /tmp/amdgpu_pm_info

 [  mclk  ]: 300 MHz

 [  sclk  ]: 324 MHz

 [GPU load]: 75%

uvd    disabled
vce    disabled

marazmista, 2016-08-15:
Ok, I see that format of  `*_pm_info` is a little different, so this is the reason that app doesn't recognize available info and that will need some work.
Can you change power profile (`sys/class/drm/card0/device/power_dpm_state`) or only just fan control work?

deri82, 2016-08-15:
sudo echo battery > /sys/class/drm/card0/device/power_dpm_state
bash: /sys/class/drm/card0/device/power_dpm_state: Permission denied
sudo echo balanced > /sys/class/drm/card0/device/power_dpm_state
bash: /sys/class/drm/card0/device/power_dpm_state: Permission denied
sudo echo performance > /sys/class/drm/card0/device/power_dpm_state
bash: /sys/class/drm/card0/device/power_dpm_state: Permission denied

sudo echo low > /sys/class/drm/card0/device/power_dpm_force_performance_level
bash: /sys/class/drm/card0/device/power_dpm_force_performance_level: Permission denied

Maybe this could help you?

https://wiki.archlinux.org/index.php/ATI

Maybe I am trying wrong directories...

deri82, 2016-08-16:
Also I can send linux amdgpu driver developers a mailinglist query if that helps, but I don't know what to ask. amd-gfx-request@lists.freedesktop.org

marazmista, 2016-08-18:
Paths seem to be ok. Strange that you can't pass value in there with sudo. Do you have `echo` added to your sudoers file? Alternatively, can you login in console as root with `su -` and then try pass some values to these files?

deri82, 2016-08-18:
It makes no differece to be in su mode. I quess the problem is that I got intel cpu that integrated gpu and it moves the default ati directories elsewhere that are mentioned at the guides.

Anyone else who has the gpu could help here? I am not so skillful with linux. It feels like that I am not the right person to provide the info needed.

deri82, 2016-08-18:
Actually. I think I got it working now. You need to type sudo...then your password. No more permissions denied.

echo low > /sys/class/drm/card0/device/power_dpm_force_performance_level

low, auto, high 

I am measuring power consumptio real time and changing between those 3 modes I get different output. Even with set low the consumption is quite high. 

around 118. 24" 144hz screen, i6600k, rx480 and external amplifer (around 30-35 wattage). all combined that 118W.

just watching youtube and similar stuff. Without this ati I get as low as 40-45wattage without external amplifier.

deri82, 2016-08-18:
If these infos help you with radeon-profile I would appreciate if you would push another build of radeon-profile via ppa since I have issues getting it compiled.

marazmista, 2016-08-20:
I updated the branch. See if you can get clock read. Build would like this:
```bash
git clone https://github.com/marazmista/radeon-profile.git &&
cd radeon-profile &&
git checkout rp-amdgpu &&
cd radeon-profile &&
qmake && make

deri82, 2016-08-20:
I did like asked and I dont see any difference. It does compile.

deri82, 2016-08-20:
Failed loading translation
Unknown power method, can't get clocks

and that second line is spammed. Also the build version 20160729

marazmista, 2016-08-20:
Can you paste content of those files: `/sys/class/drm/card0/device/power_method` and `/sys/kernel/debug/dri/0/amdgpu_pm_info`?

deri82, 2016-08-20:
The 1st command gives an error no such file or directory. And the second

cat /sys/kernel/debug/dri/0/amdgpu_pm_info

 [  mclk  ]: 300 MHz

 [  sclk  ]: 300 MHz

 [GPU load]: 0%

uvd    disabled
vce    disabled

deri82, 2016-08-20:
folders under cat /sys/class/drm/card0/device/
boot_vga                           numa_node
broken_parity_status               power/
class                              power_dpm_force_performance_level
config                             power_dpm_state
consistent_dma_mask_bits           pp_cur_state
d3cold_allowed                     pp_dpm_mclk
device                             pp_dpm_pcie
dma_mask_bits                      pp_dpm_sclk
driver/                            pp_force_state
driver_override                    pp_mclk_od
drm/                               pp_num_states
enable                             pp_sclk_od
firmware_node/                     pp_table
graphics/                          remove
hwmon/                             rescan
i2c-0/                             resource
i2c-1/                             resource0
i2c-2/                             resource0_wc
i2c-3/                             resource2
i2c-4/                             resource2_wc
i2c-5/                             resource4
i2c-6/                             resource5
i2c-7/                             rom
irq                                subsystem/
local_cpulist                      subsystem_device
local_cpus                         subsystem_vendor
modalias                           uevent
msi_bus                            vendor
msi_irqs/

deri82, 2016-08-20:
cat /sys/kernel/debug/dri/0/
amdgpu_fence_info       amdgpu_ring_comp_1.0.1  amdgpu_sa_info
amdgpu_firmware_info    amdgpu_ring_comp_1.0.2  amdgpu_vram
amdgpu_gca_config       amdgpu_ring_comp_1.0.3  amdgpu_vram_mm
amdgpu_gem_info         amdgpu_ring_comp_1.0.4  clients
amdgpu_gpu_reset        amdgpu_ring_comp_1.0.5  DP-1/
amdgpu_gtt              amdgpu_ring_comp_1.0.6  DP-2/
amdgpu_gtt_mm           amdgpu_ring_comp_1.0.7  DP-3/
amdgpu_pm_info          amdgpu_ring_gfx         gem_names
amdgpu_regs             amdgpu_ring_sdma0       HDMI-A-1/
amdgpu_regs_didt        amdgpu_ring_sdma1       name
amdgpu_regs_pcie        amdgpu_ring_uvd         ttm_dma_page_pool
amdgpu_regs_smc         amdgpu_ring_vce0        ttm_page_pool
amdgpu_ring_comp_1.0.0  amdgpu_ring_vce1

marazmista, 2016-08-20:
Thanks. I pushed some changes, can you try now?

deri82, 2016-08-20:
Graphs page still does not show bottom 2 graphs.

GPu summary page shosws 0mhz and 0mhz. 
Pressing (DPM) balanced and battery shows temporarily balanced and battery, but then it forces back to performance. Same with the changing the values near the top left.

(DPM)Pressing high and low seems to work and it has huge impact at effiency. Low gives around 65wattage whole pc, but hig value consumes at desktop around 101wattage.

Going fan page and pressing custom curve seems to crash the application, this has never been happened.

Spams error at console:

Maximum number of clients reachedError loading connectors: can't connect to X server

It's definitely better now, but not completely fixed.

marazmista, 2016-08-20:
Ok, I've looked into kernel code maybe this will help,  `sudo echo manual > /sys/class/drm/card0/device/power_dpm_force_performance_level` and then run radeon-profile and try switching between battery/balanced/performance.

deri82, 2016-08-20:
Doesn't help. I think I had this problem before too. I have a script that prints performance mode and I tried to switch it to another and it never changed. It's most likely just wrong command...

deri82, 2016-08-20:
Interesting. I booted using another kernel version 4.6 (before 4.8). The curved fan profile page doesnt crash not, but what's even more interesting is that the graph page is drawn differently. It has only 2 boxes, The lower doesnt print any values/graphs, but it has a lot of content.

GPU clock
Memory clock
UVD clock
Uvd decoder 

and volts can't be ticked or unticked using 4.8 kernel you couldnt untick it and it was enabled

marazmista, 2016-08-20:
I updated the branch, it should now show clocks correctly. Strange that fan control cause a crash. Does fixed value work?
I suspect that dpm profile may work differently with amdgpu's powerplay so maybe there is a different way to set it. I'm not sure at the moment. Good news is, that amdgpu for SI cards may be merged into 4.9, so I will have a chance to implement this decently.

deri82, 2016-08-20:
If you are ARCH user you can already try this 
https://www.gamingonlinux.com/articles/arch-users-have-a-kernel-available-that-will-allow-amdgpu-and-the-pro-driver-on-older-cards.7884

Fixed value seems to work just fine for fan. The clocks are shown now which is great. 

Gpu and memory clocks are now shown/drawn at graphs page. And the page is split into 3 boxes now. The lowest box includes voltages that are not dran and the voltages cant be untick.

Do you have any clue how to show voltages? Also the overclock thing shouldnt be hard to include. I think the command was already shown here before I joined.

deri82, 2016-08-20:
I found something interesting. 

If you press high it adjust the gpu speed at its max which is for me 1288 then if I press balanced or battery it drops gpu speed 300mhz but the text top left is still performance even the gpu speed has drop. Maybe it's just wrong text there? 

There is something wrong now with my card/drivers/kernel, it doesnt lower memory speeds at all. I will reboot. I think it has done this before too.

deri82, 2016-08-20:
After reboot it seems to drop memory speed also. When pressing high and then balanced or battery. 

 I have 4.8 kernel now but I think it's not the fault of the program. There is something wrong that the card doesnt always lower memory speed.

marazmista, 2016-08-21:
> If you are ARCH user you can already try this
https://www.gamingonlinux.com/articles/arch-users-have-a-kernel-available-that-will-allow-amdgpu-and-the-pro-driver-on-older-cards.7884

This is for CIK cards and I have a SI card. This kernel is compiled with option CONFIG_DRM_AMDGPU_CIK=Y and this option is present since first merge of amdgpu so nothing new for me.

> Fixed value seems to work just fine for fan.

hmm, try delete fan profile config `~/.radeon-profile-fanSteps`.

> Do you have any clue how to show voltages?

Info about clocks and voltages coming from `amdgpu_pm_info` file. If this file does not contain volts info, that means that driver doesn't provide such info for your system. There is nothing I can do about it. The 3 boxes on Graphs page and disabled checked volts checbox is a bug in radeon-profile (should be disabled and unchecked in this case).

> If you press high it adjust the gpu speed at its max which is for me 1288 then if I press balanced or battery it drops gpu speed 300mhz but the text top left is still performance even the gpu speed has drop. Maybe it's just wrong text there? 

You can check if file `power_dpm_force_performance_level` still has `high` inside, after you pressed balanced or battery. Also, if you have multiple monitors, memory clock are always at max(but this may not be the case, since you observed drops in frequencies). Or just bug in driver.
Text on top left comes from `power_dpm_state` file. Maybe amdgpu's powerplay just does not use this feature in that way and should be disabled. I don't know.

> Also the overclock thing shouldnt be hard to include. I think the command was already shown here before I joined.

Overclock card is disabled in code. I will enable it so it can be tested.

deri82, 2016-08-21:
After deleting that file I also lost most of the new options I got recently. I redeleted and recompiled the software and started with sudo and got them back. The 1st attempt was working (fixed curve) with the old compile and deleted fan setting, but now with all options back and new compile compile it crashes again. But I am fine with fixed value. But I have noticed that when you are able to go into curved page, it does change your fan speed. I havent digged into those settings since fixed value is enough for me. The fan noice isnt irritating at all when around 40-50% fixed speed or less.

I tried your software today with dota2, changed dpm mode during gaming. And it definitely limited my gpu speed and powerdrawn. I only got around 40fps with maxed out settings, when typically its around 144 (limited to it via console command). Powerdrawn was only around 100wattages, around 220w with 144fps.

deri82, 2016-08-21:
I don't know if these add any value, but 

 cat /sys/kernel/debug/dri/0/amdgpu_
amdgpu_fence_info       amdgpu_regs_pcie        amdgpu_ring_gfx
amdgpu_firmware_info    amdgpu_regs_smc         amdgpu_ring_sdma0
amdgpu_gca_config       amdgpu_ring_comp_1.0.0  amdgpu_ring_sdma1
amdgpu_gem_info         amdgpu_ring_comp_1.0.1  amdgpu_ring_uvd
amdgpu_gpu_reset        amdgpu_ring_comp_1.0.2  amdgpu_ring_vce0
amdgpu_gtt              amdgpu_ring_comp_1.0.3  amdgpu_ring_vce1
amdgpu_gtt_mm           amdgpu_ring_comp_1.0.4  amdgpu_sa_info
amdgpu_pm_info          amdgpu_ring_comp_1.0.5  amdgpu_vram
amdgpu_regs             amdgpu_ring_comp_1.0.6  amdgpu_vram_mm
amdgpu_regs_didt        amdgpu_ring_comp_1.0.7

marazmista, 2016-08-21:
I enabled overclock tab (if overclock is available). Not sure if it works, though.
Code looks ok, volts checkbox was probably enabled when stuff didn't work and now you need to uncheck it manually in config (or delete config). `showVoltsGraphOnStart` to `false` in `~/.radeon-profile-settings`
If you run application as `sudo radeon-profile`, it run as root, so config is applied from root's home. Otherwise, config comes from user's home directory. Same thing for fan profiles or exec definitions.
Yes, the fan speed is calculated and adjusted accordingly when user switch mode to Custom curve.
I know about those files, it is probably more detailed debug stuff for devs.

deri82, 2016-08-21:
Overclocking tab is enabled. I don't know if it works...

Anyway lots of changes recently. It's a pity that amd doesnt provide something like this themselves for linux.

ChristophHaag, 2016-08-21:
I looked a bit at the code and added some stuff for adding one value from powerplay to the graphs for polaris etc. support. I think it's actually the core clock that is added to the memory graph, but all in all it doesn't look too complicated:
https://github.com/ChristophHaag/radeon-profile/commit/6c6fa68735ba2edc816cb40f4bf97dbdd9f19f96

No idea how the daemon stuff works though. Anyone want to do the work?

deri82, 2016-08-26:
I installed some file that I shuoldnt have and it started to conflict mesa drivers. I solved it with aptitude but it reverted into old version of opengl. I wanted to test gpu mining. It seems to work only with closed source drivers. So I end up installing them. I had to force install some package. Radeon-profile seems to work with closed source drivers but, you cant impact for example dpm mode...

nomnombtc, 2016-09-07:
@marazmista 

I installed radeon-profile from the rp-amdgpu branch.
So far it works fine mostly, but there is a big problem for me:

It does not save the fansteps (the file is empty) and just hangs when I try
to close the application. It somehow seems to get stuck calling saveFanProfiles()
in uiEvents.cpp. If I comment that function out the application quits correctly.

The file gets created, but it is empty, even worse if I manually input values, it is empty
again next time:
```
-rw-r--r-- 1 1000 100    0 Sep  7 02:34 .radeon-profile-execProfiles
-rw-r--r-- 1 1000 100    0 Sep  7 02:29 .radeon-profile-fanSteps
-rw-r--r-- 1 1000 100 1364 Sep  7 02:34 .radeon-profile-settings
```

marazmista, 2016-09-10:
@ChristophHaag 
Thanks. Ok, so I see that you can read clocks from sysfs rather than debugfs. Do you know how to change clock speed or dpm profile with power play? Because that would be the only things that daemon comes in. Apparently passing something into `power_dpm_force_performance_level` works but setting different profile into `power_dpm_state` does nothing. Does it work in some other way for powerplay?  If the file looks, for example, like that:
```
0: 300Mhz *
1: 550Mhz 
2: 725Mhz 
3: 830Mhz 
4: 914Mhz 
5: 977Mhz 
6: 1029Mhz 
7: 1050Mhz 
```
I understand, that the star indicate current frequency. Would command `echo 5 > /sys/class/drm/card0/device/pp_dpm_sclk` change that frequency to 977Mhz?

@deri82 
radeon-profile with closed source driver is basically just read only. There is no dpm to control and as far as I can remember, closed source driver has its own application to control that stuff, doesn't it? Catalyst Control Center or something else with new Crimson driver.

@nomnombtc 
hmm.. that is strange, it is just few lines of code. I modified this method in latest commit (5bbc9bd5e6b86ee498d069f42997ae565ae31577), see if this change helps for this issue.

deri82, 2016-09-10:
The original issue was the heat and power consumption that rx480 produces. I have come with solution that works for me. I opened the reference cooler, but let the backlate be there. Put an old amd cpu cooler fan on top of it with few screws. I have no windows lisence anymore but I downloaded w10 iso and installed it on ssd, it allows to use it without activation for 30 days or something. 

Then I used polariseditor and atiwinflash and reduced the clocks and voltages. And I have now less heat producing and more effient gpu. But this gpu is weird, it consumes around 20-30 wattages at idle states. But in Dota 2 game I get around only 130 wattages more usage with maximum settings I can use. 1920x1080, fps capped to 144 limiter.

How is this relevant? I tried to use this program to lower my consumption with software but I have made hardware mods meanwhile.

nomnombtc, 2016-09-11:
@marazmista 
still the same with https://github.com/marazmista/radeon-profile/commit/5bbc9bd5e6b86ee498d069f42997ae565ae31577

But I think I found out why it hangs. I noticed some compiler warning scrolling by:
```
../radeon-profile/settings.cpp: In member function ‘void radeon_profile::saveFanProfiles()’:
../radeon-profile/settings.cpp:233:32: warning: range-based ‘for’ loops only available with -std=c++11 or -std=gnu++11
         for (int temperature : fanSteps.keys())
                                ^
[...]
../radeon-profile/settings.cpp: In member function ‘void radeon_profile::saveFanProfiles()’:
../radeon-profile/settings.cpp:234:43: warning: ‘temperature’ may be used uninitialized in this function 
[-Wmaybe-uninitialized]
             profile.append(QString::number(temperature)+ "#" + QString::number(fanSteps.value(temperature))  + "|");
```

That seems to be the part where it gets stuck. So I was like, maybe I should try to enable c++11?
 
Found [this](http://stackoverflow.com/questions/19398438/c-qt-how-to-add-std-c11-to-the-makefile-which-is-generated-by-qmake). And it seems the correct way is adding `CONFIG += c++11` but this only works with qt5. With qt4 you need to set QMAKE_CXXFLAGS.

Adding something like this to radeon-profile.pro makes it work fine now:
```
lessThan(QT_MAJOR_VERSION, 5) {
QMAKE_CXXFLAGS += -std=c++11
} else {
CONFIG += c++11
}
```

Not sure this is the correct way of doing it, I have qt4 so I can only confirm that adding
`QMAKE_CXXFLAGS += -std=c++11` works (saves fansteps correctly and application closes without hanging)

[9 more replies]

closed 2016-12-21 by marazmista