marazmista/radeon-profile #12: radeon-profile can't work with APU

orumin, 2015-08-06:
I use radeon-profile with A4-5000 (HD8330), within radeon-profile-daemon and debugfs mounted.
but, I get message "Can't read data."
Do I solve this problem?

marazmista, 2015-08-06:
Does it work correctly when you run radeon-profile as root without daemon running?

orumin, 2015-08-08:
No, It doesn't

marazmista, 2015-08-14:
If you have daemon running, close it. Open terminal, run daemon as root. Open second terminal and run radeon-profile as normal user. Could you post output here? Maybe it will show us something interesting. Also, do you have files in `/sys/kernel/debug/dri/0/` (or maybe in other index than 0)?

orumin, 2015-08-16:
here is the output
```
$ sudo radeon-profile-daemon
ok
connection
QFSFileEngine::open: No file name specified
0#/sys/kernel/debug/dri/0/radeon_pm_info#1
1
12balanced#/sys/class/drm/card0/device/power_dpm_state#2auto#/sys/class/drm/card0/device/power_dpm_force_performance_level#
1
1
1
1
1
1
1
1
1
1
1
1
```
And I have ```/sys/kernel/debug/dri/0/``` as below
```
$ sudo ls /sys/kernel/debug/dri/0/
bufs       HDMI-A-1  radeon_fence_info  radeon_gtt       radeon_pm_info   radeon_ring_dma1  radeon_ring_uvd   radeon_sa_info  ttm_dma_page_pool  vm
clients    LVDS-1    radeon_gem_info    radeon_gtt_mm    radeon_ring_cp1  radeon_ring_dma2  radeon_ring_vce1  radeon_vram     ttm_page_pool      vma
gem_names  name      radeon_gpu_reset   radeon_mst_info  radeon_ring_cp2  radeon_ring_gfx   radeon_ring_vce2  radeon_vram_mm  VGA-1

```

marazmista, 2015-08-20:
Hmm... have you tried checking "Refresh data without request" and then reconfigure daemon in configuration on daemon tab?

orumin, 2015-08-21:
Unfortunately, nothing happend...

marazmista, 2015-08-28:
Are you able to change profiles, like set gpu to performance, or there is no clocks and all buttons are disabled?

orumin, 2015-08-30:
I can change profile. but no line appears in graph

marazmista, 2015-08-31:
Ok, can you post your `/sys/kernel/debug/dri/0/radeon_pm_info`?

orumin, 2015-09-01:
Yeah, 
```
uvd    disabled
vce    disabled
power level 0    sclk: 26667 vddc: 3800
````

marazmista, 2015-09-02:
I created a branch for this fix, you can compile and test it by executing this command:
`git clone https://github.com/marazmista/radeon-profile && cd radeon-profile && git checkout rp-bugfix && cd radeon-profile && qmake && make `

You can replace qmake with qmake-qt4 or qmake-qt5, whatever suits you.
Binary will be in source directory. If fix will work, I will merge it to master.

orumin, 2015-09-03:
Thanks for your quick response, I can get a gpu clock and some data!

marazmista, 2015-09-05:
Merged to master, bug closed.

closed 2015-09-05 by marazmista