marazmista/radeon-profile #15: Charts are in offset of 20 seconds

Danysan1, 2016-01-19:
The 3 real-time graphs are in offset of 20 seconds:

[image]

I found the reason, in https://github.com/marazmista/radeon-profile/blob/master/radeon-profile/radeon_profile.cpp#L358 and following lines:
``` c++
ui->plotTemp->xAxis->setRange(ticksCounter+20, rangeX,Qt::AlignRight);
ui->plotColcks->xAxis->setRange(ticksCounter+20,rangeX,Qt::AlignRight);
ui->plotVolts->xAxis->setRange(ticksCounter+20,rangeX,Qt::AlignRight);
```
Removing those `+20` from the lines the graph is ok:

[image]

Do those `+20` have any reason to stay there?
Otherwise we can delete them.

marazmista, 2016-01-20:
This is 100% intentional, I wanted this gap on the right. I know that it looks like first screen on max zoom, but one can adjust zoom freely.

Danysan1, 2016-01-21:
Can we add to the options a toggle for this gap?

marazmista, 2016-01-24:
Option added.
commit: 8f8e38bed45db3285382a9ee2402639030109d95

closed 2016-01-24 by marazmista