marazmista/radeon-profile #17: Fan control not working for Radeon HD 6970 (using r600 driver)

bernandosoares, 2016-04-07:
The fan control tab is completely grayed out, even when radeon-profile is run as root. Is this because of lack of support in the driver, or in radeon-profile?

Danysan1, 2016-04-08:
Are all optional dependencies listed in https://github.com/marazmista/radeon-profile/blob/master/README.md installed?

bernandosoares, 2016-04-08:
Yep

bernandosoares, 2016-04-08:
I am running an up to date Manjaro system, if that helps

marazmista, 2016-04-08:
According to news https://www.phoronix.com/scan.php?page=news_item&px=MTg0NDM (I haven't found anything newer) only cards HD7000 and above have fan support in the driver. To be sure, you can check if files, like `/sys/class/drm/card0/device/hwmon/hwmon2/pwm1` (card index may be different) and such exists in your system.

closed 2016-11-19 by marazmista