marazmista/radeon-profile #21: PPA repository

muzena, 2016-06-02:
@marazmista 
I made PPA repository for xenial and yakkety, but for trusty I get some build error, Is this cause old version of compiler?
Build log:
https://launchpad.net/~trebelnik-stefina/+archive/ubuntu/radeon-profile/+build/9851099

marazmista, 2016-06-02:
Seems to be problem with the loop in settings.cpp
`
../settings.cpp: In member function ‘void radeon_profile::makeFanProfileGraph()’:
../settings.cpp:218:28: error: range-based ‘for’ loops are not allowed in C++98 mode
     for (int temperature : fanSteps.keys())
                            ^
../settings.cpp: In member function ‘void radeon_profile::saveFanProfiles()’:
../settings.cpp:228:32: error: range-based ‘for’ loops are not allowed in C++98 mode
         for (int temperature : fanSteps.keys())
`
If you can, try build with c++11 support or let me know and I will rewrote this part to be compatible with c++98 mode.

Danysan1, 2016-06-02:
Trysty Tahr packages gcc 4.8.2, which DOES support c++11. Try adding this to the .pro file:

CONFIG += c++11

muzena, 2016-06-02:
@marazmista 
I try to add as @Danysan1 write CONFIG += c++11 in radeon-profile.pro but it dosen't work.

Danysan1, 2016-06-02:
[Try:](http://stackoverflow.com/questions/19398438/c-qt-how-to-add-std-c11-to-the-makefile-which-is-generated-by-qmake)

QMAKE_CXXFLAGS += -std=c++11

muzena, 2016-06-02:
@Danysan1 
Thanks, this QMAKE_CXXFLAGS += -std=c++11 work.

muzena, 2016-06-02:
@marazmista 
Is it possible to fix this problrm in Unity
https://www.phoronix.com/forums/forum/linux-graphics-x-org-drivers/open-source-amd-linux/37384-radeon-profile-tool-for-changing-profiles-and-monitoring-some-gpu-parameters?p=875954#post875954
And in app indicator icon, missing radeon-profile image.

marazmista, 2016-06-03:
@muzena I've replied in that thread about icon issue.

muzena, 2016-06-09:
@marazmista @Danysan1 
system tray was removed from ubuntu and it doesn't work anymore, i tried a few modified sys trays but neither of them don't work.
Can someone write script for upstart, for ubuntu trusty, I tried but without success.

jarrard, 2017-06-28:
A PPA repo would be great as when I build the rp-dev branch version the radeon-profile app just segfaults with some Qt error (using Qt5.9).

marazmista, 2017-06-29:
Install qt5 charts package. libqt5charts5 or libqt5charts5-dev. I don't what is the actual name for that in ubuntu.

jarrard, 2017-06-29:
Yeah that is installed with 5.9, no charts compile error.

muzena, 2017-07-05:
@jarrard 
What version of Ubuntu you are using, or how you install qt5 5.9 on Ubuntu?
I can't update PPA repository because libqt5charts5 or libqt5charts5-dev packages don't exist in Ubuntu Xenial.

jarrard, 2017-07-05:
https://www.qt.io/download-open-source/

You then install the run file as per instructions, it should auto select charts for you. Can take a while to install so be patient. Once installed you will need to redefine you Qt path to where you install Qt too (somewhere in HOME or whatever). 

I can remember the exact command to redefine the library path but pretty sure google can help.

muzena, 2017-07-23:
@marazmista @Danysan1 
Why overclocking is not available on Radeon 7970 Tahiti GPU?
And in graphs missing UVD option like in older version of Radeon profile.

marazmista, 2017-07-24:
@muzena 
Overclocking doesn't work for GCN 1.0 and 1.1(?) cards, even if you use `amdgpu` module.

For UVD clocks, if you use `radeon` module, you can check option 'Use daemon as data provider', restart app and add values to plots while clocks are >0. For `amdgpu` I'm not sure if this info is exposed.

muzena, 2017-07-24:
@marazmista 
Is lm-sensors needed for dependency?

marazmista, 2017-07-24:
No, it's optional (as fallback in rare cases)

mainmachine, 2017-08-28:
Some folks might want to install QT 5.9.1 from a PPA:

https://launchpad.net/~beineri/+archive/ubuntu/opt-qt591-xenial

I had to do this on 16.04.3 to get the latest version of radeon-profile to work. :)

muzena, 2017-08-28:
@mainmachine 
I use your repository for building radeon-profile, I made copies in case that you decide delete or disable repositories https://launchpad.net/~trebelnik-stefina/+archive/ubuntu/radeon-profile-dependencies.
I add options in preinstall script to add and update repository before installing radeon-profile.

mainmachine, 2017-08-28:
@muzena 
Cool, but it's not my repo! :)

Great project BTW - it's much needed to allow folks to have good GUI support on Linux on par with the Nvidia proprietary GUI. :+1: