marazmista/radeon-profile #22: Fan control settings are not being saved; version: 20160515

Karut, 2016-07-28:
I can flawlessly adjust all settings, e.g. the clock settings are being saved and reloaded when I relaunch the application. Fan control settings are editable however they are not being saved, they are being reset to _auto_ when I restart the application. Is this a security feature or a bug? 

Graphics card: HD 7860
OS: Opensuse Tumbleweed
Radeon-profile version: 20160515

Thanks!

marazmista, 2016-07-28:
Yes, that is intentional. For instance, when user set fixed fan speed to 30% and then close the program, it will stay that way. So when something start use gpu, temperature will go up, but fan speed stay locked at 30% and this may cause overheat.
When you close radeon-profile fan control is set to auto, which is card internal mode, so it shouldn't cause problems. Overriding this behaviour, when there is nothing else that would adjust fan speed when needed is IMO bad idea, although I can add option to restore fan setting when user reopen program, no problem.
Thanks for report.

Karut, 2016-07-28:
> Yes, that is intentional. For instance, when user set fixed fan speed to 30% and then close the program, it will stay that way. So when something start use gpu, temperature will go up, but fan speed stay locked at 30% and this may cause overheat.
When you close radeon-profile fan control is set to auto, which is card internal mode, so it shouldn't cause problems. Overriding this behaviour, when there is nothing else that would adjust fan speed when needed is IMO bad idea,

Leaving the fan control in manual mode after shutting down the program certainly is not a good idea, agreed. -- Since there is no automatic start of the program at boot (at default settings) loading the last settings should be fine. The users who set it to start automatically know what they are doing and additionally should (and hopefully do) have the fan control set in a stepwise-function (genius!) so a overheating cannot occur.

Thank you for implementing that option! Keep up the good work!

marazmista, 2016-07-29:
Option added in 7265ae67e3b49a9a393572b17e7a448d7520e347 (20160729)

closed 2016-07-29 by marazmista

Karut, 2016-10-22:
I now discovered that settings are only acted on after a shutdown/reboot - after a hibernate or suspend to ram the settings are being saved but not 'active'. Its still set on fixed and the configured fan speed, but only applied when I click 'fixed'. Any idea why this might? Its as if the hibernate/suspend kills the fan control signal and radeon-profile does not resend the signal after waking up from hibernation. You may have to reopen the bug or should a file a new bug? Thanks!

marazmista, 2016-10-23:
This doesn't seem to be a bug in radeon-profile. If you set fan values manually by echo values into sysfs files, they aren't kept after resume. This may be bug in driver or intentional behaviour.
Although, you can write script to save those values to temporary file before suspend and restore them at wake like it is described here: https://askubuntu.com/a/261076