marazmista/radeon-profile #23: Starting minimized, but not as minimized to tray

onli, 2016-09-30:
It would be nice if the program could start not only minimized, but minimized to tray if the option to "minimize to tray" and "start as minimized" are active at the same time.

marazmista, 2016-10-01:
It should behave the way you've described already. What distro do you use? Also, what desktop environment and window manager?

onli, 2016-10-01:
Just to be sure, I'm only talking about application start. When afterwards raising and then minimizing it it goes as expected into the systray.

I'm using Funtoo with IceWM, no desktop environment. Stalonetray and simdock, but if there is an incompatibility it's probably with IceWM.

marazmista, 2016-10-03:
I see. Or maybe there is something wrong with launcher or task manager(simdock)? I use openbox with tint2 and, with this options selected, radeon-profile goes right into the tray. Tint2 doesn't show it. Have you tried any other program with similar capabilities to check such behaviour?

onli, 2016-10-04:
I just tested it out. Tint2 with IceWM: Minimized on start, but not only in tray. Tint2 with Openbox: Exactly the same. Raising and then minimizing fixes it, the program goes into the tray only.

Just in case it might somehow help, I recorded the behaviour: https://www.onli-blogging.de/uploads/radeon_profile_nonminimize-2.ogv. Maybe I'm missing something?

One thing to note is that I'm just realizing that I'm not on the current version anymore. Just saw you pushed some updates, but with your last commits the compilation fails for me. But going through them I don't think they should matter here, right?

marazmista, 2016-10-04:
Thanks for the video. Yes, the commits in new version weren't about this issue. Why does compilation of new version fail in your system?
Anyway, I pushed new branch `rp-bugfix`, with changes in showing window code. You can compile by running:
```bash
git clone https://github.com/marazmista/radeon-profile.git &&
cd radeon-profile &&
git checkout rp-bugfix &&
cd radeon-profile &&
qmake && make
```

onli, 2016-10-04:
Think that is the same error as in the main: https://gist.github.com/onli/acd84a1c4c4954f218e411ff1c7eef0b

I assume this is qt4 biting? I could try to switch to qt5, but I'd prefer helping you fix this compile error if I can.

marazmista, 2016-10-05:
I pushed some fixes, it should build now.

onli, 2016-10-05:
Great, Thanks. Compiles now and also fixed the bug. It properly goes directly into the tray.

marazmista, 2016-10-08:
Great to hear, thanks.

closed 2016-10-08 by marazmista

pschichtel, 2017-09-04:
I have this problem on Arch with Gnome 3.24.2

Minimize to tray and start minimized are enabled, but the windows keeps appearing after login.

Enigma0, 2017-09-05:
Ditto

Vladar4, 2018-12-19:
Same problem with Arch + openbox 3.6.1 + tint2 16.6.1.

lhorace, 2018-12-20:
@Vladar4 Might want to open a new issue as this is fairly old issue.

Vladar4, 2018-12-21:
Oh, thanks, didn't notice that this one is closed.