marazmista/radeon-profile #24: Set speed below 20

github23leech, 2016-10-20:
Setting up a custom curve won't accept any value below 20 % for the fan speed with my R9 270X, but the card works fine with only 10 %, without overheating.
It's possible to manually set the lower value by:

echo 1 > /sys/class/drm/card0/device/hwmon/hwmon4/pwm1_enable
echo 33 > /sys/class/drm/card0/device/hwmon/hwmon4/pwm1

20 % is kinda noisy while idling.

Thanks in advance, you wrote a great tool! I really appreciate it.

marazmista, 2016-10-22:
Thanks. I modified it in latest commit, f6ecff412eaa929ea39cda11b9512191715b8018, check if it works ok. I unlocked edit on the first item in fan profile, so you can change speed from old default 20% to 10%

github23leech, 2016-10-22:
Works like a charm, thanks again!

closed 2016-10-23 by marazmista