marazmista/radeon-profile #28: Sigsegv when pressing a fan control button, card HD6870 on r600

gerddie, 2016-11-22:
Hello, 

thanks for the nice program.  
However, when running it (as root) and clicking on one of the fan control buttons ("fixed" or "Custun curve") the program segfaults. 

I ran it directly from the compilation directory and my graphics hardware is a Radeon HD6870 using the r600 driver with Gentoo kernel 4.8.8.  

Any hints? 
Many thanks, 
Gert 

The log and gdb backtrace follow: 

> Failed loading translation 
> QMetaObject::connectSlotsByName: No matching signal for on_combo_fanProfiles_currentTextChanged(QString)
> [New Thread 0x7fffef484700 (LWP 14648)]
> QFSFileEngine::open: No file name specified
> Unable to open  ""  to write  "1" 
>
> Program received signal SIGSEGV, Segmentation fault.
> QList<QAction*>::QList (l=..., this=0x7fffffffd300) at ../../../qt-everywhere-opensource-src-4.8.6/src/corelib/tools/qlist.h:122
> 122	    inline QList(const QList<T> &l) : d(l.d) { d->ref.ref(); if (!d->sharable) detach_helper(); }
> (gdb) bt 
> 0  QList<QAction*>::QList (l=..., this=0x7fffffffd300) at ../../../qt-everywhere-opensource-src-4.8.6/src/corelib/tools/qlist.h:122
> 1  QWidget::actions (this=0x7fffffffe558) at /usr/src/debug/dev-qt/qtgui-4.8.6-r4/qt-everywhere-opensource-src-4.8.6/src/gui/kernel/qwidget.cpp:3344
> 2  0x000000000048699b in radeon_profile::findCurrentFanProfileMenuIndex() ()
> 3  0x0000000000486ad9 in radeon_profile::setCurrentFanProfile(QString const&, QMap<int, unsigned int> const&) ()
> 4  0x0000000000486ca9 in radeon_profile::on_btn_pwmProfile_clicked() ()
> 5  0x00000000004afe13 in radeon_profile::qt_metacall(QMetaObject::Call, int, void**) ()
> 6  0x00007ffff651091e in QMetaObject::activate (sender=sender@entry=0x7dbba0, m=m@entry=0x7ffff787cf00 <QAbstractButton::staticMetaObject>, local_signal_index=local_signal_index@entry=2,    argv=argv@entry=0x7fffffffd500) at /usr/src/debug/dev-qt/qtcore-4.8.6-r2/qt-everywhere-opensource-src-4.8.6/src/corelib/kernel/qobject.cpp:3597
> 7  0x00007ffff740f39a in QAbstractButton::clicked (this=this@entry=0x7dbba0, _t1=true) at .moc/release-shared/moc_qabstractbutton.cpp:219
> 8  0x00007ffff7146643 in QAbstractButtonPrivate::emitClicked (this=this@entry=0x8b5710)   at /usr/src/debug/dev-qt/qtgui-4.8.6-r4/qt-everywhere-opensource-src-4.8.6/src/gui/widgets/qabstractbutton.cpp:548
> 9  0x00007ffff71479b4 in QAbstractButtonPrivate::click (this=this@entry=0x8b5710)   at /usr/src/debug/dev-qt/qtgui-4.8.6-r4/qt-everywhere-opensource-src-4.8.6/src/gui/widgets/qabstractbutton.cpp:541
> 10 0x00007ffff7147ac4 in QAbstractButton::mouseReleaseEvent (this=0x7dbba0, e=0x7fffffffda40)  at /usr/src/debug/dev-qt/qtgui-4.8.6-r4/qt-everywhere-opensource-src-4.8.6/src/gui/widgets/qabstractbutton.cpp:1123
>11 0x00007ffff6dce500 in QWidget::event (this=0x7dbba0, event=0x7fffffffda40) at /usr/src/debug/dev-qt/qtgui-4.8.6-r4/qt-everywhere-opensource-src-4.8.6/src/gui/kernel/qwidget.cpp:8775
>12 0x00007ffff6d7b0cc in QApplicationPrivate::notify_helper (this=this@entry=0x734df0, receiver=receiver@entry=0x7dbba0, e=e@entry=0x7fffffffda40)   at /usr/src/debug/dev-qt/qtgui-4.8.6-r4/qt-everywhere-opensource-src-4.8.6/src/gui/kernel/qapplication.cpp:4565
>13 0x00007ffff6d821c6 in QApplication::notify (this=<optimized out>, receiver=0x7dbba0, e=0x7fffffffda40)     at /usr/src/debug/dev-qt/qtgui-4.8.6-r4/qt-everywhere-opensource-src-4.8.6/src/gui/kernel/qapplication.cpp:4108
>14 0x00007ffff64fc7ec in QCoreApplication::notifyInternal (this=0x7fffffffe290, receiver=receiver@entry=0x7dbba0, event=event@entry=0x7fffffffda40)     at /usr/src/debug/dev-qt/qtcore-4.8.6-r2/qt-everywhere-opensource-src-4.8.6/src/corelib/kernel/qcoreapplication.cpp:953
>15 0x00007ffff6d817cd in QCoreApplication::sendEvent (event=<optimized out>, receiver=<optimized out>) at ../../../qt-everywhere-opensource-src-4.8.6/src/corelib/kernel/qcoreapplication.h:231 
>16 QApplicationPrivate::sendMouseEvent (receiver=receiver@entry=0x7dbba0, event=event@entry=0x7fffffffda40, alienWidget=alienWidget@entry=0x7dbba0, 
    nativeWidget=nativeWidget@entry=0x7fffffffe2a0, buttonDown=buttonDown@entry=0x7ffff788e318 <qt_button_down>, lastMouseReceiver=..., spontaneous=true)     at /usr/src/debug/dev-qt/qtgui-4.8.6-r4/qt-everywhere-opensource-src-4.8.6/src/gui/kernel/qapplication.cpp:3173
>17 0x00007ffff6df8d8f in QETWidget::translateMouseEvent (this=this@entry=0x7fffffffe2a0, event=event@entry=0x7fffffffdda0)     at /usr/src/debug/dev-qt/qtgui-4.8.6-r4/qt-everywhere-opensource-src-4.8.6/src/gui/kernel/qapplication_x11.cpp:4519
>18 0x00007ffff6df773d in QApplication::x11ProcessEvent (this=0x7fffffffe290, event=event@entry=0x7fffffffdda0)     at /usr/src/debug/dev-qt/qtgui-4.8.6-r4/qt-everywhere-opensource-src-4.8.6/src/gui/kernel/qapplication_x11.cpp:3642
>19 0x00007ffff6e2037a in x11EventSourceDispatch (s=0x737800, callback=0x0, user_data=0x0)     at /usr/src/debug/dev-qt/qtgui-4.8.6-r4/qt-everywhere-opensource-src-4.8.6/src/gui/kernel/qguieventdispatcher_glib.cpp:146 
>20 0x00007ffff49aba8e in g_main_dispatch (context=0x736220) at /usr/src/debug/dev-libs/glib-2.48.2/glib-2.48.2/glib/gmain.c:3154
>21 g_main_context_dispatch (context=context@entry=0x736220) at /usr/src/debug/dev-libs/glib-2.48.2/glib-2.48.2/glib/gmain.c:3769
>22 0x00007ffff49abcf0 in g_main_context_iterate (context=context@entry=0x736220, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>)     at /usr/src/debug/dev-libs/glib-2.48.2/glib-2.48.2/glib/gmain.c:3840
>23 0x00007ffff49abd9c in g_main_context_iteration (context=0x736220, may_block=may_block@entry=1) at /usr/src/debug/dev-libs/glib-2.48.2/glib-2.48.2/glib/gmain.c:3901
>24 0x00007ffff652cb0e in QEventDispatcherGlib::processEvents (this=0x735eb0, flags=...)     at /usr/src/debug/dev-qt/qtcore-4.8.6-r2/qt-everywhere-opensource-src-4.8.6/src/corelib/kernel/qeventdispatcher_glib.cpp:450
>25 0x00007ffff6e2043e in QGuiEventDispatcherGlib::processEvents (this=<optimized out>, flags=...)     at /usr/src/debug/dev-qt/qtgui-4.8.6-r4/qt-everywhere-opensource-src-4.8.6/src/gui/kernel/qguieventdispatcher_glib.cpp:204
>26 0x00007ffff64fb0a1 in QEventLoop::processEvents (this=this@entry=0x7fffffffe180, flags=...) >    at /usr/src/debug/dev-qt/qtcore-4.8.6-r2/qt-everywhere-opensource-src-4.8.6/src/corelib/kernel/qeventloop.cpp:149
>27 0x00007ffff64fb3dd in QEventLoop::exec (this=this@entry=0x7fffffffe180, flags=...)     at /usr/src/debug/dev-qt/qtcore-4.8.6-r2/qt-everywhere-opensource-src-4.8.6/src/corelib/kernel/qeventloop.cpp:204
>28 0x00007ffff6500ef1 in QCoreApplication::exec () at /usr/src/debug/dev-qt/qtcore-4.8.6-r2/qt-everywhere-opensource-src-4.8.6/src/corelib/kernel/qcoreapplication.cpp:1225
>29 0x0000000000412680 in main ()

Danysan1, 2016-11-22:
This should fix it: https://github.com/Danysan1/radeon-profile/commit/1c1e051aacd4236c2ec3242d114cb541596a3795
(I say 'should' because i can't test it since I'm currently using a spare fanless card)

gerddie, 2016-11-23:
Thanks for the quick answer, unfortunately it doesn't fix the segfault. 

The more exact backtrace for clicking *Custom curve* is 

>0  QList<QAction*>::QList (l=..., this=0x7fffffffd300, this@entry=0x7fffffffd2a0) at ../../../qt-everywhere-opensource-src-4.8.6/src/corelib/tools/qlist.h:122
>1  QWidget::actions (this=0x7fffffffe558) at /usr/src/debug/dev-qt/qtgui-4.8.6-r4/qt-everywhere-opensource-src-4.8.6/src/gui/kernel/qwidget.cpp:3344
>2  0x000000000048699b in radeon_profile::findCurrentFanProfileMenuIndex (this=this@entry=0x7fffffffe2a0) at uiEvents.cpp:505
>3  0x0000000000486cd9 in radeon_profile::setCurrentFanProfile (this=this@entry=0x7fffffffe2a0, profileName=..., profile=...) at uiEvents.cpp:541
>4  0x0000000000486ea9 in radeon_profile::on_btn_pwmProfile_clicked (this=0x7fffffffe2a0) at uiEvents.cpp:84

Specifically, in radeon_profile::findCurrentFanProfileMenuIndex the 
**fanProfilesMenu->actions().count()* function call segfaults. 

When clicking *Fixed* the backtrace is 

> 0 QList<QAction*>::QList (l=..., this=0x7fffffffd3b0, this@entry=0x7fffffffd3a0) at ../../../qt-everywhere-opensource-src-4.8.6/src/corelib/tools/qlist.h:122
> 1  QWidget::actions (this=0x7fffffffe558) at /usr/src/debug/dev-qt/qtgui-4.8.6-r4/qt-everywhere-opensource-src-4.8.6/src/gui/kernel/qwidget.cpp:3344
> 2  0x0000000000486bd0 in radeon_profile::on_btn_pwmFixed_clicked (this=0x7fffffffe2a0) at uiEvents.cpp:63

and it is **fanProfilesMenu->actions()[1]->setChecked(true);** the results in the segfault. 

Now splitting of that line into 

    auto act = anProfilesMenu->actions()
    act[1]->setChecked(true); 

shows that the first line, i.e. copying the actions in the Qlist copy constructor results in the failure, which means that these actions are probably not initialized in a way they should be. 

Maybe this is related to the fact that the overclocking tab and/or  the profiles tab on the GPU Summary screen are disabled.

Danysan1, 2016-11-23:
Your last suggestion may actually be the real problem: basing on what I read here ( https://github.com/marazmista/radeon-profile/issues/17#issuecomment-207495241 ) your card should not support fan control, so your fan tab should be grayed out.

And in fact the setupUiEnabledFeatures() function does wrongly gray out the tab 2 (overclock) instead of the tab 3 (fan control): https://github.com/Danysan1/radeon-profile/commit/9d13bdab0d3e408cf6524155e5b21442eb01e63b#diff-cda7bc0581cc82ab9f765694a7a02bc6L204
Could you try this commit?

marazmista, 2016-11-23:
@gerddie 
Does the application display fan speed? I don't think that  R600 has fan control support so the bug might be that fan control tab isn't disabled and you have access to those buttons.

> Maybe this is related to the fact that the overclocking tab and/or the profiles tab on the GPU Summary screen are disabled.

Profile tab and DPM are mutually exclusive so nothing wrong here if you use DPM.

gerddie, 2016-11-23:
@Danysan:  I'll try this
@marazmista: Yes, it may be that the driver doesn't support it, the card  definitely does, since with the frglx driver and aticontrol I can set the speed.

gerddie, 2016-11-23:
@Danysan1 Okay, with the patch *overclock* and *fanspeed* are disabled

marazmista, 2016-11-23:
You can check if you have files named `pwm*` in `/sys/class/drm/cardX/device/hwmon/hwmonX/`. If not, there is no support.

marazmista, 2016-11-23:
Yes, the bug definitely is where @Danysan1 pointed out, wrong tab being disabled when no fan control available. Fixed here 47fc81b3a95622603190aecc2aa8a2c741b09483

gerddie, 2016-11-23:
Yes, no fan support in r600l. I've also looked at the r600_dpm.c kernel code, and (unlike with si_dpm.c) there is no code for fan control, and the radeon_asic.c there are also no functions assigned to the fan control interface.

gerddie, 2016-11-23:
This last commit properly grays out *fanspeed*. *overclock* is also disabled, but I guess that this is also to be expected because the file /sys/class/drm/card0/device/pp_sclk_od does not exist on my system. 
Closing this issue.

closed 2016-11-23 by gerddie

gerddie, 2016-11-23:
[Upstream bug](https://bugs.freedesktop.org/show_bug.cgi?id=90263) for fan control in R600.