marazmista/radeon-profile #33: Doesn't seem to want to work with an RX480.

tenten8401, 2016-11-27:
Hey. Got a fresh install of Antergos Linux and I'm trying to use radeon-profile, however despite whatever I try, I can't get it to work. It'll come up, "Can't read data, You need debugfs mounted and either root rights or the daemon running"... I can launch it with root and nothing. Debugfs is mounted.

My driver is radeonsi and the card is an XFX RX480 8 GB. Would love support for this program as Driconf is just horrible.

marazmista, 2016-11-27:
RX480 use `amdgpu` kernel driver. There is amdgpu branch that is not merged yet, you can try to compile and run like this: https://github.com/marazmista/radeon-profile/issues/11#issuecomment-241196351

Also, I think radeon-profile doesn't have same functionality as Driconf. It is more about card monitoring than configure OpenGL settings.

tenten8401, 2016-11-27:
Well, I'm mostly hoping to overclock with it. I'll give amdgpu a shot.

tenten8401, 2016-11-27:
Works! Thank you!

closed 2016-11-27 by tenten8401