marazmista/radeon-profile #34: 0% Fan Speed

tenten8401, 2016-11-27:
Can we have the option for 0% fan speed? I have an XFX RX480 GTR that will go to 0% in idle.

marazmista, 2016-11-27:
You can click Auto on fan control tab and then card decide itself if it should go to 0% when idle.

Enigma0, 2016-11-29:
Auto is terrible though in every other respect. Ditto on wanting 0% fan speed functionality in the Custom Fan Curve.

Danysan1, 2016-11-29:
Editing this line should be enough:
https://github.com/marazmista/radeon-profile/blob/master/radeon-profile/radeon_profile.h#L58

However I think that before we should implement some check  to prevent an user from frying its own card with an extreme fan profile...

marazmista, 2016-11-30:
Ok, I will enable this in the future.

marazmista, 2016-12-05:
Option added in commit dd35b94f6511c017424361e39be81e9045c82e1a

closed 2016-12-08 by marazmista