marazmista/radeon-profile #42: Maximum number of clients reachedError loading connectors: can't connect to X server

blisbell, 2016-12-23:
system specs
`System:    Host: htpc Kernel: 4.9.0-040900-generic x86_64 (64 bit)
           Desktop: Cinnamon 3.2.7  Distro: Linux Mint 18.1 Serena
Machine:   Mobo: ASUSTeK model: M5A97 PLUS v: Rev X.0x
           Bios: American Megatrends v: 0901 date: 04/15/2016
CPU:       Hexa core AMD Phenom II X6 1045T (-MCP-) cache: 3072 KB 
           clock speeds: max: 2700 MHz 1: 1400 MHz 2: 800 MHz 3: 1400 MHz
           4: 2000 MHz 5: 1400 MHz 6: 2000 MHz
Graphics:  Card: Advanced Micro Devices [AMD/ATI] Device 67ef
           Display Server: X.Org 1.18.4 drivers: ati,amdgpu (unloaded: fbdev,vesa,radeon)
           Resolution: 1920x1080@60.00hz
           GLX Renderer: Gallium 0.4 on AMD POLARIS11 (DRM 3.8.0 / 4.9.0-040900-generic, LLVM 4.0.0)
           GLX Version: 3.0 Mesa 13.1.0-devel - padoka PPA
Audio:     Card-1 Advanced Micro Devices [AMD/ATI] Device aae0
           driver: snd_hda_intel
           Card-2 Advanced Micro Devices [AMD/ATI] SBx00 Azalia (Intel HDA)
           driver: snd_hda_intel
           Sound: Advanced Linux Sound Architecture v: k4.9.0-040900-generic
Network:   Card: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
           driver: r8169
           IF: enp2s0 state: up speed: 1000 Mbps duplex: full mac: <filter>
Drives:    HDD Total Size: 1240.3GB (28.4% used)
           ID-1: /dev/sda model: PNY_CS1311_240GB size: 240.1GB
           ID-2: /dev/sdb model: ST1000DM003 size: 1000.2GB
Partition: ID-1: / size: 218G used: 45G (22%) fs: ext4 dev: /dev/sda1
           ID-2: swap-1 size: 2.50GB used: 0.00GB (0%) fs: swap dev: /dev/sda4
RAID:      No RAID devices: /proc/mdstat, md_mod kernel module present
Sensors:   System Temperatures: cpu: 14.5C mobo: N/A
           Fan Speeds (in rpm): cpu: 0
Info:      Processes: 245 Uptime: 14:51 Memory: 2460.1/15966.2MB
           Client: Shell (bash) inxi: 2.2.35`

I am getting this error after 5min or so of program being open or minimized to tray. As sone as its closed it goes away but nothing else can be opened until then.

`"sni-qt/8288" WARN  04:13:30.833 void StatusNotifierItemFactory::connectToSnw() Invalid interface to SNW_SERVICE 
Maximum number of clients reachedError loading connectors: can't connect to X server`

sharique, 2016-12-23:
Just curios which repo you are using for latest kernel packages?

blisbell, 2016-12-23:
[http://kernel.ubuntu.com/~kernel-ppa/mainline/v4.9/]

Danysan1, 2016-12-23:
Are you running as root or using the daemon?

blisbell, 2016-12-23:
just to add i have a xfx rx460 and can reproduce this on my mint setup and antergos setup.
an to note i need the 4.9 kernel for my win tv quad hd atsc tuner to work other wise i have to build it from media_build git with 4.4.15 or higher kernel for tuner.

marazmista, 2016-12-23:
I've found this thread: https://bbs.archlinux.org/viewtopic.php?id=188052. Can you show output of: `lsof +c0 /usr/lib/libX11.so | cut -d' ' -f1`?

Also, there are two bugs https://bugs.freedesktop.org/show_bug.cgi?id=86665 and https://bugs.freedesktop.org/show_bug.cgi?id=9209 with some changes coming to server 1.19. 
Do you have any other Qt app that use tray icon and possibly have similar problem?

blisbell, 2016-12-23:
`lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
lsof: status error on /usr/lib/libX11.so: No such file or directory
lsof 4.89
 latest revision: ftp://lsof.itap.purdue.edu/pub/tools/unix/lsof/
 latest FAQ: ftp://lsof.itap.purdue.edu/pub/tools/unix/lsof/FAQ
 latest man page: ftp://lsof.itap.purdue.edu/pub/tools/unix/lsof/lsof_man
 usage: [-?abhKlnNoOPRtUvVX] [+|-c c] [+|-d s] [+D D] [+|-E] [+|-e s] [+|-f[gG]]
 [-F [f]] [-g [s]] [-i [i]] [+|-L [l]] [+m [m]] [+|-M] [-o [o]] [-p s]
 [+|-r [t]] [-s [p:s]] [-S [t]] [-T [t]] [-u s] [+|-w] [-x [fl]] [--] [names]
Use the ``-h'' option to get more help information.
`

only qbittorrent but no problems with it

blisbell, 2016-12-23:
i used the other command in arch link an got this

lsof | awk '/libX11.so/ {clients[$1]++;} END {for(c in clients){printf "%s\t%s\n", clients[c], c;}}'
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
9	HTMLParse
12	TVRecEven
14	chrome
1	Qt\x20bea
4	SimpleCac
1	inotify_r
2	DVBRead
15	conky
1	Expire
1	Chrome_Sy
1	cinnamon
2	Chrome_Fi
4	ScriptStr
3	dbus-laun
1	D-Bus\x20
1	Chrome_Hi
1	Media
1	polkit-gn
3	FFmpegDem
1	Qt\x20HTT
1	renderer_
2	mintUpdat
3	mythbacke
1	threaded-
2	DeviceRea
1	Scheduler
1	csd-print
1	xscreensa
34	WorkerPoo
1	JobQueue
3	amdgpu_cs
5	cinnamon-
1	CachePool
1	gnome-ter
1	CrShutdow
1	DownloadM
1	Xorg
1	sandbox_i
1	SSDP
2	TFWWrite
1	update-no
1	Logger
1	gpu-proce
1	Chrome_Pr
1	TaskQueue
2	alsa-sink
1	Chrome_Ca
1	at-spi-bu
1	Chrome_IO
1	nm-applet
10	Chrome_Ch
16	gdbus
3	alsa-sour
1	HouseKeep
1	NetworkCh
2	TFWSync
1	at-spi2-r
2	mdm
46	Composito
12	si_shader
1	mono
1	PreviewGe
9	Renderer:
1	Networkin
1	AudioThre
1	pulseaudi
1	Watchdog
1	IndexedDB
1	JS\x20GC
1	Chrome_DB
16	gmain
1	LogServer
1	extension
1	docky
12	dconf\x20
1	PT17
2	RecThread
1	nemo
1	LogForwar
1	Finalizer
1	JS\x20Sou
1	BrowserWa
3	BrowserBl
1	LevelDBEn

blisbell, 2016-12-23:
ok had to modify command 

lsof +c0 /usr/lib/x86_64-linux-gnu/libX11.so | cut -d' ' -f1
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
COMMAND
mdm
mdm
Xorg
dbus-launch
cinnamon-sessio
dbus-launch
at-spi-bus-laun
at-spi2-registr
cinnamon-settin
pulseaudio
csd-printer
cinnamon-launch
cinnamon
docky
polkit-gnome-au
nemo
nm-applet
cinnamon-killer
xscreensaver
cinnamon-screen
mintUpdate
mythbackend
update-notifier
conky
conky
chrome
chrome
chrome
chrome
chrome
chrome
chrome
chrome
chrome
chrome
chrome
chrome
gnome-terminal-
dbus-launch

and output of this command 

lsof -p `pidof Xorg`
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
COMMAND  PID USER   FD      TYPE             DEVICE SIZE/OFF       NODE NAME
Xorg    1452 root  cwd       DIR                8,1     4096   14025028 /var/lib/mdm
Xorg    1452 root  rtd       DIR                8,1     4096          2 /
Xorg    1452 root  txt       REG                8,1  2402920    3409097 /usr/lib/xorg/Xorg
Xorg    1452 root  DEL       REG                0,5           142966795 /SYSV00000000
Xorg    1452 root  DEL       REG                0,5              327681 /SYSV00000000
Xorg    1452 root  DEL       REG                0,5              163840 /SYSV00000000
Xorg    1452 root  mem       REG                8,1    47600    1966289 /lib/x86_64-linux-gnu/libnss_files-2.23.so
Xorg    1452 root  mem       REG                8,1    47648    1966274 /lib/x86_64-linux-gnu/libnss_nis-2.23.so
Xorg    1452 root  mem       REG                8,1    93128    1966285 /lib/x86_64-linux-gnu/libnsl-2.23.so
Xorg    1452 root  mem       REG                8,1    35688    1966286 /lib/x86_64-linux-gnu/libnss_compat-2.23.so
Xorg    1452 root  DEL       REG                0,5           132743177 /SYSV00000000
Xorg    1452 root  mem       CHR              226,0               14510 /dev/dri/card0
Xorg    1452 root  DEL       REG                0,5             1179656 /SYSV00000000
Xorg    1452 root  DEL       REG                0,5              819206 /SYSV00000000
Xorg    1452 root  DEL       REG                0,5              524291 /SYSV00000000
Xorg    1452 root  mem       REG                8,1    30848    3417889 /usr/lib/x86_64-linux-gnu/libxcb-xfixes.so.0.0.0
Xorg    1452 root  mem       REG                8,1   214784    3801686 /usr/lib/x86_64-linux-gnu/mesa-egl/libEGL.so.1.0.0
Xorg    1452 root  mem       REG                8,1  1003624    3416826 /usr/lib/x86_64-linux-gnu/libepoxy.so.0.0.0
[... 131 lines cut ...]
Xorg    1452 root   44u     unix 0xffff9d5333d8f000      0t0      87866 @/tmp/.X11-unix/X0 type=STREAM
Xorg    1452 root   45u     unix 0xffff9d52c85bb400      0t0      89210 @/tmp/.X11-unix/X0 type=STREAM
Xorg    1452 root   46u     unix 0xffff9d535a6b5800      0t0      34855 @/tmp/.X11-unix/X0 type=STREAM
Xorg    1452 root   47u     unix 0xffff9d535a6b6400      0t0      34858 @/tmp/.X11-unix/X0 type=STREAM
Xorg    1452 root   48u     unix 0xffff9d52c85bac00      0t0      85371 @/tmp/.X11-unix/X0 type=STREAM
Xorg    1452 root   49u     unix 0xffff9d5355a52800      0t0      46675 @/tmp/.X11-unix/X0 type=STREAM
Xorg    1452 root   50u     unix 0xffff9d535ac11800      0t0      44546 @/tmp/.X11-unix/X0 type=STREAM
Xorg    1452 root   52u     unix 0xffff9d5355a53000      0t0      30532 @/tmp/.X11-unix/X0 type=STREAM
Xorg    1452 root   53u     unix 0xffff9d50e11aac00      0t0      92739 @/tmp/.X11-unix/X0 type=STREAM
Xorg    1452 root  512u      REG                0,5        4      24158 /memfd:xshmfence (deleted)
Xorg    1452 root  513u      REG                0,5        4      26588 /memfd:xshmfence (deleted)
Xorg    1452 root  514u      REG                0,5        4      26581 /memfd:xshmfence (deleted)
Xorg    1452 root  515u      REG                0,5        4      25059 /memfd:xshmfence (deleted)
Xorg    1452 root  516u      REG                0,5        4      25061 /memfd:xshmfence (deleted)
Xorg    1452 root  517u      REG                0,5        4      25065 /memfd:xshmfence (deleted)
Xorg    1452 root  518u      REG                0,5        4      25069 /memfd:xshmfence (deleted)
Xorg    1452 root  519u      REG                0,5        4      27241 /memfd:xshmfence (deleted)
Xorg    1452 root  520u      REG                0,5        4      86653 /memfd:xshmfence (deleted)
Xorg    1452 root  521u      REG                0,5        4      86673 /memfd:xshmfence (deleted)
Xorg    1452 root  522u      REG                0,5        4      32060 /memfd:xshmfence (deleted)
Xorg    1452 root  523u      REG                0,5        4      82791 /memfd:xshmfence (deleted)
Xorg    1452 root  524u      REG                0,5        4      85397 /memfd:xshmfence (deleted)
Xorg    1452 root  525u      REG                0,5        4      88306 /memfd:xshmfence (deleted)
Xorg    1452 root  526u      REG                0,5        4      90595 /memfd:xshmfence (deleted)
Xorg    1452 root  527u      REG                0,5        4     260238 /memfd:xshmfence (deleted)

marazmista, 2016-12-23:
Have you executed those commands with radeon-profile running?

blisbell, 2016-12-23:
lsof -p `pidof Xorg`
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
COMMAND  PID USER   FD      TYPE             DEVICE SIZE/OFF       NODE NAME
Xorg    1452 root  cwd       DIR                8,1     4096   14025028 /var/lib/mdm
Xorg    1452 root  rtd       DIR                8,1     4096          2 /
Xorg    1452 root  txt       REG                8,1  2402920    3409097 /usr/lib/xorg/Xorg
Xorg    1452 root  mem       CHR              226,0               14510 /dev/dri/card0
Xorg    1452 root  DEL       REG                0,5           427982861 /SYSV00000000
Xorg    1452 root  DEL       REG                0,5           427851786 /SYSV00000000
Xorg    1452 root  DEL       REG                0,5           142966795 /SYSV00000000
Xorg    1452 root  DEL       REG                0,5              327681 /SYSV00000000
Xorg    1452 root  DEL       REG                0,5              163840 /SYSV00000000
Xorg    1452 root  mem       REG                8,1    47600    1966289 /lib/x86_64-linux-gnu/libnss_files-2.23.so
Xorg    1452 root  mem       REG                8,1    47648    1966274 /lib/x86_64-linux-gnu/libnss_nis-2.23.so
Xorg    1452 root  mem       REG                8,1    93128    1966285 /lib/x86_64-linux-gnu/libnsl-2.23.so
Xorg    1452 root  mem       REG                8,1    35688    1966286 /lib/x86_64-linux-gnu/libnss_compat-2.23.so
Xorg    1452 root  DEL       REG                0,5           132743177 /SYSV00000000
Xorg    1452 root  DEL       REG                0,5             1179656 /SYSV00000000
Xorg    1452 root  DEL       REG                0,5              819206 /SYSV00000000
Xorg    1452 root  DEL       REG                0,5              524291 /SYSV00000000
Xorg    1452 root  mem       REG                8,1    30848    3417889 /usr/lib/x86_64-linux-gnu/libxcb-xfixes.so.0.0.0
Xorg    1452 root  mem       REG                8,1   214784    3801686 /usr/lib/x86_64-linux-gnu/mesa-egl/libEGL.so.1.0.0
Xorg    1452 root  mem       REG                8,1  1003624    3416826 /usr/lib/x86_64-linux-gnu/libepoxy.so.0.0.0
Xorg    1452 root  mem       REG                8,1   221248    3409587 /usr/lib/xorg/modules/libglamoregl.so
Xorg    1452 root  mem       REG                8,1   227384    3417718 /usr/lib/x86_64-linux-gnu/libtxc_dxtn_s2tc.so.0.0.0
Xorg    1452 root  mem       REG                8,1    81040    1970672 /lib/x86_64-linux-gnu/libbsd.so.0.8.2
Xorg    1452 root  mem       REG                8,1   167240    1970848 /lib/x86_64-linux-gnu/libtinfo.so.5.9
Xorg    1452 root  mem       REG                8,1   212168    3416814 /usr/lib/x86_64-linux-gnu/libedit.so.2.0.53
Xorg    1452 root  mem       REG                8,1  1566440    3412476 /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
Xorg    1452 root  mem       REG                8,1 48524184    3411679 /usr/lib/x86_64-linux-gnu/libLLVM-4.0.so.1
Xorg    1452 root  mem       REG                8,1    96824    3416819 /usr/lib/x86_64-linux-gnu/libelf-0.165.so
Xorg    1452 root  mem       REG                8,1    30936    3408443 /usr/lib/x86_64-linux-gnu/libdrm_nouveau.so.2.0.0
Xorg    1452 root  mem       REG                8,1   143408    3408545 /usr/lib/x86_64-linux-gnu/libdrm_intel.so.1.0.0
Xorg    1452 root  mem       REG                8,1   219336    3417399 /usr/lib/x86_64-linux-gnu/libnettle.so.6.2
Xorg    1452 root  mem       REG                8,1 10810792    3801195 /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
Xorg    1452 root  mem       REG                8,1   142200    3424249 /usr/lib/xorg/modules/libfb.so
Xorg    1452 root  DEL       REG                0,5           427884556 /SYSV00000000
Xorg    1452 root  DEL       REG                0,5            36667396 /SYSV00000000
Xorg    1452 root  DEL       REG                0,5             1146887 /SYSV00000000
Xorg    1452 root  mem       REG                8,1   100400    3416835 /usr/lib/x86_64-linux-gnu/libevdev.so.2.1.12
Xorg    1452 root  mem       REG                8,1    22512    3417357 /usr/lib/x86_64-linux-gnu/libmtdev.so.1.0.0
Xorg    1452 root  mem       REG                8,1    61336    3417961 /usr/lib/xorg/modules/input/evdev_drv.so
Xorg    1452 root  mem       REG                8,1    47336    3408687 /usr/lib/x86_64-linux-gnu/libdrm_radeon.so.1.0.1
Xorg    1452 root  DEL       REG                0,5               32060 /memfd:xshmfence
Xorg    1452 root  DEL       REG                0,5               90595 /memfd:xshmfence
Xorg    1452 root  DEL       REG                0,5              491522 /SYSV00000000
Xorg    1452 root  DEL       REG                0,5              356813 /memfd:xshmfence
Xorg    1452 root  DEL       REG                0,5               86673 /memfd:xshmfence
Xorg    1452 root  DEL       REG                0,5               88306 /memfd:xshmfence
Xorg    1452 root  DEL       REG                0,5               85397 /memfd:xshmfence
Xorg    1452 root  DEL       REG                0,5               86653 /memfd:xshmfence
Xorg    1452 root  DEL       REG                0,5               25069 /memfd:xshmfence
Xorg    1452 root  DEL       REG                0,5               25065 /memfd:xshmfence
Xorg    1452 root  DEL       REG                0,5               25061 /memfd:xshmfence
Xorg    1452 root  DEL       REG                0,5               25059 /memfd:xshmfence
[... cut ...]
Xorg    1452 root  mem       REG                8,1     5992    3416467 /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1.0.0
Xorg    1452 root  mem       REG                8,1    22584    3416488 /usr/lib/x86_64-linux-gnu/libXfixes.so.3.1.0
[... 150 lines cut ...]
mythbackend
update-notifier
conky
conky
chrome
chrome
chrome
chrome
chrome
chrome
chrome
chrome
chrome
chrome
chrome
chrome
gnome-terminal-
chrome
chrome
dbus-launch
chrome
chrome
radeon-profile

this is with window open looking to build xorg 1.19 any1 know of a ppa with it?

blisbell, 2016-12-23:
well cant build xserver some package in mint 18.1-ubuntu 16.04 are too old.

marazmista, 2016-12-23:
Have you had this issue with previous versions or this is your first install?

blisbell, 2016-12-23:
at 1208 version i had a cant read mount error it was fixed but this issue popped up with the last 2.

blisbell, 2016-12-23:
also tried building with qt4 and qt5 the same.

blisbell, 2017-01-19:
seems to have fixed it's self closing

closed 2017-01-19 by blisbell