marazmista/radeon-profile #46: Switching cards

intagger, 2017-03-26:
I have two RX 480. When I switch from card0 to card1 Radeon Profile just disappears.
```
[1]    5606 segmentation fault (core dumped)  radeon-profile
```

edison12347, 2017-05-21:
I have exactly the same issue on RX 570, RX580, RX470

marazmista, 2017-05-25:
Could you post output of `ls /sys/class/drm/*` and contents of `uevent` files inside `/sys/class/drm/cardX/device` directory?

joekiller, 2017-05-26:
I too have the problem where only the intel card shows up

```
 ls /sys/class/drm/* 
/sys/class/drm/version

/sys/class/drm/card0:
dev
device
power
subsystem
uevent

/sys/class/drm/card1:
card1-DP-1
card1-DP-2
card1-DP-3
card1-eDP-1
card1-HDMI-A-1
card1-HDMI-A-2
card1-HDMI-A-3
dev
device
error
gt_act_freq_mhz
gt_boost_freq_mhz
gt_cur_freq_mhz
gt_max_freq_mhz
gt_min_freq_mhz
gt_RP0_freq_mhz
gt_RP1_freq_mhz
gt_RPn_freq_mhz
power
subsystem
uevent

/sys/class/drm/card1-DP-1:
device
dpms
drm_dp_aux1
edid
enabled
i2c-13
modes
power
status
subsystem
uevent

/sys/class/drm/card1-DP-2:
device
dpms
drm_dp_aux2
edid
enabled
i2c-14
modes
power
status
subsystem
uevent

/sys/class/drm/card1-DP-3:
device
dpms
drm_dp_aux3
edid
enabled
i2c-15
modes
[... 358 lines cut ...]
	Capabilities: [b0] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [148] Device Serial Number 00-00-00-01-00-4c-e0-00
	Capabilities: [158] Latency Tolerance Reporting
	Capabilities: [160] L1 PM Substates
	Kernel driver in use: rtsx_pci
	Kernel modules: rtsx_pci

3d:00.0 Non-Volatile memory controller: Toshiba America Info Systems Device 010f (rev 01) (prog-if 02 [NVM Express])
	Subsystem: Toshiba America Info Systems Device 0001
	Flags: bus master, fast devsel, latency 0, IRQ 16, NUMA node 0
	Memory at ec000000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [40] Power Management version 3
	Capabilities: [50] MSI: Enable- Count=1/8 Maskable- 64bit+
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [b0] MSI-X: Enable+ Count=8 Masked-
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [168] Alternative Routing-ID Interpretation (ARI)
	Capabilities: [178] #19
	Capabilities: [198] Latency Tolerance Reporting
	Capabilities: [1a0] L1 PM Substates
	Kernel driver in use: nvme
	Kernel modules: nvme

```

edison12347, 2017-05-28:
Here is stats that you asked:
ls /sys/class/drm/* 
/sys/class/drm/version

/sys/class/drm/amdttm:
buffer_objects  memory_accounting  power  subsystem  uevent

/sys/class/drm/card0:
card0-DP-1  card0-DP-2  card0-DVI-D-1  card0-HDMI-A-1  card0-HDMI-A-2  dev  device  power  subsystem  uevent

/sys/class/drm/card0-DP-1:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card0-DP-2:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card0-DVI-D-1:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card0-HDMI-A-1:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card0-HDMI-A-2:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card1:
card1-DP-3  card1-DP-4  card1-DVI-D-2  card1-HDMI-A-3  card1-HDMI-A-4  dev  device  power  subsystem  uevent

/sys/class/drm/card1-DP-3:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card1-DP-4:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card1-DVI-D-2:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card1-HDMI-A-3:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card1-HDMI-A-4:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card2:
card2-DP-5  card2-DP-6  card2-DVI-D-3  card2-HDMI-A-5  card2-HDMI-A-6  dev  device  power  subsystem  uevent

/sys/class/drm/card2-DP-5:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card2-DP-6:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card2-DVI-D-3:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card2-HDMI-A-5:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card2-HDMI-A-6:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card3:
card3-DP-7  card3-DP-8  card3-DVI-D-4  card3-HDMI-A-7  card3-HDMI-A-8  dev  device  power  subsystem  uevent

/sys/class/drm/card3-DP-7:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card3-DP-8:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card3-DVI-D-4:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card3-HDMI-A-7:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card3-HDMI-A-8:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/controlD64:
dev  device  power  subsystem  uevent

/sys/class/drm/controlD65:
dev  device  power  subsystem  uevent

/sys/class/drm/controlD66:
dev  device  power  subsystem  uevent

/sys/class/drm/controlD67:
dev  device  power  subsystem  uevent

/sys/class/drm/renderD128:
dev  device  power  subsystem  uevent

/sys/class/drm/renderD129:
dev  device  power  subsystem  uevent

/sys/class/drm/renderD130:
dev  device  power  subsystem  uevent

/sys/class/drm/renderD131:
dev  device  power  subsystem  uevent

for file in /sys/class/drm/card*/device/uevent; do echo $file; cat $file;done
/sys/class/drm/card0/device/uevent
DRIVER=amdgpu
PCI_CLASS=30000
PCI_ID=1002:67DF
PCI_SUBSYS_ID=1DA2:E366
PCI_SLOT_NAME=0000:02:00.0
MODALIAS=pci:v00001002d000067DFsv00001DA2sd0000E366bc03sc00i00
/sys/class/drm/card0-DP-1/device/uevent
MAJOR=226
MINOR=0
DEVNAME=dri/card0
DEVTYPE=drm_minor
/sys/class/drm/card0-DP-2/device/uevent
MAJOR=226
MINOR=0
DEVNAME=dri/card0
DEVTYPE=drm_minor
/sys/class/drm/card0-DVI-D-1/device/uevent
MAJOR=226
MINOR=0
DEVNAME=dri/card0
DEVTYPE=drm_minor
/sys/class/drm/card0-HDMI-A-1/device/uevent
MAJOR=226
MINOR=0
DEVNAME=dri/card0
DEVTYPE=drm_minor
/sys/class/drm/card0-HDMI-A-2/device/uevent
MAJOR=226
MINOR=0
DEVNAME=dri/card0
DEVTYPE=drm_minor
/sys/class/drm/card1/device/uevent
DRIVER=amdgpu
PCI_CLASS=30000
PCI_ID=1002:67DF
PCI_SUBSYS_ID=174B:E347
PCI_SLOT_NAME=0000:03:00.0
MODALIAS=pci:v00001002d000067DFsv0000174Bsd0000E347bc03sc00i00
/sys/class/drm/card1-DP-3/device/uevent
MAJOR=226
MINOR=1
DEVNAME=dri/card1
DEVTYPE=drm_minor
/sys/class/drm/card1-DP-4/device/uevent
MAJOR=226
MINOR=1
DEVNAME=dri/card1
DEVTYPE=drm_minor
/sys/class/drm/card1-DVI-D-2/device/uevent
MAJOR=226
MINOR=1
DEVNAME=dri/card1
DEVTYPE=drm_minor
/sys/class/drm/card1-HDMI-A-3/device/uevent
MAJOR=226
MINOR=1
DEVNAME=dri/card1
DEVTYPE=drm_minor
/sys/class/drm/card1-HDMI-A-4/device/uevent
MAJOR=226
MINOR=1
DEVNAME=dri/card1
DEVTYPE=drm_minor
/sys/class/drm/card2/device/uevent
DRIVER=amdgpu
PCI_CLASS=30000
PCI_ID=1002:67DF
PCI_SUBSYS_ID=1DA2:E353
PCI_SLOT_NAME=0000:04:00.0
MODALIAS=pci:v00001002d000067DFsv00001DA2sd0000E353bc03sc00i00
/sys/class/drm/card2-DP-5/device/uevent
MAJOR=226
MINOR=2
DEVNAME=dri/card2
DEVTYPE=drm_minor
/sys/class/drm/card2-DP-6/device/uevent
MAJOR=226
MINOR=2
DEVNAME=dri/card2
DEVTYPE=drm_minor
/sys/class/drm/card2-DVI-D-3/device/uevent
MAJOR=226
MINOR=2
DEVNAME=dri/card2
DEVTYPE=drm_minor
/sys/class/drm/card2-HDMI-A-5/device/uevent
MAJOR=226
MINOR=2
DEVNAME=dri/card2
DEVTYPE=drm_minor
/sys/class/drm/card2-HDMI-A-6/device/uevent
MAJOR=226
MINOR=2
DEVNAME=dri/card2
DEVTYPE=drm_minor
/sys/class/drm/card3/device/uevent
DRIVER=amdgpu
PCI_CLASS=30000
PCI_ID=1002:67DF
PCI_SUBSYS_ID=1DA2:E353
PCI_SLOT_NAME=0000:05:00.0
MODALIAS=pci:v00001002d000067DFsv00001DA2sd0000E353bc03sc00i00
/sys/class/drm/card3-DP-7/device/uevent
MAJOR=226
MINOR=3
DEVNAME=dri/card3
DEVTYPE=drm_minor
/sys/class/drm/card3-DP-8/device/uevent
MAJOR=226
MINOR=3
DEVNAME=dri/card3
DEVTYPE=drm_minor
/sys/class/drm/card3-DVI-D-4/device/uevent
MAJOR=226
MINOR=3
DEVNAME=dri/card3
DEVTYPE=drm_minor
/sys/class/drm/card3-HDMI-A-7/device/uevent
MAJOR=226
MINOR=3
DEVNAME=dri/card3
DEVTYPE=drm_minor
/sys/class/drm/card3-HDMI-A-8/device/uevent
MAJOR=226
MINOR=3
DEVNAME=dri/card3
DEVTYPE=drm_minor

marazmista, 2017-06-13:
Thanks for outputs. I pushed update to rp-dev branch. Build instructions are here: https://github.com/marazmista/radeon-profile/tree/rp-dev#build, just replace first line to: `git clone -b rp-dev https://github.com/marazmista/radeon-profile.git`

predacon, 2017-06-15:
same problem here after trying your update with two rx 470 and iGPU HD530 as main.

## ls /sys/class/drm/* ##

/sys/class/drm/version

/sys/class/drm/card0:
card0-DP-1  card0-DP-2  card0-DP-3  card0-HDMI-A-1  card0-HDMI-A-2  card0-HDMI-A-3  dev  device  error  gt_act_freq_mhz  gt_boost_freq_mhz  gt_cur_freq_mhz  gt_max_freq_mhz  gt_min_freq_mhz  gt_RP0_freq_mhz  gt_RP1_freq_mhz  gt_RPn_freq_mhz  power  subsystem  uevent

/sys/class/drm/card0-DP-1:
device  dpms  drm_dp_aux0  edid  enabled  i2c-3  modes  power  status  subsystem  uevent

/sys/class/drm/card0-DP-2:
device  dpms  drm_dp_aux1  edid  enabled  i2c-4  modes  power  status  subsystem  uevent

/sys/class/drm/card0-DP-3:
device  dpms  drm_dp_aux2  edid  enabled  i2c-5  modes  power  status  subsystem  uevent

/sys/class/drm/card0-HDMI-A-1:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card0-HDMI-A-2:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card0-HDMI-A-3:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card1:
card1-DP-4  card1-DVI-D-1  card1-DVI-D-2  card1-HDMI-A-4  dev  device  power  subsystem  uevent

/sys/class/drm/card1-DP-4:
device  dpms  drm_dp_aux3  edid  enabled  i2c-15  modes  power  status  subsystem  uevent

/sys/class/drm/card1-DVI-D-1:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card1-DVI-D-2:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card1-HDMI-A-4:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card2:
card2-DP-5  card2-DVI-D-3  card2-DVI-D-4  card2-HDMI-A-5  dev  device  power  subsystem  uevent

/sys/class/drm/card2-DP-5:
device  dpms  drm_dp_aux4  edid  enabled  i2c-24  modes  power  status  subsystem  uevent

/sys/class/drm/card2-DVI-D-3:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card2-DVI-D-4:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/card2-HDMI-A-5:
device  dpms  edid  enabled  modes  power  status  subsystem  uevent

/sys/class/drm/renderD128:
dev  device  power  subsystem  uevent

/sys/class/drm/renderD129:
dev  device  power  subsystem  uevent

/sys/class/drm/renderD130:
dev  device  power  subsystem  uevent

/sys/class/drm/ttm:
buffer_objects  memory_accounting  power  subsystem  uevent

 ## uevent files ##

DRIVER=i915
PCI_CLASS=30000
PCI_ID=8086:1912
PCI_SUBSYS_ID=1849:1912
PCI_SLOT_NAME=0000:00:02.0
MODALIAS=pci:v00008086d00001912sv00001849sd00001912bc03sc00i00
/sys/class/drm/card0-DP-1/device/uevent
MAJOR=226
MINOR=0
DEVNAME=dri/card0
DEVTYPE=drm_minor
/sys/class/drm/card0-DP-2/device/uevent
MAJOR=226
MINOR=0
DEVNAME=dri/card0
DEVTYPE=drm_minor
/sys/class/drm/card0-DP-3/device/uevent
MAJOR=226
MINOR=0
DEVNAME=dri/card0
DEVTYPE=drm_minor
/sys/class/drm/card0-HDMI-A-1/device/uevent
MAJOR=226
MINOR=0
DEVNAME=dri/card0
DEVTYPE=drm_minor
/sys/class/drm/card0-HDMI-A-2/device/uevent
MAJOR=226
MINOR=0
DEVNAME=dri/card0
DEVTYPE=drm_minor
/sys/class/drm/card0-HDMI-A-3/device/uevent
MAJOR=226
MINOR=0
DEVNAME=dri/card0
DEVTYPE=drm_minor
/sys/class/drm/card1/device/uevent
DRIVER=amdgpu
PCI_CLASS=30000
PCI_ID=1002:67DF
PCI_SUBSYS_ID=1043:04B0
PCI_SLOT_NAME=0000:01:00.0
MODALIAS=pci:v00001002d000067DFsv00001043sd000004B0bc03sc00i00
/sys/class/drm/card1-DP-4/device/uevent
MAJOR=226
MINOR=1
DEVNAME=dri/card1
DEVTYPE=drm_minor
/sys/class/drm/card1-DVI-D-1/device/uevent
MAJOR=226
MINOR=1
DEVNAME=dri/card1
DEVTYPE=drm_minor
/sys/class/drm/card1-DVI-D-2/device/uevent
MAJOR=226
MINOR=1
DEVNAME=dri/card1
DEVTYPE=drm_minor
/sys/class/drm/card1-HDMI-A-4/device/uevent
MAJOR=226
MINOR=1
DEVNAME=dri/card1
DEVTYPE=drm_minor
/sys/class/drm/card2/device/uevent
DRIVER=amdgpu
PCI_CLASS=30000
PCI_ID=1002:67DF
PCI_SUBSYS_ID=1043:04B0
PCI_SLOT_NAME=0000:07:00.0
MODALIAS=pci:v00001002d000067DFsv00001043sd000004B0bc03sc00i00
/sys/class/drm/card2-DP-5/device/uevent
MAJOR=226
MINOR=2
DEVNAME=dri/card2
DEVTYPE=drm_minor
/sys/class/drm/card2-DVI-D-3/device/uevent
MAJOR=226
MINOR=2
DEVNAME=dri/card2
DEVTYPE=drm_minor
/sys/class/drm/card2-DVI-D-4/device/uevent
MAJOR=226
MINOR=2
DEVNAME=dri/card2
DEVTYPE=drm_minor
/sys/class/drm/card2-HDMI-A-5/device/uevent
MAJOR=226
MINOR=2
DEVNAME=dri/card2
DEVTYPE=drm_minor

predacon, 2017-06-15:
my bad. your new build works well without segfault when switching card. But dont shows any info about clocks. In console shows this error continuosly: DRM_IOCTL_AMDGPU_INFO: Invalid argument

lhorace, 2017-06-15:
Clocks are not available until Kernel 4.12 with AMDGPU.

See https://github.com/marazmista/radeon-profile/issues/54#issuecomment-308265956

predacon, 2017-06-15:
Then, why I can see memory clock and core clock with actual radeon-profile build from AUR?
[image]

predacon, 2017-06-15:
oh, it seems im using radeonsi driver instead AMDGPU. strange...

lhorace, 2017-06-15:
Are you sure you are using radeonsi on a RX470? AFAIK, only AMDGPU support Polaris and newer GPUs. Also, the stable, current build, polls information form /sys. The Dev branch uses IOCTLS and AMDGPU in <4.11 don't provide clock information.

predacon, 2017-06-15:
I have no xf86-video-ati installed. no radeon module loaded but AMDGPU.
Makes me doubt this >> 
[image]

lhorace, 2017-06-16:
Probably a bug but is the current build working for you?

predacon, 2017-06-16:
yes, its working good. tested fan control and overclocking. this last dont work so well but works :)
only one problem with actual build: segfault when changing between cards. Love your software sr. you're doing so well!

marazmista, 2017-06-16:
Thanks for the tests. The driver confusion is explained pretty well here: https://www.reddit.com/r/linuxmasterrace/comments/4j2qtu/can_someone_quickly_explain_the_amd_open_drivers/d336nqx/
Basically, `radeon` and `amdgpu` are kernel drivers (you can see the name of module used in `uevent` files, first line DRIVER=?) and `radeonsi` is opengl implementation for radeons in mesa. I will show kernel module name somewhere on the UI in later builds

You see the clocks in current stable build because daemon read `amdgpu_pm_file` from debugfs and provide this info to the UI, but for kernel 4.12, it is reported, that this behaviour cause a shuttering (https://github.com/marazmista/radeon-profile/issues/54). I have a 'checkbox to force read of the info in old way' on my todo list, just haven't managed to implement it before my travels.

predacon, 2017-06-17:
thankyou for the explanation. if you need test something on radeon-profiler, i will be happy to help.

marazmista, 2017-07-03:
Fixed by merge of `rp-dev`: ac594203b2b89001304d73c38bb3c87a4adbd4c8

closed 2017-07-03 by marazmista