marazmista/radeon-profile #52: auto run isn't possible

szymag, 2017-05-19:
Recently, I've switched kernel to 4.11.1 and something wrong happen to radeon-profile. When system starts, radeon-profile also, but I couldn't read memory and GPU clock. Demon can't work without root privilege, so I need to run program manually:
```bash
sudo radeon-profile
``` 
I'm working on Linux Mint 18.1

marazmista, 2017-05-20:
Do you start daemon as systemd service or in some other way? After boot you can check if it started:
`systemctl status radeon-profile-daemon.service`

szymag, 2017-05-21:
Before upgrading kernel, I added `radeon-profile` to `startup application` and everything was OK. 
I've run your command and I got correct output, same as running `radeon-profile` with `sudo`.