marazmista/radeon-profile #53: radeon-profile appears to open up unlimited X server or Xhost connections causing major issues.

jarrard, 2017-05-24:
I have been testing radeon-profile with tumbleweed for a while now, I got the daemon and main application fully working by just making sure the correct files were write accessible etc.

However I noticed that radeon-profile is eating all the x server connections up which results in all other applications you attempt to start fail due to unable to connect to x server related issues.

I'm not complete sure how to fix this or why its happening. I believe I can get you the error that is typical when attempting to launch anything from terminal, but I don't know howto monitor a processes connections to a X server.

Note when radeon-profile is closed all programs can now be launched successfully without issue.

jarrard, 2017-05-25:
Here is the error messages all programs that use X produce.

Maximum number of clients reachedQXcbConnection: Could not connect to display :0

If I close radeon-profile down, everything works fine again.

marazmista, 2017-05-25:
I've searched a bit and found this topic: https://bbs.archlinux.org/viewtopic.php?id=188052. You can try execute `lsof +c0 /usr/lib/libX11.so | cut -d' ' -f1` or `lsof -p $(pidof Xorg)`

Which desktop environment do you use? What is the X server version?

jarrard, 2017-05-26:
I'm running 1.19.3 with KDE Plasma 5 (Tumbleweed). I will try your suggestion, it just seems like the issue shouldn't exist, and if it connects unlimited amount of times surely it will kill the x server in due time?

jarrard, 2017-05-27:
lsof +c0 /usr/lib64/libX11.so | cut -d' ' -f1   

Didn't seem to have an affect unfortunately, radeon-profile still within 15minutes locks up xorg server to the point where nothing else opens (just quest due to x window issues).  Closing Radeon-profile totally fixes the issue so its definitely isolated to this program unfortunately.

marazmista, 2017-05-27:
What was the output?

jarrard, 2017-05-27:
It basically listed all the processes running, and radeon-profile was on that process list so assumed it did what it was meant to do? Was I meant to run it only when the problem occurs?

jarrard, 2017-05-27:
Output at present, issues haven't arisen just yet.

theriddick@linux-84u0:~> lsof +c0 /usr/lib64/libX11.so | cut -d' ' -f1
COMMAND
firefox
kmozillahelper
radeon-profile
kdeinit5
klauncher
kded5
kaccess
ksmserver
kglobalaccel5
kwin_x11
baloo_file
kdeconnectd
krunner
plasmashell
xembedsniproxy
kscreen_backend
pulseaudio
docky
python2
skypeforlinux
org_kde_powerde
kactivitymanage
skypeforlinux
at-spi-bus-laun
at-spi2-registr
kuiserver5
skypeforlinux
ts3client_linux
goa-daemon
theriddick@linux-84u0:~>

jarrard, 2017-05-28:
I've since moved to git20170110 and turned off the daemon refresh data without request (daemon) option and it appears stable. I believe I was using a older version before from 2016.

I use the tumbleweed repo (https://software.opensuse.org/package/radeon-profile) because when I compile my own version is doesn't launch for some reason, not sure what the problem is but it appears to build it correctly. Hopefully someone uploads a newer package to opensuse sometime in future.

jarrard, 2017-07-11:
This issue I believe still happens if I run the application as root. However I have not tested it with the latest update from dev branch because of segmentation fault issue.