marazmista/radeon-profile #54: Don't read amdgpu_pm_info anymore

ChristophHaag, 2017-05-25:
With linux 4.12 reading this file causes stuttering in 3D apps.

See: https://bugs.freedesktop.org/show_bug.cgi?id=101182

lhorace, 2017-05-27:
If and when this pull request get's merge https://github.com/marazmista/radeon-profile/pull/44. It will likely solve that issue.

Danysan1, 2017-05-27:
Actually it would not solve completely the problem (in that PR ioctls are still used together with data from amdgpu_pm_info), but it would be a step toward the solution.

ChristophHaag, 2017-05-27:
Another quick fix would be reading the specific relevant files like for the frequencies
/sys/class/drm/card0/device/pp_dpm_mclk
and
/sys/class/drm/card0/device/pp_dpm_sclk
for GPUs that have powerplay support.

marazmista, 2017-05-27:
@lhorace yes, I've merged that into a dev branch and I'm working on an update, although some stuff is not available in ioctls, like volts, but I guess this was removed info from `amdgpu` module debug info anyway, and may stay as an option when someone use `radeon` module.

lhorace, 2017-05-27:
Great, it's been a very useful software here. Hopefully, that could fix some stability issues that I experience, that don't bother me so much. 

Additionally, I don't particular care about volts, however, how about asking amdgpu developers to expose it somehow? I am certainly sure that plenty of other users would find that information useful.

marazmista, 2017-06-13:
I pushed update to rp-dev branch. For `amdgpu` clocks info, Kernel 4.12 is a minimum because ioctls for this information are not available in earlier releases. It is still work in progress (for example, I'm having problems getting info about gpu usage on `amdgpu` but it is ok when I use `radeon`), and I want to add couple of things, but in overall, stuff should work.

If you use daemon, install updated version. Qt4 won't work and also qt5-charts package is needed.

Danysan1, 2017-06-14:
I tried compiling it but it did not work. It compiles only with the changes in https://github.com/marazmista/radeon-profile/compare/rp-dev...Danysan1:rp-dev

Danysan1, 2017-06-14:
Also, pwm speed is shown even if it is not available. This should fix it: https://github.com/Danysan1/radeon-profile/compare/rp-dev...rp-dev-2

marazmista, 2017-07-03:
Ioctls are now primary data source with optional debugfs, merged ac594203b2b89001304d73c38bb3c87a4adbd4c8

closed 2017-07-03 by marazmista