marazmista/radeon-profile #57: Optionally read the GPU load from amdgpu_pm_info

higuita, 2017-06-28:
amdgpu_pm_info have the GPU load, it would be good if one could read and graph that info too.

in another ticket, it is referred that it cause small delays, but to help debug performance issued, it is useful, so it should be disabled by default, and user get a warning if enabled.

Thanks for the tool!

marazmista, 2017-06-28:
Already done in `rp-dev` branch https://github.com/marazmista/radeon-profile/tree/rp-dev

marazmista, 2017-07-03:
Merged: ac594203b2b89001304d73c38bb3c87a4adbd4c8

closed 2017-07-03 by marazmista

higuita, 2017-07-14:
I tried the latest version and my GPU still reports 0% load, yet the amdgpu_pm_info reports load without any problem

marazmista, 2017-07-14:
Do you get `Invalid argument` messages in output when you run it in terminal?

higuita, 2017-07-14:
No, this is my output:

```
# radeon-profile
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
```
the GPU always show 0
when closing, this extra message show up:
```
context mis-match in pipe_sampler_view_release()
```

running with radeon-profile-daemon and running as a user, i get the same result

higuita, 2017-08-02:
just to update: i updated to the latest version and still the same problem, no gpu load show up

Scoopta, 2017-08-05:
Using the latest version with an R9 Fury running mesa and amdgpu and I can confirm that I also have 0% load.

marazmista, 2017-08-05:
What is your kernel version?

Scoopta, 2017-08-05:
4.12.4

lhorace, 2017-08-25:
@higuita Was this issue resolved for you? It's been closed without any indication is working for you?

higuita, 2017-09-07:
Sorry, no, this is not solved

using the latest radeon-profile, both with and without radeon-profile-daemon, on a update system:

OpenGL renderer string: AMD Radeon (TM) RX 480 Graphics (AMD POLARIS10 / DRM 3.18.0 / 4.13.0-slack, LLVM 4.0.1)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 17.2.0
my libdrm is 2.4.83

Enigma0, 2017-10-02:
Same issue:

[image]