marazmista/radeon-profile #58: Also add the "manual" performance level and manual frequency switch

higuita, 2017-06-28:
we have auto, low, high, but there is one extra level, the "manual"

With this option, one can manually switch frequency for the cart using pp_dpm_sclk and maybe pp_dpm_mclk (i'm unable to change this, but maybe this also depends of cards/firmware). This frequency change could then also be added to radeon-profile, as a drop down menu on the already showing mclk and sclck entries

This way we can manually set a frequency if needed, to improve performance on certain apps, or simply reduce the temperature on games that have already "enough" fps or the card gets too hot

marazmista, 2017-06-28:
Already partially implemented in `rp-dev` branch.

higuita, 2017-06-29:
oops, didn't know about it!
sorry for the spam then, i will try the rp-dev branch

jarrard, 2017-06-29:
I tried the rip-dev branch and compiled with qt5.9 (installed via run file from website and env set) and it resulted in a segfault possibly related to Qt Query or something when executing the program. Oddly the Master version compiles and runs fine.

higuita, 2017-06-29:
also have the same problem, also qt 5.9, slackware-current 64bit:
```
$ sudo /usr/bin/radeon-profile 
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QMetaObject::connectSlotsByName: No matching signal for on_cb_gpuData_clicked(bool)
Segmentation fault
```
the gdb output is this:
```

Thread 1 "radeon-profile" received signal SIGSEGV, Segmentation fault.
0x00007ffff5ac4c00 in QTimer::setInterval(int) () from /usr/lib64/libQt5Core.so.5
(gdb) bt 
#0  0x00007ffff5ac4c00 in QTimer::setInterval(int) () from /usr/lib64/libQt5Core.so.5
#1  0x00000000004745f3 in radeon_profile::qt_metacall(QMetaObject::Call, int, void**) ()
#2  0x00007ffff5ab883e in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib64/libQt5Core.so.5
#3  0x00007ffff6d3f890 in QDoubleSpinBox::valueChanged(double) () from /usr/lib64/libQt5Widgets.so.5
#4  0x00007ffff6d3fa72 in ?? () from /usr/lib64/libQt5Widgets.so.5
#5  0x00007ffff6cb631a in ?? () from /usr/lib64/libQt5Widgets.so.5
#6  0x00007ffff6d4119f in QDoubleSpinBox::setValue(double) () from /usr/lib64/libQt5Widgets.so.5
#7  0x0000000000455814 in radeon_profile::loadConfig() ()
#8  0x0000000000419b91 in radeon_profile::radeon_profile(QWidget*) ()
#9  0x00000000004153f2 in main ()
```

Running without sudo it starts without any problem, but as i do not have (yet) the daemon running, it is not doing anything.

Starting the daemon, it can can control the auto/low/high and the battery/balanced/performance and fan speed, but i do not see any GPU frequency (sclk nor mclk)

jarrard, 2017-06-29:
When I running the original master version with sudo (since dev one doesn't get past Q errors) it causes PID lockup after 5mins or so which basically means that no more processes can run until radeon-profile is exited. Running with daemon instead appears to resolve the issue however, from what I can see.

Just keep in mind you need to edit the daemon file to point to the correct radeon-profile program as it defaults to the typical etc/bin locations.

higuita, 2017-06-30:
> Just keep in mind you need to edit the daemon file to point to the correct radeon-profile program location (were you compile and keep radeon-profile) as it defaults to the typical etc/bin locations.

Sorry, i didn't understand this part... change what?

jarrard, 2017-06-30:
when you compile radeon-profile-daemon open the file in the extra directory and have a good look at the paths to ensure they are correct.

higuita, 2017-06-30:
ohh, the systemd unit file! i use slackware, it do not use systemd, so i do not need to change that! :)
Thank you anyway

marazmista, 2017-07-03:
Manual frequency control merged to master: ac594203b2b89001304d73c38bb3c87a4adbd4c8

closed 2017-07-03 by marazmista

higuita, 2017-07-13:
thanks, it works!