marazmista/radeon-profile #63: segmentation fault

jarrard, 2017-07-11:
As mentioned before, now that you have moved from dev branch I can no longer run this application. 

Segmentation fault.

It compiles fine with Qt 5.9+charts with no errors that I can see, I'm using kernel 4.12 but this also happened on 4.11. The issue did not happen with the OLD build so is something newly introduced. 

no other error codes appear

marazmista, 2017-07-11:
Could you build it as debug and paste output?
```
git clone https://github.com/marazmista/radeon-profile.git &&
cd radeon-profile/radeon-profile
qmake CONFIG+=debug &&
make
```

jarrard, 2017-07-12:
Here is the output:

`
the_riddick@underverse:~/Downloads/radeon-profile/radeon-profile$ qmake CONFIG+=debug && >make
Info: creating stash file /home/the_riddick/Downloads/radeon-profile/radeon-profile/.qmake.stash
Project MESSAGE: Building for debug
/home/the_riddick/Desktop/the_riddick/Qt/5.9/gcc_64/bin/uic radeon_profile.ui -o ui_radeon_profile.h
/home/the_riddick/Desktop/the_riddick/Qt/5.9/gcc_64/bin/uic components/pieprogressbar.ui -o ui_pieprogressbar.h
/home/the_riddick/Desktop/the_riddick/Qt/5.9/gcc_64/bin/uic dialogs/dialog_defineplot.ui -o ui_dialog_defineplot.h
/home/the_riddick/Desktop/the_riddick/Qt/5.9/gcc_64/bin/uic dialogs/dialog_rpevent.ui -o ui_dialog_rpevent.h
/home/the_riddick/Desktop/the_riddick/Qt/5.9/gcc_64/bin/uic dialogs/dialog_topbarcfg.ui -o ui_dialog_topbarcfg.h
/home/the_riddick/Desktop/the_riddick/Qt/5.9/gcc_64/bin/uic dialogs/dialog_deinetopbaritem.ui -o ui_dialog_deinetopbaritem.h
g++ -c -pipe -std=c++11 -Wall -Wextra -Wpedantic -g -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_CHARTS_LIB -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtCharts -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtPrintSupport -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtWidgets -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtGui -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtNetwork -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtCore -I. -isystem /usr/include/libdrm -I. -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/mkspecs/linux-g++ -o main.o main.cpp                                                                                                                                                                          
g++ -c -pipe -std=c++11 -Wall -Wextra -Wpedantic -g -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_CHARTS_LIB -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtCharts -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtPrintSupport -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtWidgets -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtGui -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtNetwork -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtCore -I. -isystem /usr/include/libdrm -I. -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/mkspecs/linux-g++ -o radeon_profile.o radeon_profile.cpp                                                                                                                                                      
g++ -c -pipe -std=c++11 -Wall -Wextra -Wpedantic -g -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_CHARTS_LIB -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtCharts -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtPrintSupport -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtWidgets -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtGui -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtNetwork -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtCore -I. -isystem /usr/include/libdrm -I. -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/mkspecs/linux-g++ -o uiElements.o uiElements.cpp                                                                                                                                                              
g++ -c -pipe -std=c++11 -Wall -Wextra -Wpedantic -g -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_CHARTS_LIB -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtCharts -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtPrintSupport -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtWidgets -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtGui -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtNetwork -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtCore -I. -isystem /usr/include/libdrm -I. -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/mkspecs/linux-g++ -o uiEvents.o uiEvents.cpp                                                                                                                                                                  
g++ -c -pipe -std=c++11 -Wall -Wextra -Wpedantic -g -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_CHARTS_LIB -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtCharts -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtPrintSupport -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtWidgets -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtGui -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtNetwork -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/include/QtCore -I. -isystem /usr/include/libdrm -I. -I../../../Desktop/the_riddick/Qt/5.9/gcc_64/mkspecs/linux-g++ -o gpu.o gpu.cpp                                                                                                                                                                            
[... cut ...] topbaritem.o qrc_radeon-resource.o moc_radeon_profile.o moc_gpu.o moc_daemonComm.o moc_execbin.o moc_rpplot.o moc_pieprogressbar.o moc_dialog_defineplot.o moc_dialog_rpevent.o moc_dialog_topbarcfg.o moc_dialog_deinetopbaritem.o   -lXrandr -lX11 -L/home/the_riddick/Desktop/the_riddick/Qt/5.9/gcc_64/lib -lQt5Charts -L/home/the_riddick/Qt/5.9/gcc_64/lib -lQt5PrintSupport -lQt5Widgets -lQt5Gui -lQt5Network -lQt5Core -lGL -lpthread 
the_riddick@underverse:~/Downloads/radeon-profile/radeon-profile$ 
`

marazmista, 2017-07-12:
What is the output when you try to run radeon-profile compiled with debug flag?

jarrard, 2017-07-12:
Segmentation fault:

Originally when I compiled and ran the rp-dev branch before you moved it to mainline it did produce some other Qt error but I didn't save that snap shot.

marazmista, 2017-07-14:
I pushed some updates, can you build latest master as debug and post output when you try to run it?

jarrard, 2017-07-14:
Sure, it may take me a while as I'm still sorting out the kinks in my new computer system I built late last night. _I sort of borked my Linux install trying to convert a btrfs to ext4 so my QT install is probably not setup right._

dhardy, 2018-02-12:
I also get segfaults with Qt 5.9 (Fedora 27):
```
15:56 dhardy@localhost:~/other/radeon-profile/radeon-profile$ ./radeon-profile 
Segmentation fault (core dumped)
15:57 dhardy@localhost:~/other/radeon-profile/radeon-profile$ valgrind ./radeon-profile 
==7570== Memcheck, a memory error detector
==7570== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==7570== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==7570== Command: ./radeon-profile
==7570== 
==7570== 
==7570== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==7570==  Bad permissions for mapped region at address 0x19B63850
==7570==    at 0x19B63850: ??? (in /usr/lib64/qt5/plugins/platforminputcontexts/libcomposeplatforminputcontextplugin.so)
==7570==    by 0xE7A4AFF: ???
==7570==    by 0xE7A459F: ???
==7570== 
==7570== HEAP SUMMARY:
==7570==     in use at exit: 536,232 bytes in 4,305 blocks
==7570==   total heap usage: 15,008 allocs, 10,703 frees, 1,781,954 bytes allocated
==7570== 
==7570== LEAK SUMMARY:
==7570==    definitely lost: 0 bytes in 0 blocks
==7570==    indirectly lost: 0 bytes in 0 blocks
==7570==      possibly lost: 896 bytes in 4 blocks
==7570==    still reachable: 535,336 bytes in 4,301 blocks
==7570==                       of which reachable via heuristic:
==7570==                         newarray           : 128 bytes in 4 blocks
==7570==         suppressed: 0 bytes in 0 blocks
==7570== Rerun with --leak-check=full to see details of leaked memory
==7570== 
==7570== For counts of detected and suppressed errors, rerun with: -v
==7570== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
Segmentation fault (core dumped)
```
That was built in debug mode. There were two compiler warnings about implicit fall-through but probably unrelated.

closed 2019-02-24 by marazmista