marazmista/radeon-profile #64: gpu.h:12:37: fatal error: QtConcurrent/QtConcurrent: No such file or directory

Enigma0, 2017-07-18:
```
g++ -c -m64 -pipe -std=c++11 -O2 -Wall -W -D_REENTRANT -DQT_NO_DEBUG_OUTPUT -DQT_NO_DEBUG -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4 -I. -I. -o main.o main.cpp
In file included from radeon_profile.h:4:0,
                 from main.cpp:1:
gpu.h:12:37: fatal error: QtConcurrent/QtConcurrent: No such file or directory
 #include <QtConcurrent/QtConcurrent>
                                     ^
compilation terminated.
Makefile:383: recipe for target 'main.o' failed
make: *** [main.o] Error 1
```

Danysan1, 2017-07-19:
Radeon Profile does not support Qt 4 anymore since https://github.com/marazmista/radeon-profile/commit/ac594203b2b89001304d73c38bb3c87a4adbd4c8 .
The last version that supports Qt 4 is [a77dddc](https://github.com/marazmista/radeon-profile/tree/a77dddc959b25cef82b9c83f65ffc376a851fb9b).

Enigma0, 2017-07-21:
@Danysan1 Alright - how do I build using Qt5 (presumably)?

Enigma0, 2017-07-21:
Had to reinstall `qt5-default` as per : https://stackoverflow.com/a/39742962/1666510

Enigma0, 2017-07-21:
Still seeing same error above despite:

```
qmake --version
QMake version 3.0
Using Qt version 5.7.1 in /usr/lib/x86_64-linux-gnu
```

```
:~/radeon-profile/radeon-profile$ qmake
Project MESSAGE: Building for release
Project ERROR: Unknown module(s) in QT: charts
:~/radeon-profile/radeon-profile$ make
/usr/lib/x86_64-linux-gnu/qt4/bin/uic radeon_profile.ui -o ui_radeon_profile.h
/usr/lib/x86_64-linux-gnu/qt4/bin/uic components/pieprogressbar.ui -o ui_pieprogressbar.h
/usr/lib/x86_64-linux-gnu/qt4/bin/uic dialogs/dialog_defineplot.ui -o ui_dialog_defineplot.h
/usr/lib/x86_64-linux-gnu/qt4/bin/uic dialogs/dialog_rpevent.ui -o ui_dialog_rpevent.h
/usr/lib/x86_64-linux-gnu/qt4/bin/uic dialogs/dialog_topbarcfg.ui -o ui_dialog_topbarcfg.h
/usr/lib/x86_64-linux-gnu/qt4/bin/uic dialogs/dialog_deinetopbaritem.ui -o ui_dialog_deinetopbaritem.h
g++ -c -m64 -pipe -std=c++11 -O2 -Wall -W -D_REENTRANT -DQT_NO_DEBUG_OUTPUT -DQT_NO_DEBUG -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4 -I. -I. -o main.o main.cpp
In file included from radeon_profile.h:4:0,
                 from main.cpp:1:
gpu.h:12:37: fatal error: QtConcurrent/QtConcurrent: No such file or directory
 #include <QtConcurrent/QtConcurrent>
                                     ^
compilation terminated.
Makefile:383: recipe for target 'main.o' failed
make: *** [main.o] Error 1
```

Enigma0, 2017-08-05:
Essentially a duplicate of this: https://github.com/marazmista/radeon-profile/issues/66
Now resolved with pull request: https://github.com/marazmista/radeon-profile/pull/65

closed 2017-08-05 by Enigma0