marazmista/radeon-profile #66: Project ERROR: Unknown module(s) in QT: charts (Ubuntu 17.04)

Enigma0, 2017-07-21:
Ubuntu 17.04 seems to not have `libqt5charts5*.deb`

README should probably mention:

* http://launchpadlibrarian.net/316647367/libqt5charts5_5.7.1-3_amd64.deb
* http://launchpadlibrarian.net/316647362/libqt5charts5-dev_5.7.1-3_amd64.deb

marazmista, 2017-07-21:
Looks like this may be related: https://bugs.launchpad.net/ubuntu/+source/qtcharts-opensource-src/+bug/1682643
Can you try install qtcharts-opensource-src as noted in link? Maybe this will help.

Enigma0, 2017-07-21:
@marazmista I did see that. That's where I found the .debs that seemed relevant. Not sure how to install that as a whole - it looks like it just has a lot of .deb's

I also just installed http://launchpadlibrarian.net/316647368/qml-module-qtcharts_5.7.1-3_amd64.deb but still same error.

As per https://launchpad.net/ubuntu/+source/qtcharts-opensource-src/5.7.1-3/+build/12462945 the only packages I didn't install are the -dbgsym, docs & example ones.

marazmista, 2017-07-24:
Can you check if you have QtCharts directory in `/usr/include/qt`?

Enigma0, 2017-07-24:
@marazmista Not entirely sure what's going on here but this is all I have:

```
:/usr/include$ ls -l | grep qt
drwxr-xr-x 19 root root   4096 Jul 13 12:43 qt4
drwxr-xr-x  3 root root   4096 Jul 18 20:05 qt4qevercloud
drwxr-xr-x  2 root root   4096 Jul 18 19:50 qt5keychain
drwxr-xr-x  3 root root   4096 Jul 18 19:50 qt5qevercloud
```

Enigma0, 2017-08-05:
@marazmista I reinstalled GNOME 17.04 and I have nothing in `/usr/include`.

What am I supposed to install?

I have this but how do I use it? Some instructions in the README would be useful.

```
:/opt/Qt5.9.1/5.9.1/gcc_64/include$ ls -l | grep Chart
drwxrwxr-x 3 root root  4096 Jul 31 01:23 QtCharts
```

Enigma0, 2017-08-05:
Issue is resolved for me following the updated README in pull request: https://github.com/marazmista/radeon-profile/pull/65#pullrequestreview-54519918

MasterCATZ, 2017-12-17:
Ubuntu 17.10

sudo apt install libqt5charts5 libqt5charts5-dev

worked for me 

libqt5charts5 is already the newest version (5.9.1-3).
libqt5charts5-dev is already the newest version (5.9.1-3).

Enigma0, 2017-12-17:
That's awesome. Must have just been an oversight with `17.04`.

closed 2017-12-17 by Enigma0

MishkaRogachev, 2018-06-26:
sudo apt install qml-module-qtcharts