marazmista/radeon-profile #67: Program seems outdated? Doesn't match screenshots - says daemon needs to be running even though it is

Enigma0, 2017-07-21:
Am I running the latest version? Why doesn't it match the screenshots?

[image]

Also says it can't read data as daemon needs to be running but it is:

```
:~$ sudo service radeon-profile-daemon status
● radeon-profile-daemon.service - radeon-profile daemon
   Loaded: loaded (/etc/systemd/system/radeon-profile-daemon.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2017-07-21 10:46:51 EDT; 6min ago
 Main PID: 1191 (radeon-profile-)
    Tasks: 2 (limit: 4915)
   Memory: 2.0M
      CPU: 96ms
   CGroup: /system.slice/radeon-profile-daemon.service
           └─1191 /usr/bin/radeon-profile-daemon

Jul 21 10:53:08 radeon-profile-daemon[1191]: Shared memory is not attached, can't write data:  "QSharedMemory::handle: key is empty"
Jul 21 10:53:09 radeon-profile-daemon[1191]: Shared memory is not attached, can't write data:  "QSharedMemory::handle: key is empty"
Jul 21 10:53:10 radeon-profile-daemon[1191]: Shared memory is not attached, can't write data:  "QSharedMemory::handle: key is empty"
Jul 21 10:53:11 radeon-profile-daemon[1191]: Shared memory is not attached, can't write data:  "QSharedMemory::handle: key is empty"
Jul 21 10:53:12 radeon-profile-daemon[1191]: Shared memory is not attached, can't write data:  "QSharedMemory::handle: key is empty"
Jul 21 10:53:13 radeon-profile-daemon[1191]: Shared memory is not attached, can't write data:  "QSharedMemory::handle: key is empty"
Jul 21 10:53:14 radeon-profile-daemon[1191]: Shared memory is not attached, can't write data:  "QSharedMemory::handle: key is empty"
Jul 21 10:53:15 radeon-profile-daemon[1191]: Shared memory is not attached, can't write data:  "QSharedMemory::handle: key is empty"
Jul 21 10:53:16 radeon-profile-daemon[1191]: Shared memory is not attached, can't write data:  "QSharedMemory::handle: key is empty"
Jul 21 10:53:17 radeon-profile-daemon[1191]: Shared memory is not attached, can't write data:  "QSharedMemory::handle: key is empty"
```

I also tried setting up the PPA and installing the packages from there but it seems the same.

marazmista, 2017-07-21:
Version is in the bottom right corner of the window.
Latest can be checked here: https://github.com/marazmista/radeon-profile/releases

Enigma0, 2017-07-21:
Alright so that confirms I am definitely outdated. What's odd is that I installed the packages via the PPA also and I am still seeing the same package version...

Presumably if I did manage to build it (I have some other issues open preventing me) I'd get to that version - OK.

closed 2017-07-21 by Enigma0

muzena, 2017-08-05:
@Enigma0 
From where you install radeon-profile and, which version of Linux do you have?

Enigma0, 2017-08-05:
@muzena From latest git but I am unable to build because qt5-charts is missing in 17.04: https://github.com/marazmista/radeon-profile/issues/66

muzena, 2017-08-05:
@Enigma0 
Because qt5-charts is missing it's only possible to make ppa for [Ubuntu 17.10 artful](https://launchpad.net/~trebelnik-stefina/+archive/ubuntu/radeon-profile/+sourcepub/8104429/+listing-archive-extra)

Enigma0, 2017-08-06:
Fair enough.

muzena, 2017-08-19:
@Enigma0 
Try now repository, I add qt5-charts dependencies.

Enigma0, 2017-08-19:
@muzena I'm all set building from source for now but if/when I get a new installation I'll give that a try.

This was the pull request that allowed me to build from source: https://github.com/marazmista/radeon-profile/pull/65

Thanks for the effort.