marazmista/radeon-profile #68: Translation not found. segmentation fault

predacon, 2017-07-24:
build radeon-profile-git-20170714.r0.g4e02ea6-1 from AUR.
No problem reported on installation but when I try to open it, on the console report this error: Translation not found. Segmentation fault
Running on 4.11.9-1-ARCH with AMDGPU open source driver.

Can I help with some more info or outputs?

predacon, 2017-07-24:
update: radeon-profile only runs fine if its executed with root privileges. Even with radeon-profile-daemon running, error described before continues happening

marazmista, 2017-07-24:
First, you can try delete configuration: `.radeon-profile-auxstuff` and `.radeon-profile-settings` in home directory and run it as regular user.
If that doesn't help, could you build it as debug and paste output when you try to run it? Script for build debug: 
```
git clone https://github.com/marazmista/radeon-profile.git &&
cd radeon-profile/radeon-profile &&
qmake CONFIG+=debug &&
make
```

predacon, 2017-07-25:
Problem solved deleting config files in .radeon-profile-auxstuff and .radeon-profile-settings.
Thankyou so much.

closed 2017-07-25 by predacon