marazmista/radeon-profile #70: GPU usage always 0%

pschichtel, 2017-08-18:
I have an RX 460 graphics card in my system. I'm on Arch with Kernel 4.12.8 and the fully open source driver stack (mesa 17.1.6).
Frequencies, memory usage, temperatures, fan speeds, ... everything is displayed correctly (and with reasonable values), except the GPU usage value. This values is always on 0.

SethBarberee, 2017-08-18:
Same problem with my RX 470 on Arch linux 4.12.7-2-ck-skylake

pschichtel, 2017-08-18:
Probably amdgpu related I guess.

pschichtel, 2017-08-19:
Radeontop seems to work fine.

blisbell, 2017-08-23:
I have an rx 480 an mine is doing the same i also have an r5 220 to test with it does show the gpu usage but it uses the radeon driver not the amdgpu. it is really a 5450 rebrand.

lhorace, 2017-08-25:
Seem to be a duplicate of https://github.com/marazmista/radeon-profile/issues/57.

pschichtel, 2017-08-27:
I'm on Kernel 4.12+ (```Linux phillip-desktop 4.12.8-2-ARCH #1 SMP PREEMPT Fri Aug 18 14:08:02 UTC 2017 x86_64 GNU/Linux```) and on latest maser of radeon-profile and radeon-profile-daemon (with the daemon running), but there is still no GPU usage.

lhorace, 2017-08-28:
IIRC, he made no mention about GPU usage. Just to be clear, he was talking about GPU and Memory clock on AMDGPU driver in regards to Kernel 4.12+.

marazmista, 2017-08-30:
What `libdrm` version do you have?

pschichtel, 2017-08-31:
@marazmista 2.4.83

blisbell, 2017-08-31:
I am using mesa-git repo for arch libdrm-git  6141.f19dbb79-1

higuita, 2017-09-07:
as commended in #57 

OpenGL renderer string: AMD Radeon (TM) RX 480 Graphics (AMD POLARIS10 / DRM 3.18.0 / 4.13.0-slack, LLVM 4.0.1)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 17.2.0

libdrm version is 2.4.83

Enigma0, 2017-10-02:
Same issue:

[image]

arigit, 2018-01-14:
Same issue with RX-560
Fedora 27.

OpenGL renderer string: Radeon RX 560 Series (AMD POLARIS11 / DRM 3.19.0 / 4.14.11-300.fc27.x86_64, LLVM 4.0.1)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 17.2.4

Name         : libdrm
Version      : 2.4.89
Release      : 1.fc27
Arch         : i686
Size         : 372 k
Source       : libdrm-2.4.89-1.fc27.src.rpm

higuita, 2018-03-24:
As the other bug is close, lets use this one...
what can we do to help solve this? :)

rigred, 2018-05-08:
Had a look at the code. 
The reason is awful hacky code. Both in the linux dri files and here for radeon-profile.
So radeon-profile is using regex to parse the stats out of `/sys/kernel/debug/dri/0/amdgpu_pm_info`.

[image]

Which is a really horrible solution. We actually need to be reading GPU registers like radeontop is doing.

https://github.com/clbr/radeontop

I've made a fork where I've started patching a few oddities, but I've still got to work myself into the code

https://github.com/rigred/radeon-profile/

rigred, 2018-05-08:
Also the kernel developers seem to be doing some really dumb things with the amdgpu_pm_info output with VDDC and VDDCI voltages being reported as Watts? What is this nonsense?

[image]

rigred, 2018-05-08:
Someone really wasn't thinking when writing the amdgpu_pm code

https://github.com/torvalds/linux/blob/86bbbebac1933e6e95e8234c4f7d220c5ddd38bc/drivers/gpu/drm/amd/amdgpu/amdgpu_pm.c

mainmachine, 2018-05-08:
Looks like they did a copypasta and forgot to change "W" to "V"... LOL :)

rigred, 2018-05-08:
No. It doesn't actually report voltage. I'm not sure what magic number it reports. Current perhaps? VDDC goes up to 87 and more under load. Like wattage would. But VDDC isn't supposed to read watts. It's commonly supposed to be mV with Current (Amps) as another entry and if multiplied that gives Watts. But this number is just way too low at idle. The VDDCI is also apparently always at 16? The debugfs output just looks like some quick and dirty number fudging without explanation.

blisbell, 2018-05-11:
I hope you get this cleaned up @rigred tried the new commits still no gpu usage with an rx 480. will sen any logs you might want just have to tell me what an how.been watching this program for a while it looks like the original creater  has abandoned it.

rigred, 2018-05-11:
Yeah I haven't yet patched the GPU usage part. Will be doing that this weekend. I've been I'll recently and didn't get to doing this.

marazmista, 2018-05-11:
radeon-profile does use ioctl to read GPU data, so unless you didn't check 'Use daemon as data provider' in settings, the regex code you've mentioned is not used and no info is passed through IPC between daemon and radeon-profile about GPU data (and daemon is simple, because I don't want split logic into two programs).
And for kernels 4.12<= there is this: https://bugs.freedesktop.org/show_bug.cgi?id=101182 (and IMO because of this, there is no point in refactoring regex code).

What I've found when I investigated this bug some time ago, is that reading GPU usage using amdgpu iocts falis with error: `Invalid argument` thrown here: https://github.com/marazmista/radeon-profile/blob/38132804e94b4f1cdceda7ccef7bd78ff1b99278/radeon-profile/ioctl_amdgpu.cpp#L60

`AMDGPU_INFO_SENSOR_GPU_LOAD` is type of sensor, defined for a more general argument `AMDGPU_INFO_SENSOR` for ioctl to read GPU Usage, it is defined here:
https://github.com/torvalds/linux/blob/86bbbebac1933e6e95e8234c4f7d220c5ddd38bc/include/uapi/drm/amdgpu_drm.h#L662

Function to read ioctl data from `AMDGPU_INFO_SENSOR` in radeon-profile is here: https://github.com/marazmista/radeon-profile/blob/38132804e94b4f1cdceda7ccef7bd78ff1b99278/radeon-profile/ioctl_amdgpu.cpp#L16
and it works fine. I tested it for all values that can be passed as sensor type and it works for: `AMDGPU_INFO_SENSOR_GFX_SCLK` , `AMDGPU_INFO_SENSOR_GFX_MCLK`, `AMDGPU_INFO_SENSOR_GPU_TEMP` and fails for `AMDGPU_INFO_SENSOR_GPU_LOAD` and everything that comes after (values`0x4` till end). 

In kernel, the ioctl request for GPU usage info is handled here: https://github.com/torvalds/linux/blob/94d7dbf108813ea45a91e27e9a8bd231d5a23fa7/drivers/gpu/drm/amd/amdgpu/amdgpu_kms.c#L737
Maybe there is a bug in kernel, and it work when value is put in `amdgput_pm_info` but it somehow doesn't when value is returned by ioctl. If so, I didn't find it.

rigred, 2018-05-11:
Yes I've just looked at more of the code and found the ioctl logic.
I'm currently testing the ioctl code and everything currently only returns -1 from the kernel so something is going wrong.
Kernel 4.16.6 btw.

rigred, 2018-05-11:
I've gone over the values in the pointers from the result of `AMDGPU_INFO_SENSOR_GPU_LOAD` query and it's returning some strange value. 

All other queries return normal int's except for GPU LOAD which gives -1.

First loop:
[image]

Second Loop:
[image]

In my case it also very often seems to for no apparent reason fallback to using the PM info file instead of ioctl.

marazmista, 2018-05-11:
Yep, same results as mine.

For temperature, yes, hwmon is used.
I think all the `get..()` methods go through `gpu` class (it is more or less adapter pattern).

rigred, 2018-05-11:
A possible workaround is to use something like this via the xf86drm `drmCommandWrite()` function, which surprisingly works:

```C
#include <xf86drm.h>
```
-------------
```C
int get_gpu_reg(int fd, int reg, uint32_t* value)
{
	struct drm_amdgpu_info request;
	memset(&request, 0, sizeof request);
	request.return_pointer = (uint64_t)value;
	request.return_size = sizeof *value;
	request.query = AMDGPU_INFO_SENSOR;
	request.sensor_info.type = reg;

	return drmCommandWrite(fd, DRM_AMDGPU_INFO, &request, sizeof request);
}

int get_gpu_load(int fd, uint32_t* value)
{
	return get_gpu_reg(fd, AMDGPU_INFO_SENSOR_GPU_LOAD, value);
}
```

rigred, 2018-05-11:
Actually this should work for just about all the same values. There's a whole duplicated drm API almost

```
// Shader core clock speed in Mhz
int get_gpu_sclk(int fd, uint32_t* value)
{
	return get_gpu_reg(fd, AMDGPU_INFO_SENSOR_GFX_SCLK, value);
}

// Memory clock speed in MHz
int get_gpu_mclk(int fd, uint32_t* value)
{
	return get_gpu_reg(fd, AMDGPU_INFO_SENSOR_GFX_MCLK, value);
}

// GPU temperature in millidegrees C
int get_gpu_temp(int fd, uint32_t* value)
{
	return get_gpu_reg(fd, AMDGPU_INFO_SENSOR_GPU_TEMP, value);
}

// GPU load measurement between 0 and 100
// Fluctuates like crazy and needs to be polled rapidly for an accurate reading
int get_gpu_load(int fd, uint32_t* value)
{
	return get_gpu_reg(fd, AMDGPU_INFO_SENSOR_GPU_LOAD, value);
}

// GPU power draw in watts (W)
int get_gpu_power(int fd, uint32_t* value)
{
	return get_gpu_reg(fd, AMDGPU_INFO_SENSOR_GPU_AVG_POWER, value);
}

// GPU core voltage in millivolts (mV)
int get_gpu_voltage(int fd, uint32_t* value)
{
	return get_gpu_reg(fd, AMDGPU_INFO_SENSOR_VDDGFX, value);
}
```

marazmista, 2018-08-28:
So, I tested kernel with this patch and unfortunately, there is something wrong with `gpu_busy_percent` file on my machine:
```
cat /sys/class/drm/card0/device/gpu_busy_percent
cat: /sys/class/drm/card0/device/gpu_busy_percent: Invalid argument
```
 It shows as empty if checked differently than through `cat`. Could someone else check?

blisbell, 2018-08-28:
I have an vega 56 now all i get is 0 even when converting a video file using gpu

marazmista, 2018-09-02:
Ok, thanks, maybe it just doesn't work for my setup.
I pushed to rp-dev branch some initial implementation that read gpu usage from this file.

marazmista, 2018-12-13:
`gpu_busy_percent` sysfs's file, which is supposed to have info about GPU load, was implemented and merged into master, so all should work. On my system, this file exist but doesn't have any info, so it is difficult to debug if something is wrong.

pschichtel, 2018-12-14:
I'll try and check on my system (vega64) in the coming days

rigred, 2018-12-14:
I have checked on my RX580 and confirm that it is partially working.

For GPU compute workloads that max out the Core I can see 100% load
However for any finer workloads or 3d rendering the value is clearly not updated consistently.

Using the Gallium HUD overlay I can get very accurate fine GPU load measurements.
In one particular scenario (using my tesseract game fork) running at 60FPS with 28% GPU load (according to OpenGL Gallium)the GPU busy percent is 0 or jumping around to very different values around the 28%.

While idle I also get very different measurements between rocm-smi and gpu_busy_percent

The only conclusion I can make is that none of the tools are consistently in agreement with each other.

rigred, 2018-12-14:
For others:
```
/sys/class/drm/card0/device/gpu_busy_percent
```

X0rg, 2019-01-20:
I have a new graphic card (RX 580), and I can confirm this issue on Arch Linux, with Linux 4.20.3 and Mesa 18.3.1.

Some investigation (sorry for the debugging with `qWarning()`, the output was not displayable in GDB):
```
amdgpuIoctlHandler::getGpuUsage success = true
amdgpuIoctlHandler::getGpuUsage data = 0
dXorg::getGPUUsage() gpuUsage = 0
dXorg::getGPUUsage() gpu_busy_percent.isEmpty() = false
amdgpuIoctlHandler::getGpuUsage success = true
amdgpuIoctlHandler::getGpuUsage data = 4.2039e-45
dXorg::getGPUUsage() gpuUsage = 4.2039e-45
dXorg::getGPUUsage() gpu_busy_percent.isEmpty() = false
```
`4.2039e-45`, what? So the problem come from `amdgpuIoctlHandler::getGpuUsage()`.

According to [this line](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/gpu/drm/amd/amdgpu/amdgpu_kms.c#n854), `AMDGPU_INFO_SENSOR_GPU_LOAD` is a integer, not a float.

Here is a quick and dirty patch:
```
diff --git a/radeon-profile/ioctl_amdgpu.cpp b/radeon-profile/ioctl_amdgpu.cpp
index 7c82ef9..48555cd 100644
--- a/radeon-profile/ioctl_amdgpu.cpp
+++ b/radeon-profile/ioctl_amdgpu.cpp
@@ -57,7 +57,10 @@ bool amdgpuIoctlHandler::getValue(void *data, unsigned dataSize, unsigned comman
 
 bool amdgpuIoctlHandler::getGpuUsage(float *data) const {
 #ifdef AMDGPU_INFO_SENSOR_GPU_LOAD
-    return getSensorValue(data, sizeof(data), AMDGPU_INFO_SENSOR_GPU_LOAD);
+    int tmp = -1;
+    bool success = getSensorValue(&tmp, sizeof(tmp), AMDGPU_INFO_SENSOR_GPU_LOAD);
+    *data = success ? (float) tmp : -1.0;
+    return success;
 #else
     Q_UNUSED(data);
     return false;
```
And the new output is correct:
```
amdgpuIoctlHandler::getGpuUsage success = true
amdgpuIoctlHandler::getGpuUsage data = 45
dXorg::getGPUUsage() gpuUsage = 45
dXorg::getGPUUsage() gpu_busy_percent.isEmpty() = false
amdgpuIoctlHandler::getGpuUsage success = true
amdgpuIoctlHandler::getGpuUsage data = 49
dXorg::getGPUUsage() gpuUsage = 49
dXorg::getGPUUsage() gpu_busy_percent.isEmpty() = false
amdgpuIoctlHandler::getGpuUsage success = true
amdgpuIoctlHandler::getGpuUsage data = 46
dXorg::getGPUUsage() gpuUsage = 46
```

min-xu-ai, 2020-08-20:
I am having this problem on arch linux with kernel 5.7.x.

Thanks for your checkin with qDebug, but when I enabled debug log level, I see a bunch of output, but not the sensor number. So I hacked it and got some more info:

```
[owen@honor radeon-profile]$ gksu ./target/radeon-profile 
triggered PW
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QFSFileEngine::open: No file name specified
DRM_IOCTL_AMDGPU_INFO: Invalid argument
DRM_IOCTL_AMDGPU_INFO: Invalid argument
DRM_IOCTL_AMDGPU_INFO: Invalid argument
DRM_IOCTL_AMDGPU_INFO: Invalid argument
DRM_IOCTL_AMDGPU_INFO: Invalid argument
DRM_IOCTL_AMDGPU_INFO: Invalid argument
DRM_IOCTL_AMDGPU_INFO: Invalid argument
DRM_IOCTL_AMDGPU_INFO: Invalid argument
Requested sensor: 0X04 - GPU_LOADRequested sensor: 0X04 - GPU_LOADRequested sensor: 0X04 - GPU_LOADRequested sensor: 0X04 - GPU_LOADRequested sensor: 0X04 - GPU_LOADRequested sensor: 0X04 - GPU_LOADRequested sensor: 0X04 - GPU_LOADRequested sensor: 0X04 - GPU_LOAD[owen@honor radeon-profile]$ git diff
diff --git a/radeon-profile/ioctl_amdgpu.cpp b/radeon-profile/ioctl_amdgpu.cpp
index ae0e03a..69152c7 100644
--- a/radeon-profile/ioctl_amdgpu.cpp
+++ b/radeon-profile/ioctl_amdgpu.cpp
@@ -1,3 +1,5 @@
+#include <stdio.h>
+
 #include "ioctlHandler.h"
 
 #include <QDebug>
@@ -26,6 +28,7 @@ bool amdgpuIoctlHandler::getSensorValue(void *data, unsigned dataSize, unsigned
     if (Q_UNLIKELY(!success)) {
         sensor_name = getSensorName(sensor);
         qDebug("Requested sensor: %s", qUtf8Printable(sensor_name));
+        printf("Requested sensor: %s", qUtf8Printable(sensor_name));
         perror("DRM_IOCTL_AMDGPU_INFO");
     }
     return success;
```

What do you think things could go wrong for my setup? Thanks a lot for your help. BTW, I had to add "iommu=pt" to my kernel params to get the Ryzen GPU to work in the first place.

min-xu-ai, 2020-08-20:
Never mind. I found out why:

```
Aug 20 06:59:49 honor kernel: amdgpu: [powerplay] pp_dpm_get_sclk_od was not implemented.
Aug 20 06:59:49 honor kernel: amdgpu: [powerplay] pp_dpm_get_mclk_od was not implemented.
```