marazmista/radeon-profile #71: Power limit setting

tacchinotacchi, 2017-08-23:


lhorace, 2017-08-25:
Hi,

I can't see to find `power level`, where is it located?

marazmista, 2017-08-27:
Not sure if I understand, but what happens when you set specific clock in Wattman and then increase power limit and put some load on card? Frequencies go up and ignore your setting? 
If so, I guess power limit is not implemented in kernel. There is only PowerPlay table in `pp_dpm_sclk` and `pp_dpm_mclk` files that enable user to set frequency while `manual` is selected.

lhorace, 2017-08-28:
Well, if you increase the power limit, you are telling the card that it's okay to draw my juice to maintain that frequency. On windows, on RX 480, you can increase the power limit by 50%. 

> If so, I guess power limit is not implemented in kernel.

I surmise that's the case but I could swore I did this in the past. I can ask on Phoronix Forums about it.

tacchinotacchi, 2017-08-29:
It hasn't got much to do with the performance profile, rather with how much
power ( as in wattage ) the card is allowed to use. For many custom cards,
the retail power limit is so low that the card can't even keep up with the
retailer OC and throttles. That's my case, as I need to set it to +14% not
to have my 470 throttle down from 1216 MHz.

tacchinotacchi, 2017-08-29:
Dude, I'll never reply from my email again. Can't even edit for some reason

ruben2000de, 2018-09-15:
I would realy apreaceate a easy "gui-way" to set my V64 to +/-50% (W)