marazmista/radeon-profile #77: No GPU Usage Information

NSchepsen, 2017-10-01:
Hello,

I've compiled this app, but there is no gpu usage information. My config is:

Kubuntu 14.04
AMD 7950 using radeon driver
glxinfo - installed
xdriinfo - installed
xrandr - installed

[image]

Thanks

Enigma0, 2017-10-02:
Same issue here:

[image]

lhorace, 2017-10-02:
Duplicate of #70

NSchepsen, 2017-10-03:
No, it is not, I have no entry shows me the GPU usage, the other guy has it, but it shows 0,0 %.

lhorace, 2017-10-14:
In regards to @Enigma0, I would refer you to issue #70 to track progress as I am currently doing.

@NSchepsen I'll compile the latest version here on my RX480 and report back in regards to the icon not showing up. Frankly, to be quite honest with you as many have already reported, GPU usage is currently broken.

mainmachine, 2018-03-08:
@lhorace - Am I missing something? How does the output of ```glxinfo -B``` give you GPU usage info...?

I also have this problem. I'm on Ubuntu 16.04.4 LTS, 4.13 kernel, RX 580, Mesa 17.3.2 from Padoka stable PPA

Everything else works fine, and I can scrape GPU usage like this:
```
sudo grep -i load /sys/kernel/debug/dri/0/amdgpu_pm_info
```

```
~$ sudo grep -i load /sys/kernel/debug/dri/0/amdgpu_pm_info
GPU Load: 100 %
```

lhorace, 2018-03-11:
@mainmachine In regards to the original OP's issue, his distribution of glxinfo thrown a glxinfo usage error because he is missing `-B`.

As far as GPU usage is concerned, that's still broken, originally `radeon-profile` was stating /sys/kernel/debug/dri/0/amdgpu_pm_info for GPU usage but unfortunately as a side effect, it caused stutter in 3D games. Last I read, the developer was still having issues greping GPU usuage and I haven't seen much commits from him in awhile now.

See #54 and https://bugs.freedesktop.org/show_bug.cgi?id=101182

mainmachine, 2018-03-13:
@lhorace - thanks, makes sense now. :)

colesdav, 2018-08-20:
@lhorace - Thank you for pointing me to this. I have been trying to get Radeon Profile to run on Ubuntu 16.04LTS with AMDGPU/PRO 18.10 drivers and ROCm install. I in stalled rocm-smi as well but it reports no GPU Utilization. I tried Radeon Profile GUI - which I think AMD should point to since they do not provide a GUI in Ubuntu 16.04 for 2 years. I tried it here: https://youtu.be/iro19Yldd2Y?t=135 

Reading https://github.com/marazmista/radeon-profile/issues/70
It looks like a new Linux kernel patch will be released to expose GPU usage information in future: 

"A patch has been submitted to expose GPU usage via Sysfs: https://lists.freedesktop.org/archives/amd-gfx/2018-June/023283.html It would likely appear in Linux 4.19.
--"

marazmista, 2019-01-27:
Fixed in commit e870733ed0a1ba4ca8bfb0cefd7cec979ee26413

Podesta, 2019-02-25:
Problem persists for me on latest git (RX580 for reference). I didn't have it a couple months ago. Fresh install on ef512eb I noticed this issue. Just updated to c29341c to fix the fan rpm issue and the gpu usage problem persists. Both the icon on the 'Summary' is missing, and also there is no option to add Gpu utilization on the graphs tab.

glutanimate, 2019-06-02:
@Podesta same here on an RX460

Podesta, 2019-09-03:
Small update on the issue. Awhile back, not sure when, the gpu usage information came back to the property list, maybe after a kernel update. But the bar/graph at the top is still missing (only shows vram usage and fan speed). I cal also add it to the graph tab now.

Oxalin, 2020-02-04:
@Podesta Is this still relevant? Are you able to add the CPU usage pie chart manually: Configuration > General > User interface > Configure top bar > Add > GPU usage (as a pie)

Oxalin, 2020-02-04:
@Podesta Is this still relevant (according to #70)?

Podesta, 2020-03-01:
@Oxalin Sorry, I won't have access to machine with the AMD card for a few months. I'll update as soon as I can, but of somebody else could try it would be better.