marazmista/radeon-profile #79: I can't set fan speed

lhorace, 2017-10-14:
Hello,

Kernel: 4.13.5-1-zen
GPU: RX 480

I can't seem to set fan speed here? Setting `auto` to `manual` simply resets back to `auto`? Applying fixed or custom curves does nothing?

cj360, 2017-12-01:
Does running radeon-profile as root cause the fan speed to change? Until I started the daemon the qt-ui wouldn't affect the gpu.

anjandev, 2017-12-30:
Kernel: 4.14.8-1-ARCH
GPU: R9 270

I was having the same issue as @lhorace. I switched to using the daemon:
https://github.com/marazmista/radeon-profile-daemon
and ran radeon-profile as non-root as @cj360 described. Fan speed on the gpu now obeys what I put into radeon-profile. I just have to have radeon profile running all the time. I entered the following options in my settings for radeon-profile:

傉䝎਍ਚ

lhorace, 2018-02-18:
I apologize for the long absence as I haven't been on Linux for quite bit of time. I am now on Ubuntu 17.10 and compiled both radeon-profile (v. 20170714) and radeon-profile-daemon from source and it appears the fan control to be functioning properly. Although I'll need to do more testing to check for reliability.

lhorace, 2018-02-20:
After further testing, fan control is a bit finicky, there appears to be latency between when you click on `activate` and settings to take affect. Furthermore, perhaps on reboot, while radeon-profile states that the active profile, ie, `overclock` in my case. Fan speeds fluctuates at a great margin, which should remain relatively constant due to the settings I have in place, furthermore, clicking on `activate` doesn't make any difference. It's only when switching profile from `overclock`, as `default` does nothing, to `fixed` and back to `overclock` did I notice a difference in fan control.

This is a perfect example in reference to #72, perhaps a green text `settings applied` under `activate` where `unsaved changes` would appear would be very helpful.