marazmista/radeon-profile #80: GPU memory clock stuck at 300Mhz

sanchitsamuel, 2017-10-27:
Been experiencing this for a while now.
As long as I don't run Radeon Profile, everything run great (games).
After running Radeon profile, a lot of times (not all the time) the Memory clock gets stuck at 300Mhz and the performance in games gets affected. 
I'm using 'amdgpu' driver with latest kernel ie. 4.12.
My card is R7 m260.
Overclocking also doesn't change the clock.

lhorace, 2018-03-16:
There was a bug in 'amdgpu' and Windows Radeon drivers, but last I checked, that was suppose to be fixed. You not doing any overclock? AFAIK, radeon-profile won't touch anything unless you make a change? Can you go on the /sys interface and double check the mclk?

Oxalin, 2020-02-04:
@sanchitsamuel Is this still relevant?