marazmista/radeon-profile #83: Depending on OS Ubuntu 16.04 / 17.10 some features work/notwork

MasterCATZ, 2017-12-16:
on Ubuntu 16.04 The clock speeds will not show but fan controller works 

on Ubuntu 17.10 clock speeds show but no fan controller , also the information seems to be mixed 
in-between the cards 

[image]
[image]

MasterCATZ, 2017-12-17:
booting Ubuntu 17.10 with just the R9 290 with just the amdgpu module loaded fan control is their , 
but the gauge does not show it 
[image]

eode, 2018-11-18:
I don't know how related this is, but on 18.04, some features appear to work, others don't:

* Installed with PPA
* Lenovo e585 -- RX Vega 10
* Kernel 4.19.2-041902-generic #201811132032
  * Needed as prior versions have significant keyboard driver issues

Pardon if I'm using the wrong terms here, but:
* Profile (battery/balanced/performance) doesn't work
  * Momentarily appears to switch, but promptly changes back to "Performance"
* Power Level (auto/low/high/manual/profile_min_sclk, etc) definitely *does* work
* Manual fan control doesn't work.
* `DRM_IOCTL_AMDGPU_INFO: Invalid argument` repeated in terminal
* Connects to daemon fine, no other log messages from daemon

I can transfer this to a new issue if you think it's unrelated.