marazmista/radeon-profile #86: Qtcharts file not found-Solus Plasma desktop

Boethias, 2018-05-12:
I'm trying to build and compile in Solus.  I downloaded the Qt-charts file from Solus software centre but I'm still getting the compile error: **QtCharts: No such file or directory**
[image]

I have qt-charts 5.10.0 though I'm not sure what the file path is. 
Solus with the KDE Plasma 5.12 desktop

Cytomax55, 2018-06-05:
i seem to be getting the same issue
when i run qmake && make

g++ -c -pipe -std=c++11 -O2 -Wall -W -D_REENTRANT -fPIC -DQT_NO_DEBUG_OUTPUT -DQT_NO_DEBUG -DQT_CHARTS_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCharts -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I. -isystem /usr/include/libdrm -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o main.o main.cpp
In file included from components/pieprogressbar.h:10:0,
                 from radeon_profile.h:9,
                 from main.cpp:1:
./ui_pieprogressbar.h:13:10: fatal error: QtGui/QAction: No such file or directory
 #include <QtGui/QAction>
          ^~~~~~~~~~~~~~~
compilation terminated.
Makefile:608: recipe for target 'main.o' failed
make: *** [main.o] Error 1

OS: Kubuntu 18.04 up to date

CPU: Ryzen 1600X

MB: ASROCK 350B Pro (latest non Beta Bios) http://www.asrock.com/mb/AMD/AB350%20Pro4/

GPU: XFX RX580 gpu http://xfxforce.com/en-us/products/amd-radeon-rx-500-series/rx-580-gts-8gb-dd-rx-580p828d6

Qt 5.6<= (qt5-base and qt5-charts) (qt5-defaults for Ubuntu/Debian)

qmake -v
QMake version 3.1
Using Qt version 5.9.5 in /usr/lib/x86_64-linux-gnu

installed using 
sudo apt install qt5-base
sudo apt install libqt5charts5 libqt5charts5-dev

libxrandr
sudo apt install ibxrandr-dev
ibxrandr-dev is already the newest version (2:1.5.1-1)

libdrm (for amdgpu, 2.4.79<=, more recent, the better)
https://launchpad.net/~paulo-miguel-dias/+archive/ubuntu/pkppa
currently at version 2.4.91

recent kernel (for amdgpu 4.12<=, more recent, the better)
radeon card

moriel5, 2018-11-12:
I'm joining this issue, from Solus as well, just with the Budgie DE (though I also have Plasma installed, as well as GNOME and Maté).
From what I gather from the `qt5-charts` eopkg (you can manually download the packages from packages.getsol.us (formerly packages.solus-project.com), the root should probably be set to /usr/lib64/qt5.
As to the specific file, well, there isn't a binary, with the closest thing (in my opinion, I'm not very well aquainted with QT) being either libQt5Charts.pri in /usr/lib64, or qt_lib_charts.pri within /usr/lib64/qt5/mkspecs/modules.

@Cytomax55, your issue is certainly not the same issue, as it has nothing to do with `qt-charts`, but rather with (I think) `qt-actions` (or something like that), but it is similar.
In light of that, my information will not be of much practical use to you, since the file locations are different between Solus and Ubuntu.

moriel5, 2018-11-25:
Alright, this issue is resolved by installing `qt5-charts-devel`.
For Ubuntu users, I'm guessing that it may be something like `qt5-charts-dev`.

closed 2019-01-27 by marazmista

Kabouik, 2019-05-27:
Thanks, I was facing the same issue with Solus too!

moriel5, 2019-05-28:
I'm happy that this helped you.
However Radeon Profile is already in the repos, so why did you compile it from source?