marazmista/radeon-profile #89: Constant 0 RPM fan speed

kajocina, 2018-07-05:
Hi there,

I am trying to fix a problem with my system (Ubuntu 18.04 + Dell XPS8700 + Radeon R9 270). My system seems to only use the GPU fan at 100% when the GPU temperature reaches ~80 degrees (without even using the card!). Otherwise it's at 0%.

I installed radeon-profile and can control the fan speed, i.e. I can set it manually to 100%. However, the software always displays 0 RPM fan speed. I also can't set anything like 30-50%, only max.

Is there anything I can do it about it?
Here is a screen:

[image]

marazmista, 2018-08-23:
Hi
Please check `fan1_input` file in `/sys/class/drm/card0/device/hwmon/hwmon2/` (numbers in path may be different). If 0 is inside that file, there must be something wrong with driver reporting 0 RPM speed.

kajocina, 2018-08-27:
Thanks for your reply! I already removed that card from my PC and now use an integrated video chip. It seems that indeed the driver wouldn't work properly in my setup. Since I can't follow up the problem anymore, I will close the thread.

Best!

closed 2018-08-27 by kajocina