marazmista/radeon-profile #91: Core or Memory Clock manual options grayed out with "use daemon as data provider" unchecked

xxmodxx788, 2018-07-26:
I am somewhat new to linux to hi o/.
OS: Ubuntu 18.04
GPU: XFX R7 240 2gb
I tried the issue with the same problem already but I already have it Unchecked in the screenshots below. Any idea? (daemon ran as sudo) (radeon profile not ran as sudo)
[image]
[image]

marazmista, 2018-11-03:
Manual frequency control is not related to 'use daemon as data provider' setting.
Check if you have `/sys/class/drm/card0/device/pp_dpm_sclk` file because it it used to set frequency manually.

marazmista, 2019-08-26:
Implemented in 9a364be47b751165504362b49b38d011c6712636 and e7922095c2cea1de4f8cee61441496891ddb0bd7

closed 2019-09-03 by marazmista