marazmista/radeon-profile #92: GPU Utilization not displayed on R9 FuryX/Nano.

colesdav, 2018-08-12:
Hi, 
First of all thank you for creating the GUI. 
I am running Radeon Profile on Ubuntu 16.04LTS. 
Video here: https://www.youtube.com/watch?v=iro19Yldd2Y
GPU Utilization is not reported. 
Any ideas why not? 

CPU = i7-4790K. 
GPU1 = R9 Fury X. 
GPU2 = R9 Nano. 
MB = Asus Z97 Deluxe. 
SSHD = 2TB Seagate. 

Thanks.

colesdav, 2018-08-20:
I looked at the code and it looks like an attempt is made to launch a separate CPU Thread to interrogate gpu usage information to avoid lag  and performance issues in the GUI. 

Searching the issues list earlier I can see the following post recarding issue 70: 

https://github.com/marazmista/radeon-profile/issues/70

Specifically the following very helpful  statement 

"lhorace commented on Mar 11 •

@mainmachine In regards to the original OP's issue, his distribution of glxinfo thrown a glxinfo usage error because he is missing -B. Additionally, his GPU Usage icon is missing, and I don't know his glxinfo is the cause for it.

As far as GPU usage is concerned, that's still broken, originally radeon-profile was stating /sys/kernel/debug/dri/0/amdgpu_pm_info for GPU usage but unfortunately as a side effect, it caused stutter in 3D games. Last I read, the developer was still having issues greping GPU usage via libdrm and I haven't seen much commits from him in awhile now.
See #54 and https://bugs.freedesktop.org/show_bug.cgi?id=101182"

Next Steps - Thank lhorace for posting that information - it has saved me lots of time. 
                 - Post this information to AMD Community Forum Thread. 
                 - Investigate #54.

colesdav, 2018-08-20:
Here is my response to lhorace and the summary which is: wait for a new Linux 4.19 Kernel Patch.

Hi,

Thank you for your summary of GPU_U usage problem.
I have been trying to get Radeon Profile to run on Ubuntu 16.04LTS with AMDGPU/PRO 18.10 drivers and ROCm install. I in stalled rocm-smi as well but it does not report GPU Utilization.

I tried Radeon Profile GUI - which I think AMD should point to since they do not provide a GUI in Ubuntu 16.04 LTS for 2 years.

I had a few problems getting the correct version of Qt installed and a few other compile issues, however I show it running here: https://youtu.be/iro19Yldd2Y?t=135

Reading #70
It looks like a new Linux kernel patch will be released to expose GPU usage information in future:

"A patch has been submitted to expose GPU usage via Sysfs: https://lists.freedesktop.org/archives/amd-gfx/2018-June/023283.html It would likely appear in Linux 4.19.
--"

Bye.

marazmista, 2018-08-24:
Hi, thanks for your investigation and effort.
It looks like this patch has been merged into 4.19, so after release of -rc1 I will take a look and try to implement this.

colesdav, 2018-09-10:
Hello, Thank you so much for replying and looking into this further. 
I really appreciate your efforts on this.

Do you have a Patreon Page so people could sponsor your work?
I think you may have an opportunity here with this Radeon Profile Software.

With the advent of Steam Play Beta and the general unhappiness I have seen  with Windows 10 over past 2 years I think the need for a Radeon Adrenalin type GUI may increase rapidly. 
I think many people will start to move to Linux from Windows 10 now. 

I have been testing Steam Play Beta over the past week. 
I was asked to look at testing Fedora 28 with latest Mesa 3.0 devel drivers. 

Here are a few games that run very well indeed on Fedora 28 with latest MESA 3.0 devel drivers. 

1. Monster Hunter World.
https://www.youtube.com/watch?v=nOIsF8g5EJQ
https://www.youtube.com/watch?v=twBfPz0_5WI

2. Hellblade Sensua's Sacrifice. 
https://www.youtube.com/watch?v=uFZhieTb798
https://www.youtube.com/watch?v=UDexxpwc7yY

3. MURDERED: Soul Suspect: 
https://www.youtube.com/watch?v=csSe8f7_LAQ

4. Crysis 2: Maximum Edition: 
https://www.youtube.com/watch?v=bIonDcl7yCw
https://www.youtube.com/watch?v=IaRXjReeinU

There are many more. 
Also some Impressive games run natively on Linux already: 

5. Rise of the Tomb Raider 
https://www.youtube.com/watch?v=b7d6t36R_Nw

As you will see at my Youtube page from I have tested many other games already.  

I tried to install Radeon Profile on Fedora 28 before I ran the Steam Play Beta testing. 
I wanted to be able to show it running and working in the testing videos if possible and also use it to monitor the GPU performance. 
I hit a few problems during compile. 
The compilation reported some files as missing. 
I tried using some files that were already on my Ubuntu 16.04 installation, and the compile did get further. But I had to push on with the Steam Play Beta testing asap. 
Could you please tell me if you think there is ant way I could get Radeon Proifile running on Fedora 28? Should I open an Issue report on that or is it pointless?

Please note there is no AMDGPU/PRO now known as "Radeon™ Software for Linux® 18.30" driver support from AMD specifically for Fedora 28. 

I note that AMD have now implemented some basic Wattman Type functionality in their "Radeon™ Software for Linux® 18.30" driver 

It says the following in the Release Notes:
"Added preliminary Wattman-like functionality; please refer to the sample script included in the /examples folder of the amdgpu-dkms package for more information. "

I have not installed that on Ubuntu 16.04/18.04 yet, but when I do I will post some information about it here. 

Thanks again for all of your work on this. 
Bye.

X0rg, 2019-01-27:
@colesdav: Is your problem fixed since e870733ed0a1ba4ca8bfb0cefd7cec979ee26413, by chance?

Podesta, 2019-02-25:
@X0rg problem persists for me on a RX580. Made the comment on #77, since I'm not sure if this issue is specific to the R9 or not.

Oxalin, 2020-02-04:
@Podesta Please update status according to #77