marazmista/radeon-profile #101: UVD (cclk) (dclk) graph problem

muzena, 2018-10-28:
Hi @marazmista 
Can you add back options to show in graphs UVD core frequency (cclk) and UVD decoding frequency (dclk) like it was in older version of radeon-profile.

marazmista, 2018-11-03:
Hi, it hasn't been removed but as far as I know such info is not available through ioctls when amdgpu module is used.

muzena, 2018-11-03:
@marazmista 
I don't use amdgpu, i have old HD5700 gpu it uses radeon I think,
When I translating strings.hr.ts file, I saw message that is those two options disabled in source code!

marazmista, 2018-11-03:
If you have 'Use daemon as data provider' unchecked, try check it. Then it should show something nonetheless I see that some fixes in code are needed.