marazmista/radeon-profile #102: Ubuntu 18.04.1 unable to install radeon-profile

GottaSlay, 2018-10-31:
I keep getting this when trying to install. 
` sudo apt-get install radeon-profile
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 radeon-profile : Depends: qt59charts-no-lgpl but it is not installable
E: Unable to correct problems, you have held broken packages.
`

muzena, 2018-11-06:
@GottaSlay 
Try now, I think it should work now.

GottaSlay, 2018-11-06:
Finally got it working! Thanks!

GottaSlay, 2018-11-06:
Everything else works apart from the Overclock sliders, the max speed just wont change no matter what i set the overclock values to.

muzena, 2018-11-07:
I don't know, i think this isn't problem in packaging, ask @marazmista

raphaelbeckmann, 2018-11-13:
I cannot install radeon-profile either:

```
$ sudo apt-get install radeon-profile
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package radeon-profile
```

Why cannot the package be located?

GottaSlay, 2018-11-13:
Did you add the repository by doing `sudo add-apt-repository ppa:trebelnik-stefina/radeon-profile
` and then doing `sudo apt update`? After this you should be able to install it via `sudo apt-get install radeon-profile`.

win8linux, 2019-02-19:
I installed the program from the PPA on KDE neon, which is based on Ubuntu 18.04.2 as of the time of writing this. I cannot change any settings ***at all*** since they would all revert back immediately after. Should I file this as a seperate issue?

Specs (`neofetch --stdout`):
```
OS: KDE neon User Edition 5.15 x86_64 
Host: Nitro AN515-42 V1.12 
Kernel: 4.18.0-15-generic 
Uptime: 8 hours, 29 mins 
Packages: 2463 
Shell: bash 4.4.19 
Resolution: 1920x1080 
DE: KDE 
WM: KWin 
Theme: Breeze Dark [KDE], Breeze-Dark [GTK2/3] 
Icons: Breeze-dark [KDE], Breeze-dark [GTK2/3] 
Terminal: konsole 
CPU: AMD Ryzen 5 2500U with Radeon Vega Mobile Gfx (8) @ 2.000GHz 
GPU: AMD Radeon Vega Series 
GPU: AMD Radeon RX 460/560D 
Memory: 2647MiB / 6930MiB
```

marazmista, 2019-02-19:
I see you have mobile vega. Are you able to change any of settings described here (https://wiki.archlinux.org/index.php/ATI#Dynamic_power_management) manually?

Do you have daemon installed and running? I don't know which version of radeon-profile is in the PPA. Also what kernel version?

win8linux, 2019-02-19:
Unfortunately, I am unable to change any settings described in the link. In particular, running `echo low > /sys/class/drm/card0/device/power_dpm_force_performance_level` shows this error:
```sh
bash: echo: write error: Invalid argument
```
The daemon is installed and running. The version of `radeon-profile` in the PPA is 0.2.4 and the kernel version is `4.18.0-15-generic` (from HWE stack).

marazmista, 2019-02-20:
If you unable to change values by hand then `radeon-profile` won't work either and problem may be somewhere else.
Also I don't know what 0.2.4 means. Version is in bottom-right corner of the window or in About tab.

win8linux, 2019-02-20:
Oh, okay then @marazmista. The ***actual*** version is `20181028`. The previous number was obtained by checking the package version.

muzena, 2019-02-24:
@win8linux 
If you install radeon-profile from PPA last changes are not available in PPA.
It caused by these commits in [radeon-profile-daemon](https://github.com/marazmista/radeon-profile-daemon/commit/354961e5f3f559ee3580b0ff9d359e66241fb9c3) and [radeon-profile](https://github.com/marazmista/radeon-profile/commit/260af611d3318dcabaf856d701af3fe9b23d4f32) 
It is in conflict with [debian rules](https://github.com/muzena/radeon-profile/blob/packaging_bionic/radeon-profile/debian/rules) building file, so this is reason why automated PPA building is no longer working.
I tried to configure debian rules file but without success for now.

muhamed89, 2019-02-25:
Whats the best way to install it on Linux Mint?