marazmista/radeon-profile #104: When the app is closed, fan settings are are lost and speed goes up.

HubKing, 2018-12-03:
I only need fan control. I have installed the daemon. This is like Windows' service, right? Then, why do I need to keep the heavy GUI application running? Shouldn't the GUI only be used for user interaction, and the work be done by the daemon?

When I close the GUI app, the fan speed goes higher even than before running the GUI app. The GPU temperature is not even high (40 degrees), so there is no reason the fan speed goes up to 2000 RPM. It looks as if when exiting the app, fan control is completely lost and the card just applies a high fan speed.

So, in short,
1. How can I make the fan control work when only the daemon is running?
or if that is not possible,
2. How can I at least make the original (before running the GUI app) fan controlling restored after exiting the GUI app?

lhorace, 2018-12-03:
Hello,

1) Actually the intended purpose of the daemon is to provide access to video card settings, see readme https://github.com/marazmista/radeon-profile-daemon, since only superuser has such access.

2) AFAIK, out abundance of caution, `radeon-profile` simply set's Fan speed to %100 before exiting to avoid potential  overheating. You may try setting Fan control back to default settings, which imply allowing the VBios manage fan speeds prior to quiting `radeon-profile`.

It might be good idea, to simply set Fan Control to it's default state after exiting, in lieu of it's current behavior. If indeed is what's happening there.

Question, when you logging out, are you planning to shutting it down?

lhorace, 2018-12-03:
Well, I've been following this repository for quite some time, I highly doubt the developer would implement any sophisticated features from within `radeon-profile-daemon`, unless you submitted a PR and he accepted it. Alternatively, perhaps look into `fancontrol` from `lm_sensors`, albeit rudimentary implementation in lieu of controlling the fan speed from `radeon-profile`?

marazmista, 2018-12-13:
Hi @HubKing 
> Changing it back to default before exiting the app and changing it back to custom settings when the app starts, every time? That is too cumbersome.

This should happen automatically if you have 'Save selected fan control mode' checked in settings. 

> When I close the GUI app, the fan speed goes higher even than the speed before running the GUI app. 

> How can I at least make the original (before running the GUI app) fan controlling restored after exiting the GUI app?

When GUI is closing, fan profile is being set to Auto and speed is controlled by kernel. You can check this in file: `/sys/class/drm/card0/device/hwmon/hwmon2/pwm1_enable`. If value is `2` after closing GUI, then radeon-profile doesn't control fan and the speed should drop to value that kernel or card's BIOS think is ok for current temperature.

> I do not know about the technical details or philosophy, but as a user's perspective, it is better that a constant thing like fan controlling is done in the background by a service without the need to start a GUI application. So, maybe adding an option to make the daemon do the work would be helpful.

> How can I make the fan control work when only the daemon is running?
or if that is not possible,

That is not possible and I don't think that overriding card's internal fan profile and do it in background is good. Someone can just set to low fixed speed and forget about it, start game and overheat. GUI is more visible. Besides, all the logic is in GUI because app can work without daemon (as root) and implementing logic in two places is not the best idea I think. GUI is always heavier, so at least daemon can be simple.

> KDE cannot log out when the GUI app is running. It says the logout was cancelled by it.

I've installed KDE to test, but couldn't reproduce that. Although, after some digging, I think that commit 6413b3b097a54bc080ef3934e1bee95a049ea8c3 may fix this. If you can compile and test that would be cool.

> The control does not seem to be applied immediately when the GUI app is started. I have to click "Activate".

Fan profile is activated, but calculations about new fan speed according to selected profile are delayed until temperature change.