marazmista/radeon-profile #106: tint2: Tray icon disappears after restarting the systray

Vladar4, 2018-12-19:
After tint2's systray restart (when turning on/off additional monitors, for example) tray icon disappears.

Looks like the radeon-profile does not handle the icon reloading properly at the moment, see https://gitlab.com/o9000/tint2/wikis/FAQ#why-did-some-systray-icons-disappear-from-the-panel

marazmista, 2019-01-20:
It's difficult to find something about reloading tray icon. It seems like app doesn't get any info when tray icon is not visible anymore. 
Similar issue I think: https://bbs.archlinux.org/viewtopic.php?id=243148
Anyway, I assume that app is not killed, just the tray icon disappear, right?
I pushed a commit 48f1770f3ef0c114fdf46aa1e306218cca9efca1 to `rp-dev` branch that checks if tray icon is visible, could you compile and test to see if this help?

Vladar4, 2019-01-20:
> Anyway, I assume that app is not killed, just the tray icon disappear, right?

Correct

> I pushed a commit 48f1770 to rp-dev branch that checks if tray icon is visible, could you compile and test to see if this help?

Sadly, it does not fix the issue. `isVisible()` always returns `true', and even `setVisible(true)` or `show()` do not restore the icon.

I've found recent potentially related issue here: https://gitlab.com/o9000/tint2/issues/725