marazmista/radeon-profile #108: Graphs are always black

Zamundaaa, 2019-01-03:
As the title says.
Also I can't set the performance level, it stays stuck on Performance, if I try setting it to battery or balanced it switches back in a fraction of a second. Could it have something to do with me having a RX 580 Sapphire Nitro+ that has a hardware switch for performance level (albeit just 2 possible states)?

marazmista, 2019-02-08:
Not sure what you mean, could you provide screenshot of this black graphs?
About changing states, it seems that writing value to sysfs failed, do you have daemon installed?

Zamundaaa, 2019-02-11:
Here's the screenshot of black graphs. I have no idea which is which.
[image]
I do have the deamon installed with the AUR package and I'm sure it's running
[image]
I don't know if that could point to the issue but there seems to be some file path empty or something
[image]

parkerlreed, 2019-02-11:
Have you set a color? Default is black. It doesn't auto pick colors

[image]

Zamundaaa, 2019-02-11:
Ah yes. I feel a little stupid now...
Should I open a seperate issue about power states not being set? The whole Overclock tab doesn't work for me either. No clockspeed setting and the "States Table" sub-tab is greyed out

parkerlreed, 2019-02-11:
@Zamundaaa Yeah separate issue if one doesn't already exist.

Zamundaaa, 2019-02-11:
Ok the rest of the issues is now tracked on #120

closed 2019-02-11 by Zamundaaa