marazmista/radeon-profile #109: Question about removing/modifying first and last fan curve items

rigrassm, 2019-01-26:
I noticed that you are not able to remove the first and last items from the fan curve and that this seems to be a hard coded block put in.

I was curious as to why this limitation was put into place since it leads to having some odd profile configurations.

For example, I don't want my fan curve waiting until it's at 90 degrees C before hitting 100%. Since I'm not able to modify the temp in the last step I have to create a new step to max out the fan speed at my desired temp. In this case you end up with the last step not even being reachable since the fan speed will be set to 100% before that temp is ever reached.

After seeing the behavior above and looking at the source, i noticed that you can still define a step that comes after the original "last step". Once the new step is created, the step that was the last step can then be deleted with the one created to go after it becoming "stuck". 

Example:

1. Create a copy of the default profile.
2. Add a new step with the temp set to 99 and the fan % set to 100
3. Notice you can delete the the 90c/100% step but cannot remove the 99c/100% step that was just created. 

I get having a fixed 0 degree step which I assume is meant to be used to set a minimum fan speed for the profile but I feel like the restriction of altering/removing that last step could probably be safely removed.

I've personally worked around this by manually editing the radeon-profile-auxstuff XML file to remove the unwanted steps from the curve profile.

marazmista, 2019-02-03:
Basically the last step is locked, because plot next to the table is exact representation and it has to reach the 100% speed somewhere. I started to think about it and maybe after putting some artificial point on the plot (to have speed line continuity to the end) and adjusting the algorithm (so it would set speed to 100% after last defined point), the current restrictions on table could be removed.

rigrassm, 2019-02-03:
Is there a specific reason the curve has to end with the fan speed being set to 100%?

Neither Wattman nor MSI afterburner(only two other programs I've used personally for setting fan speeds) require the curve to end with fan speeds being set to 100%.

I think a simple check to ensure the last step has both a higher temp and higher speed than the second to last step would be sufficient. Maybe pop up a warning if the user tries to set a temp or speed that doesn't meet those requirements.

marazmista, 2019-02-05:
Back then I thought about safety so this was the only reason but now such limitations may not be needed.
I removed locked steps, modified the fan steps list and plot, now it should work similar to Afterburner. (commit 0013a4f3c7040ec57bc780c207b1430f59bca837)

closed 2019-02-11 by marazmista