marazmista/radeon-profile #110: Build error on Ubuntu 18.04

muzena, 2019-01-26:
@marazmista 
After recent changes you made I get this build error in [build.log](https://launchpadlibrarian.net/408297346/buildlog_ubuntu-bionic-amd64.radeon-profile_0.2.4-201901252233~ubuntu18.04.1_BUILDING.txt.gz)

Danysan1, 2019-01-26:
This could be caused by a bug in the Qt User Interface Compiler which has been solved on December 14, 2018: https://bugreports.qt.io/browse/QTBUG-72555
Maybe Ubuntu packages haven't included the bugfix yet.

marazmista, 2019-01-28:
Maybe you can apply this patch (https://gist.github.com/marazmista/31474fb438a787995c9c4cca29f4851c) and try to compile until Ubuntu update its packages.

Cyclic3, 2019-02-03:
This still occurs in debian/sid.