marazmista/radeon-profile #112: Segfault on debian/sid

Cyclic3, 2019-02-03:
On running the software, I get a segfault. Running a quick git bisect, I found that this break was introduced on a5218fe532ab72690ce547c51064b7c2e53487d8. I have a Vega 56. This still occurs when run as root.

GDB backtrace:
```
#0  0x00007ffff695181c in QString::toUInt(bool*, int) const () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00005555555a6fe2 in dXorg::loadOcTable() ()
#2  0x00005555555a7d6a in dXorg::figureOutDriverFeatures() ()
#3  0x00005555555a80b9 in dXorg::configure() ()
#4  0x00005555555a82bd in dXorg::dXorg(GPUSysInfo const&) ()
#5  0x000055555559c41a in gpu::initialize() ()
#6  0x00005555555791ea in radeon_profile::radeon_profile(QWidget*) ()
#7  0x0000555555574ef2 in main ()
```

`uname -a`: "Linux chevron 4.19.0-2-amd64 #1 SMP Debian 4.19.16-1 (2019-01-17) x86_64 GNU/Linux"

blisbell, 2019-02-03:
I can confirm on arch linux with a vega 56 and get the same segfault.

Sebastian-Hirsch, 2019-02-03:
I can confirm it too on the current Manjaro Linux Stable Version with my Radeon RX Vega 56.

marazmista, 2019-02-03:
Could you provide `pp_od_clk_voltage` file? Maybe for Vega 56 it looks different than samples I've found over internet.

Sebastian-Hirsch, 2019-02-03:
Here is mine. I have already overclocked my card, but that shouldn't matter
[pp_od_clk_voltage.txt](https://github.com/marazmista/radeon-profile/files/2826060/pp_od_clk_voltage.txt)

.

marazmista, 2019-02-03:
Thanks, I've pushed a commit 51fd135c1b6da8a12b0a6c230fdb98bb217c9fa1 with a fix.

Cyclic3, 2019-02-04:
@marazmista This is the quickest github issue fix I have seen in a long while. Thanks!

closed 2019-02-04 by Cyclic3