marazmista/radeon-profile #113: Graph Settings are not saved between launches

Sebastian-Hirsch, 2019-02-04:
When setting Graphs in the Graphs Menu and restarting the application, all graphs are lost and i have to set them up again.

Tested with the latest git version of this tool, Linux Kernel 4.20 and Mesa 18.3.2.

marazmista, 2019-02-16:
After setting them up, try clicking Apply and later go to the Configuration tab and click Save all. Are they restored after next application launch?

Sebastian-Hirsch, 2019-02-17:
Going to the config page and press "Save all" there seems to work with the latest git code. But it is really not intuitive not to save the graphs after pressing the apply button. It would be good to change this behavior as it is not visible to the user, that you have to save the config in a separate menu.

marazmista, 2019-02-17:
Save config to file is also called when you close the application, so I don't know why it worked when you manually pressed the button but not when closing normally (killing app won't save config).
Yes, I will modify Apply button behavior.

marazmista, 2019-02-18:
Fixed 3ff78b282be9a9c73c990159effc41ed8adead5f

Sebastian-Hirsch, 2019-02-18:
Can confirm this, I will close this issue.

closed 2019-02-18 by Sebastian-Hirsch