marazmista/radeon-profile #115: Fan percentage and fan rpm shown wrong for RX Vega 56

Sebastian-Hirsch, 2019-02-04:
The shown fan rpm and fan percentage is completly wrong. When the fans are at 1000rpm, it shows 50 - 60 rpm. This seems not to be an amdgpu specific issue as i use an gnome shell extension which shows the correct rpm values. I use the following gnome shell extension:

https://extensions.gnome.org/extension/1460/vitals/

I hope the code of this extension could help fixing the bug

System: Manjaro Linux, Kernel 4.20, Mesa 18.3.2, Radeon RX Vega 56.

[image]

fatino, 2019-02-20:
I have the same issue with RX 470. Percentage seems to be correct. Just the RPM number seems wierd.

aaulia, 2019-02-20:
+1, RX 470, same issue.

marazmista, 2019-02-21:
probably fixed in commit c29341c3a363a9fb14a64865f685966f083f97f7

Sebastian-Hirsch, 2019-02-21:
It works for me now. I will close the issue.

closed 2019-02-21 by Sebastian-Hirsch