marazmista/radeon-profile #116: Move profile storage and profile application to the daemon so the app doesn't need autostart to apply the fan profile

Gaming4LifeDE, 2019-02-08:
The fan profiles should be moved to the daemon. This would make it possible to just have the daemon autostart to be able to apply a fan profile on boot.

On the same note, maybe also move all the logic to a library so GUIs in differen toolkits can be implemented to integrate better with different desktop environments.

marazmista, 2019-02-09:
This was somewhat discussed here: #104

Lomomn, 2019-08-02:
Would this option be viable if a disclaimer was added to the program warning people that their fan settings would be applied at startup and it was up to them to make sure that the correct curve is being used?
I assumed that "Save selected fan control mode" would do this, but the program doesn't actually remember this because it's a safety feature. 

For me, setting even the 20% fan setting automatically at startup would be beneficial as my card's auto setting means that the temp reaches 75C when idle. Any profile is better than auto in my case.