marazmista/radeon-profile #119: Can't start radeon-profile: Daemon is not connected, therefore it can't be configured

alex9k1, 2019-02-10:
When I try to run ```radeon-profile``` it shows me the following:

> Translation not found.
Daemon is not connected, therefore it can't be configured
zsh: segmentation fault (core dumped)  radeon-profile

The first line can be avoided with ```LANG=en_US radeon-profile```, but it still won't launch. I've built it with ```CONFIG+=debug```, and when I try to launch it now, it shows:

> Creating application object
Creating radeon_profile
Loading configuration
Initializing device
Card detected (amdgpu):  "card0"
hwmon path:  "/sys/class/drm/card0/device/hwmon/hwmon2/"
Opened /dev/dri/renderD128 , fd number: 12
Connecting to daemon
Daemon is not connected, therefore it can't be configured
ioctlHandler: everything ok
Power method: DPM
ASSERT failure in QList<T>::operator[]: "index out of range", file /usr/include/qt/QtCore/qlist.h, line 552
zsh: abort (core dumped)  LANG=en_US radeon-profile

```journalctl -u radeon-profile-daemon.service``` gives me:

> Feb 10 22:44:22 alex-desktop systemd[1]: radeon-profile-daemon.service: Succeeded.
Feb 10 22:44:22 alex-desktop systemd[1]: Stopped radeon-profile daemon.
Feb 10 22:44:22 alex-desktop systemd[1]: Started radeon-profile daemon.
Feb 10 22:44:22 alex-desktop radeon-profile-daemon[25060]: Starting in debug mode
Feb 10 22:44:22 alex-desktop radeon-profile-daemon[25060]: ok

When I then try to launch ```radeon-profile``` it adds:

> Feb 10 22:45:30 alex-desktop radeon-profile-daemon[25060]: Connecting to the client
Feb 10 22:45:31 alex-desktop radeon-profile-daemon[25060]: Disconnecting from the client

I've tried it from the AUR  an by building it manually, same results.

I'm on Arch with 4.20.7-zen1-1-zen. Over the weekend I also upgraded my graphics card from a R9 Nano to RadeonVII, so idk. if it's related to that or some code changes during that time.

marazmista, 2019-02-11:
My first guess is that this may be something with reading overclock table. Could you provide `pp_od_clk_voltage` file? Maybe for new Radeons it has something extra inside that cause this seg fault.

alex9k1, 2019-02-11:
Sure. https://gist.github.com/alex9k1/dab93a66c209c4099dd258eb516cb8e9

alex9k1, 2019-02-18:
In the meantime I tried to manually undervolt this card, but noticed, that it wouldn't let me overwrite the level values, even though I've set amdgpu.ppfeaturemask=0xffffffff. So, could this issue rather be related to amdgpu?

marazmista, 2019-02-18:
I guess you need to change performance level to `manual` before applying changes to `pp_od_clk_voltage table`:
`echo "manual" > /sys/class/drm/card0/device/power_dpm_force_performance_level`

alex9k1, 2019-02-18:
You're right. I did that, too, but forgot to mention it.

marazmista, 2019-02-18:
Maybe call this after all adjustments?:
`echo "c" > /sys/class/drm/card0/device/pp_od_clk_voltage`

Anyway, thanks for the sample of sysfs file above. I've added support for overclocking for new cards (commit 02d74c01afd0304b2fbd36b2a59ffeb35cec2c11) and this also should fix this crash.

alex9k1, 2019-02-18:
Yep, did that too, idk. Fortunately it doesn't matter anymore :) Just built radeon-profile and it's working again. I can adjust the fan curve and undervolt the GPU - just what I needed. 

A very big thank you!

closed 2019-02-18 by alex9k1