marazmista/radeon-profile #120: Power Modes and Overclocking don't work

Zamundaaa, 2019-02-11:
Power mode is always on performance. If I try to change it to battery or balanced it just goes back after a second or so.
in the overclock tab if I activate "Overclock" or "Manual frequency control" then the I still can't do anything because all the sliders go from 0 to 0 and are greyed out. The sub-tab "States table" is always greyed out.
[image]
The daemon is installed and running. There may be a little problem with it anyhow:
https://user-images.githubusercontent.com/20477130/52562624-5eafca00-2dff-11e9-9fe3-093074e82032.png

parkerlreed, 2019-02-11:
I can confirm I am experiencing both there too.

````
[parker@stealth ~]$ uname -a
Linux stealth 4.20.7-arch1-1-ARCH #1 SMP PREEMPT Wed Feb 6 18:42:40 UTC 2019 x86_64 GNU/Linux
[parker@stealth ~]$ pacman -Q radeon-profile-daemon-git radeon-profile-git 
radeon-profile-daemon-git 20190124.r6.g05bdb90-1
radeon-profile-git 20190127.r11.g19de258-1
````

marazmista, 2019-02-13:
What card do you have @Zamundaaa?

Zamundaaa, 2019-02-13:
Oh yeah I probably could've given more Information there.
I'm on Manjaro KDE. My CPU is an i7 6700k and the GPU is a RX 580 Sapphire Nitro+ edition

SethBarberee, 2019-02-13:
My overclock works fine but power states are locked to Performance on my Sapphire RX Vega 56 Reference.

BlazeKl, 2019-02-17:
I'm experiencing the same issue, installed radeon-profile from AUR today

[image]

I'm using Arch KDE, GPU is RX 470 using recently compiled mesa

Sebastian-Hirsch, 2019-02-18:
I too can confirm this problem with the current stable version of manjaro.
[image]

marazmista, 2019-02-18:
Is it ok after commit 6b4125b856950538f150a3fad7c3668ba4b63a15?

Sebastian-Hirsch, 2019-02-18:
The sliders are now usable, but they seem to have no effect. Could be related to the use of state table overclocking on my side.

closed 2019-02-20 by marazmista

Zamundaaa, 2019-02-21:
Overclock did now begin to work (after a reinstall!) but manual frequency control still shows 0 for both greyed out sliders and the states table too

marazmista, 2019-02-21:
Yes, because manual freq control will be implemented differently.

belenos, 2019-03-05:
I'm having the same issue on v.20181028. Seems to be related to the "refresh interval [s]" inside the Configuration tab. If I increase the value, the new profile set sticks for longer. It also works if I click on "pause refresh temporarily" (btw, it has a little typo. It says "temporAIRly"), which is not an ideal workaround. 

I'm running a KDE Neon fresh install (kernel 4.15.0-45) on a RX 560 OC.

marazmista, 2019-03-05:
Can you change value manually, as described here? https://wiki.archlinux.org/index.php/ATI#Dynamic_power_management

If you click pause or increase interval, new value doesn't stick for longer, it is just that interface doesn't refresh system info.

belenos, 2019-03-06:
> Can you change value manually, as described here? https://wiki.archlinux.org/index.php/ATI#Dynamic_power_management

Changing operation modes manually doesn't work, even after adding the kernel parameter radeon.dpm=1 to grub and rebooting (but the performance mode is set correctly when I change it to low either on terminal or the GUI).

Marco-MP, 2019-06-06:
I can confirm that this issue still persists. 

Overclock seems to be changeable but has no apparent effect and posts this message in the console upon checking the box:
QFSFileEngine::open: No file name specified
Unable to open  ""  to write  "0"

The manual frequency control sliders aren't movable and produce no output in the console.

Manually changing the values seems to be not working as even as root I have no permission to change the necessary files.
However setting the Profile to low for example seems to have an effect as the lowest State is not leavable in this setting.

Regarding the Profile[Battery,Balanced,Performance] changing:
Apparently the changed Profile is indeed correctly written to the file but does indeed get reset.
power_dpm_state registers an update if opened while changing the value. 

Software Version is 2019063
Card is R9 290X

PS: Sorry for necroing a Closed Issue but I was unsure if I should open a new one for this apparently unsolved Problem.

mrgfy, 2020-10-15:
I'm not having any problems using the tool. I have set the menu launcher exec as such: gksudo -P radeon-profile
and i have installed radeon-profile-daemon. First I set systemd service as follows: sudo systemctl enable radeon-profile-daemon.service then start it. After that exec the menu launcher with the command explained earlier. All modes appear to work and are functional. Just my two cents after reading the issues, hope it helps.

mrgfy, 2020-10-15:
[image]
[image]