marazmista/radeon-profile #122: Poor performance after undervolting & pstate changes doesn't stick after restart.

aaulia, 2019-02-16:
My system information

```
~ 
➜ uname -a && pacman -Q radeon-profile-git radeon-profile-daemon-git && inxi -G 
Linux archbox 4.20.8-arch1-1-ARCH #1 SMP PREEMPT Wed Feb 13 00:06:58 UTC 2019 x86_64 GNU/Linux
radeon-profile-git 20190211.r0.g52f3ee3-1
radeon-profile-daemon-git 20190124.r6.g05bdb90-1
Graphics:  Device-1: AMD Ellesmere [Radeon RX 470/480/570/570X/580/580X] driver: amdgpu v: kernel 
           Display: tty server: X.Org 1.20.3 driver: amdgpu resolution: 1920x1080~60Hz 
           OpenGL: renderer: AMD Radeon RX 470 Graphics (POLARIS10 DRM 3.27.0 4.20.8-arch1-1-ARCH 
           LLVM 7.0.1) 
           v: 4.5 Mesa 18.3.3 
```

I've tried and succeed in manually setting the p-state voltage for 6th and 7th state through amdgpu's `sysfs`

Every time I tried changing the voltage from `radeon-profile` it always result in poor performance in games or benchmark that I did after (ie. what usually run around 60 fps are now run in 15 fps). Also after restarting the computer, all settings are back to normal.

marazmista, 2019-02-16:
After changing values from radeon-profile, could you check the file manually to see if they were applied?

aaulia, 2019-02-17:
This is the status of the `sysfs` p-state table before and after I apply the changes from `radeon-profile`. It seems that it's doing it correctly.  

[image]  
  
One suspicious thing that I notice is how the power cap is at `0w`?  

[image]  

This is my gpu status during a benchmark, notice how the gpu clock speed seems to stuck at `300MHz` and voltage is at `1000mV`.  

[image]

Lastly this is my `power1_cap` status, it show normal `130w` default power cap, AFAIK
```
~ 
➜ cat /sys/class/drm/card0/device/hwmon/hwmon0/power1_cap    
130000000
```

aaulia, 2019-02-17:
Hello, I have more information. After reading [archwiki](https://wiki.archlinux.org/index.php/AMDGPU#Overclocking) entry on overclocking. I decided to check on the p-state's state. It seems that after setting the voltage (or maybe anything that related to the p-states) it's stuck/locked on the first state.  

[image]

Sebastian-Hirsch, 2019-02-17:
I can confirm the power cap issue with my RX Vega 56. It is at 0 watts even though it should be at 247W (the value i have set with my own oc script). The max value of the slider is erratically at 99W, which is wrong too.

marazmista, 2019-02-18:
Commit e089053a3f1f8f906424263ed61d585f4eb6964c should fix power cap issue.
Saving pstates adjustments, this on the todo here: https://github.com/marazmista/radeon-profile/projects/1#card-16937033

What if you try to enable other states after adjusting values? Following archwiki it should look like this: `echo "0 1 2 3 4 5 6 7" >  /sys/class/drm/card0/device/pp_dpm_sclk`
It is interesting that the states got stuck after pstate table modifications. I don't quite understand the relation between some files. For example, I'm not sure if percent overclock via `pp_sclk_od` can be applied in the same time as `pp_od_clk_voltage` adjustments.

Sebastian-Hirsch, 2019-02-18:
I can confirm, that the correct power target is now shown correctly.
For the percentage overclocking issue I would suggest to just drop it for all cards, as the state table method is all around better. There could indeed a problem using both but I am not knowledgeable enough in this regard. I would by default just enable the state table overclocking for all cards that support it and the percentage oc for older cards that not support state tables. A list with supported cards for state table oc would be helpful here.

aaulia, 2019-02-19:
The latest commit(s) actually fix the problem. The GPU clock correctly goes to `1260MHz` and the voltage to `1025mV`. The last problem I guess is that the configuration doesn't survive reboot. Because after rebooting the values goes back to default.  

One minor UI problem, the p-state graph only show changed states instead of the whole states, after modification.

[image]

marazmista, 2019-02-19:
Plot issue fixed 4b10fe23e5ed8d45da31fe31954784be20d380a7
Reboots will always reset values, even after I implement saving pstates adjustments, you will need to launch radeon-profile to apply modified numbers.

aaulia, 2019-02-20:
Nope, it doesn't save p-state configurations after reboot.

[image]  

[image]  

In comparison, the fan control configurations survive reboot, and being (re-)applied every time.

aaulia, 2019-02-20:
> Plot issue fixed [4b10fe2](https://github.com/marazmista/radeon-profile/commit/4b10fe23e5ed8d45da31fe31954784be20d380a7)
> Reboots will always reset values, even after I implement saving pstates adjustments, you will need to launch radeon-profile to apply modified numbers.

Just to be clear, maybe I'm mistaken with my assumption. So I have to re-input my voltages and hit apply every time I reboot my PC? Because I'm under the impression that it would work just like the fan control config.

marazmista, 2019-02-20:
I assume you have radeon-profile in autostart.
> So I have to re-input my voltages and hit apply every time I reboot my PC?

Correct. Restoring modified values at application startup (like fan profiles) is not implemented yet.

aaulia, 2019-02-20:
Great, so I can close this issue then. Thank you

closed 2019-02-20 by aaulia