marazmista/radeon-profile #123: Critical issue with client disconnecting from daemon.

fatino, 2019-02-23:
Client randomly loses connection to daemon which results in GUI being greyed out and fans spining up. But today fans didn't even spin up, they stayed low. I'm glad I have GPU temp monitoring in the panel, because I noticed that my GPU temp went up to 60C while watching YT video. I'm just glad I didn't start a game.
This showed in the logs:
```lut 23 09:52:34 greyhill radeon-profile-daemon[814]: Connecting to the client
lut 23 09:52:34 greyhill radeon-profile-daemon[814]: The file path indicated to be set is empty. Lost value:  ""
lut 23 09:55:34 greyhill radeon-profile-daemon[814]: Restoring system defaults
lut 23 09:55:34 greyhill radeon-profile-daemon[814]: Setting fan control mode to auto
lut 23 09:55:34 greyhill radeon-profile-daemon[814]: Client disconnected
lut 23 09:55:34 greyhill radeon-profile-daemon[814]: Awaiting connections...
lut 23 09:55:36 greyhill radeon-profile[1381]: Unable to open  "/sys/class/drm/card0/device/hwmon/hwmon0/pwm1"  to write  "28"
lut 23 09:55:43 greyhill radeon-profile[1381]: Unable to open  "/sys/class/drm/card0/device/hwmon/hwmon0/pwm1"  to write  "22"
lut 23 09:55:54 greyhill radeon-profile[1381]: Unable to open  "/sys/class/drm/card0/device/hwmon/hwmon0/pwm1"  to write  "22"
```
Usually I could just restart the radeon-profile app to get it in shape again. But now I couldn't.
Major issue, almost overheated my GPU.
My system:
Arch Linux
Kernel - 4.20.10-arch1-1-ARCH
AMDGPU drivers
Radeon RX 470
Radeon-profile and radeon-profile-daemon v20190220 from AUR

marazmista, 2019-02-23:
Your GPU would not overheat as system defaults of fan control are restored when connection is lost.

Could you stop system service, compile daemon as debug and start it from another terminal? Logs may provide some more info about how connection was lost.

fatino, 2019-02-23:
That's the case. System defaults weren't restored somehow because the fan was at 5% and my GPU temperature was over 60C. Usually it boosted to about 45% straight up and just stayed there until I restarted the app. 
Anyway I compiled with debug. Started the service. Started radeon-profile. Instant crash. Now my fan speed is locked to 7.8% and my temperature rose to about 58C. I guess if I use more GPU it would overheat.
```lut 23 22:24:08 greyhill radeon-profile-daemon[8323]: Connecting to the client
lut 23 22:24:09 greyhill kdeinit5[1228]: Registering ":1.115/StatusNotifierItem" to system tray
lut 23 22:24:09 greyhill plasmashell[1284]: QQuickItem::stackAfter: Cannot stack StatusNotifierItem_QMLTYPE_295(0x55ab72f1f8a0, parent=0x55ab70b4ab80,>
lut 23 22:24:09 greyhill plasmashell[1284]: QQuickItem::stackAfter: Cannot stack StatusNotifierItem_QMLTYPE_295(0x55ab72971cc0, parent=0x55ab711f2320,>
lut 23 22:24:09 greyhill radeon-profile-daemon[8323]: Performing task:  "0#pwm1_enable#/sys/class/drm/card0/device/hwmon/hwmon0/pwm1_enable#2#1#/sys/c>
lut 23 22:24:09 greyhill radeon-profile-daemon[8323]: Elaborating a CONFIG signal
lut 23 22:24:09 greyhill radeon-profile-daemon[8323]: Fan control mode file:  "/sys/class/drm/card0/device/hwmon/hwmon0/pwm1_enable"
lut 23 22:24:09 greyhill radeon-profile-daemon[8323]: Elaborating a SET_VALUE signal
lut 23 22:24:09 greyhill radeon-profile-daemon[8323]: "1"  will be written into  "/sys/class/drm/card0/device/hwmon/hwmon0/pwm1_enable"
lut 23 22:24:09 greyhill radeon-profile-daemon[8323]: Elaborating a SET_VALUE signal
lut 23 22:24:09 greyhill radeon-profile-daemon[8323]: "25"  will be written into  "/sys/class/drm/card0/device/hwmon/hwmon0/pwm1"
lut 23 22:24:09 greyhill radeon-profile-daemon[8323]: Elaborating a SET_VALUE signal
lut 23 22:24:09 greyhill radeon-profile-daemon[8323]: ""  will be written into  "/sys/class/drm/card0/device/pp_sclk_od"
lut 23 22:24:09 greyhill radeon-profile-daemon[8323]: Elaborating a SET_VALUE signal
lut 23 22:24:09 greyhill radeon-profile-daemon[8323]: The file path indicated to be set is empty. Lost value:  ""
lut 23 22:24:09 greyhill radeon-profile-daemon[8323]: Elaborating a SET_VALUE signal
lut 23 22:24:09 greyhill radeon-profile-daemon[8323]: "auto"  will be written into  "/sys/class/drm/card0/device/power_dpm_force_performance_level"
lut 23 22:24:09 greyhill radeon-profile-daemon[8323]: *** stack smashing detected ***: <unknown> terminated
lut 23 22:24:09 greyhill audit[8323]: ANOM_ABEND auid=4294967295 uid=0 gid=0 ses=4294967295 pid=8323 comm="radeon-profile-" exe="/usr/bin/radeon-profi>
lut 23 22:24:09 greyhill kernel: audit: type=1701 audit(1550957049.034:99): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=8323 comm="radeon-profile-">
lut 23 22:24:09 greyhill systemd[1]: Created slice system-systemd\x2dcoredump.slice.
-- Subject: Pomyślnie ukończono zadanie uruchamiania dla jednostki system-systemd\x2dcoredump.slice
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Pomyślnie ukończono zadanie uruchamiania dla jednostki system-systemd\x2dcoredump.slice.
-- 
-- Identyfikator zadania: 1819.
lut 23 22:24:09 greyhill systemd[1]: Started Process Core Dump (PID 8373/UID 0).
-- Subject: Pomyślnie ukończono zadanie uruchamiania dla jednostki systemd-coredump@0-8373-0.service
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Pomyślnie ukończono zadanie uruchamiania dla jednostki systemd-coredump@0-8373-0.service.
-- 
-- Identyfikator zadania: 1818.
lut 23 22:24:09 greyhill audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@0-8373-0 comm="systemd" exe="/u>
lut 23 22:24:09 greyhill kernel: audit: type=1130 audit(1550957049.044:100): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@0-8>
lut 23 22:24:09 greyhill systemd[1]: radeon-profile-daemon.service: Main process exited, code=dumped, status=6/ABRT
-- Subject: Proces jednostki zakończył działanie
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Proces ExecStart= należący do jednostki radeon-profile-daemon.service zakończył działanie.
-- 
-- Kod wyjścia procesu: „dumped”, jego stan wyjścia: 6.
lut 23 22:24:09 greyhill systemd[1]: radeon-profile-daemon.service: Failed with result 'core-dump'.
-- Subject: Jednostka się nie powiodła
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Jednostka radeon-profile-daemon.service przeszła do stanu „failed” (niepowodzenia)
-- z wynikiem „core-dump”.
lut 23 22:24:09 greyhill audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=radeon-profile-daemon comm="systemd" exe="/usr/li>
lut 23 22:24:09 greyhill kernel: audit: type=1131 audit(1550957049.411:101): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=radeon-profile-daemo>
lut 23 22:24:09 greyhill systemd-coredump[8374]: Removed old coredump core.spotify.1000.50a4ba12f284417fb9194fab706556e4.24240.1550942276000000.lz4.
lut 23 22:24:09 greyhill systemd-coredump[8374]: Process 8323 (radeon-profile-) of user 0 dumped core.
                                                 
                                                 Stack trace of thread 8323:
                                                 #0  0x00007fda525b2d7f raise (libc.so.6)
[... cut ...]
lut 23 22:24:12 greyhill radeon-profile[8344]: Unable to open  "/sys/class/drm/card0/device/hwmon/hwmon0/pwm1"  to write  "25"
lut 23 22:24:13 greyhill plasmashell[1284]: qml: temp unit: 0
lut 23 22:24:16 greyhill plasmashell[1284]: qml: temp unit: 0
lut 23 22:24:16 greyhill plasmashell[1284]: qml: temp unit: 0
lut 23 22:24:20 greyhill radeon-profile[8344]: Unable to open  "/sys/class/drm/card0/device/hwmon/hwmon0/pwm1"  to write  "28"
lut 23 22:24:22 greyhill plasmashell[1284]: qml: temp unit: 0
lut 23 22:24:25 greyhill plasmashell[1284]: qml: temp unit: 0```

marazmista, 2019-02-24:
You can always check `/sys/class/drm/card0/device/hwmon/hwmon0/pwm1_enable`. If it is `2` then fan is controlled by system/card itself.

Could you run daemon in separate root terminal? Logs would not be obfuscated by systemd stuff.

fatino, 2019-02-24:
IDK how to do that. But you can see all the logs here. There's some from KDE but overall it's all there.

marazmista, 2019-02-24:
Open new terminal window and build binary as described in Build section https://github.com/marazmista/radeon-profile-daemon#build but replace line `qmake &&` with `qmake CONFIG+=debug`. After build finishes, go to `target` directory, then type `su` to switch user to root and start daemon `./radeon-profile-daemon` and launch radeon-profile gui normally. Daemon messages should appear in terminal window. Use it for a while and paste logs.

fatino, 2019-02-25:
```lut 25 19:46:27 greyhill radeon-profile-daemon[6966]: "43"  will be written into  "/sys/class/drm/card0/device/hwmon/hwmon0/pwm1"
lut 25 19:46:27 greyhill radeon-profile-daemon[6966]: Performing task:  "7#1#"
lut 25 19:46:27 greyhill radeon-profile-daemon[6966]: ALIVE signal received
lut 25 19:46:45 greyhill radeon-profile-daemon[6966]: Asking connection confirmation...
lut 25 19:47:05 greyhill radeon-profile-daemon[6966]: Confirmation not received, closing connection
lut 25 19:47:05 greyhill radeon-profile-daemon[6966]: Restoring system defaults
lut 25 19:47:05 greyhill radeon-profile-daemon[6966]: Setting fan control mode to auto
lut 25 19:47:05 greyhill radeon-profile-daemon[6966]: "2"  will be written into  "/sys/class/drm/card0/device/hwmon/hwmon0/pwm1_enable"
lut 25 19:47:05 greyhill radeon-profile-daemon[6966]: Client disconnected
lut 25 19:47:05 greyhill radeon-profile-daemon[6966]: Awaiting connections...
```
Now failsafe managed to work.

aaulia, 2019-02-27:
Just noticed it today, but the daemon have been disconnecting for couple of times now (I restart daemon and the radeon-profile each time). This turn my fan up to 40-50% which causes a noticeable noise.  

```
● radeon-profile-daemon.service - radeon-profile daemon
   Loaded: loaded (/usr/lib/systemd/system/radeon-profile-daemon.service; enabled; vendor preset: disabled)
   Active: active (running) since Wed 2019-02-27 08:41:18 WIB; 26min ago
 Main PID: 20877 (radeon-profile-)
    Tasks: 2 (limit: 4915)
   Memory: 1.4M
   CGroup: /system.slice/radeon-profile-daemon.service
           └─20877 /usr/bin/radeon-profile-daemon

Feb 27 08:41:18 archbox systemd[1]: Started radeon-profile daemon.
Feb 27 08:41:27 archbox radeon-profile-daemon[20877]: Connecting to the client
Feb 27 08:41:27 archbox radeon-profile-daemon[20877]: The file path indicated to be set is empty. Lost value:  ""
Feb 27 08:50:07 archbox radeon-profile-daemon[20877]: Restoring system defaults
Feb 27 08:50:07 archbox radeon-profile-daemon[20877]: Setting fan control mode to auto
Feb 27 08:50:07 archbox radeon-profile-daemon[20877]: Disconnecting from the client
```

aaulia, 2019-03-01:
Hello, is there any more info about this, the past couple of days, radeon-profile have been constantly disconnecting for me. I noticed it since my fans kept going wild. This happen repeatedly even after I restart both the service and the GUI.

marazmista, 2019-03-02:
Could you pase full log of daemon? Maybe it will show something.
Also, if disconnect happen, you don't have to restart daemon.

aaulia, 2019-03-04:
After checking with `journalctl`, it's pretty much repetition of what I'm already posting, is there any flag/way to make the daemon spit out more info?

marazmista, 2019-03-04:
You can stop system service and compile daemon in debug mode and run in root console as described in https://github.com/marazmista/radeon-profile/issues/123#issuecomment-466813979

BlazeKl, 2019-03-14:
this is the log that I get from the daemon, restarting the client restores the connection tho

Received signal:  "7#1#"
ALIVE signal received
Received signal:  "2#114#/sys/class/drm/card0/device/hwmon/hwmon1/pwm1#"
Elaborating a SET_VALUE signal
"114"  will be written into  "/sys/class/drm/card0/device/hwmon/hwmon1/pwm1"
Received signal:  "2#112#/sys/class/drm/card0/device/hwmon/hwmon1/pwm1#"
Elaborating a SET_VALUE signal
"112"  will be written into  "/sys/class/drm/card0/device/hwmon/hwmon1/pwm1"
Asking connection confirmation...
Confirmation not received, closing connection
Restoring system defaults
Setting fan control mode to auto
"2"  will be written into  "/sys/class/drm/card0/device/hwmon/hwmon1/pwm1_enable"
Client disconnected
Awaiting connections...

kfilippop, 2019-03-17:
I can also confirm that after some time fan control and overclock tabs become unclickable and the custom fan profile I had active is not working. Just restarting the client restores normality. If you want me to take diagnostic action let me know. v.20190311

kfilippop, 2019-03-18:
I also confirm that /sys/class/drm/card0/device/hwmon/hwmon0/pwm1_enable had a value of 2 when the issue reappeared so fear not!

aaulia, 2019-03-28:
Just want to add (while checking this issue out again), a bit anecdotal but might shed some light into this issue. I manage to stay connected without intermittent disconnection, by keeping my Radeon-Profile opened the whole time (not minimized in the system tray). I'm using i3wm and I just let it open in a workspace (i use 9th). No disconnection so far. While previously I can have two in the span of half an hour.

kfilippop, 2019-03-28:
I "solved" the problem by running automatically R-P as root. This of course is well known already but if it saves so much trouble checking if it works or not, let it be like that.

BlazeKl, 2019-03-28:
for time to time the client stops sending alive signals to the daemon or the daemon is not receiving those signals, this happen to me even with the client opened on my second screen, so I forked the daemon with a workaround which makes the daemon not close the connection with the client even if the signals are not received. this is in no way a proper fix for the issue though since I just commented two lines of code

marazmista, 2019-03-31:
It seems that disconnecting is very inconsistent, sometimes it works for hours without issues and sometimes disconnect happen after few minutes. 
I will add reconnect attempts when connection lost and also option to disable connection checking.

lss4, 2019-04-13:
I encountered this issue myself as I'm setting a fan curve on my Radeon VII so its temperature won't go too high when mining. Currently using the new fan curve I managed to keep the temperature below 80 celsius when mining.

Originally I ran the client as root without any issues. However, given it can be inconvenient to run it as root I decided to try the daemon, and this time I encountered this disconnection issue that I almost fried the card as it continued mining operation at around 105 celsius due to the default fan control is way too inadequate.

```
Apr 10 01:05:40 linuxsys radeon-profile-daemon[39839]: Awaiting connections...
Apr 13 22:33:15 linuxsys radeon-profile-daemon[39839]: Connecting to the client
Apr 13 23:09:15 linuxsys radeon-profile-daemon[39839]: Restoring system defaults
Apr 13 23:09:15 linuxsys radeon-profile-daemon[39839]: Setting fan control mode to auto
Apr 13 23:09:15 linuxsys radeon-profile-daemon[39839]: Client disconnected
Apr 13 23:09:15 linuxsys radeon-profile-daemon[39839]: Awaiting connections...
```

The disconnection seemed to have happened when the system went standby.

I'm not sure if the daemon could be made to save previous settings in case the client got disconnected, as seriously this is **unsafe** in my use case as I risk frying the expensive video card if the fan curve meant to keep the temperature in check failed when I'm mining due to the client disconnected from the daemon (under any circumstances).

Podesta, 2019-04-21:
Also would like to know if this is a issue that affects anyone else. If completely unrelated, I could open a different issue.

Whenever my card loses connection to the daemon, for the reasons of this thread, or even if I simply close the program, the fan on my card starts quickly cycling from something like 20% to 70%, like it's confused and doesn't really know what to do, while making a buzzing sound. It only happens if I open the profile (and connect to the daemon..) and at some point lose the connection. If I don't open the profile, the default profile on the card works fine, and I never hear the buzzing (and fan speed going up and down) that always comes when the profile stops talking to the daemon.

lss4, 2019-04-21:
What was your settings? Did you set a higher fan speed than default? And when that happens, how was your GPU temperature? If it was very high (like about 100) then it's possible the default fan curve has kicked in.

From what I looked, if radeon-profile loses connection to the daemon it'll revert to default hardware settings. The daemon doesn't store any settings on its own at present.

Podesta, 2019-04-22:
It happens irrespective of the card temperature, and also irrespective of whether or not I use a different fan profile or 'performance profile' from auto. It is extremely consistent, as in whenever it loses connection or I close the window, the fan speed goes to ~1862RPM and the percentage keeps cycling between ~26% and ~48%, as if there was a conflict and it didn't know what to do, always accompanied by this buzzing sound. The actual RPM doesn't change much, after going to the 1862, no matter if it was idling at 35C and 900RPM before or not.

When I used to run as a normal user, and had the constant disconnections, I would usually fix it by going to "profile_peak" which would max the fan speed, and back to "auto". Since I started running as sudo, I only encounter it when if I close the window before a shutdown or restart. Even if, while running as sudo, I open a new window as normal user and close it, the same thing happens, but only for a few seconds, when I believe the rooted application gets back control automatically.

Maybe it is my card, a Sapphire Nitro+ RX580. I believe it is a quiet GPU under Windows, as in no fans running until it reaches about 60C, but by default always runs with at least 20% fan speed with the open source drivers, at least. I'll try the flipping the switch to the other BIOS, since it is supposed to have dual mode for miners, and see if the same thing still happens.

lss4, 2019-04-22:
It seems there's some strange interactions between your card and radeon-profile the moment when your profile is disabled (when closing radeon-profile, sudo or not, or when a normal user radeon-profile loses connection to the daemon). Also, from your comment it seems radeon-profile-daemon has higher priority than a sudo-ed radeon-profile.

You can put your case in another issue (and reference this one), as your case is probably something else that's specific to your card or your profile settings.

marazmista, 2019-04-27:
After commit 8e1a4446499b62bb8511c679ca2a3bcc1df44219 it should reconnect in case of connection loss and restore fan state.

lss4, 2019-04-28:
Is it still possible for the client to disconnect from daemon under certain circumstances after the commit? And when it disconnects, would it still revert to default settings? If so, the problem is not solved, as there would still be "downtime" where the client might get disconnected and causing the fan curve settings to be reverted to default during the period.

Under scenarios where the default, conservative fan curve could result in overheat, even a single second of "downtime" would not be acceptable, while there would be zero downtime when running radeon-profile as root.

ssnakee99, 2019-05-10:
I installed the ROCm package and my radeon-profile stopped working in the same way you guys are discribing, all the GUI is greyed out, it may be related to the fact that the ROCm package Kernel Driver, but this goes beyond my head, so i hope someone can help me to get it to work again thanks

lss4, 2019-05-11:
Just run radeon-profile as root. If it's still greyed out then it's possibly driver-related (I don't know about ROCm kernel driver as I'm only using its OpenCL part, with original amdgpu, on 5.0 kernel).

The daemon in its current business logic is not recommended. Occasionally the client would disconnect from the daemon causing the GUI to grey out and temporarily revert all settings you configured until it reconnects.

marazmista, 2019-05-11:
> revert all settings 

Not all settings, just fan control.

Reconnect is attempted at next refresh (timer) event (configured in settings), so the 'downtime' you are describing is actually the same as refreshing temperature and adjusting fan speed if needed.

> I installed the ROCm package and my radeon-profile stopped working in the same way you guys are discribing, all the GUI is greyed out, it may be related to the fact that the ROCm package Kernel Driver, but this goes beyond my head, so i hope someone can help me to get it to work again thanks

How do you use ROCm? Is this something that is running in background or just display info in console and quits?

lss4, 2019-05-12:
> Not all settings, just fan control.

Is it possible to avoid doing so when disconnected with the daemon? Because I'm trying to keep the card's temperature in check while mining using my own fan settings, this would defeat the purpose.

Right now I have to keep it running in the background as root so the fan control settings would have zero downtime.

HanFox, 2019-05-23:
I'm having this issue, too, with an AMD 290X on Arch Linux.

I noticed that when the client loses connection the daemon is setting `pwm1_enable` to 2 (or auto). This seems like the intelligent choice, however, it's noted on the Arch Linux wiki that setting `pwm1_enable` to 2 can trigger a bug, which gets the card stuck in auto mode which can only be fixed with a reboot. From my experience this is true.

See: https://wiki.archlinux.org/index.php/Fan_speed_control#AMDGPU_sysfs_fan_control

Thus Radeon Profile can never take back over, because it's locked out (everything's locked out, even using su/root to echo in to the file).

When I boot up with no programs that have touched fan controls that file seems to be set to 1, but causes the default "odd" behaviour of the driver (I don't know if it's the card itself that has an emergency mode?) just trying to keep the card alive when it hits its "meltdown" temperate (mine being >95C when the card will ramp the fans).

Maybe some testing w/o setting `pwm1_enable` to 2, or maybe just do what fancontrol does and just set the fan to 100% on connection loss?

I'm unaware of any upstream bug report for the Arch Linux note. Maybe someone can find it, or find a mention of someone involved in the driver/graphics stack talking about it?