marazmista/radeon-profile #125: 0RPM fan speed

maln0ir, 2019-03-04:
I have exactly the same issue like here #89, but older card. I don't have any plans to replace that card :)

[My card](http://xfxforce.com/en-us/products/amd-radeon-r9-200-series/amd-radeon-r9-280x-black-double-dissipation-edition-r9-280x-tdbd) has two fans, but very often only one is spinning. I don't think this is an issue, because under heavy load both are spinning and temperature is almost always below 80 degrees. I manually set fan to low speed to make it quiet, but fan never stops, so fan speed should not be 0 RPM.

[image]

More details

```
~ $ uname -a
Linux Nidhoggr 4.20.13-zen1-1-zen #1 ZEN SMP PREEMPT Wed Feb 27 19:12:55 UTC 2019 x86_64 GNU/Linux

~ $ LC_ALL=C pacman -Qi radeon-profile-git radeon-profile-daemon-git | grep -E Version
Version         : 20190220.r9.ga539519-1
Version         : 20190301.r0.gae026bd-1

~ $ lspci -k | grep -A 2 VGA
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Tahiti XT [Radeon HD 7970/8970 OEM / R9 280X]
	Subsystem: XFX Pine Group Inc. Tahiti XTL [Radeon R9 280X]
	Kernel driver in use: amdgpu

~ $ lsmod | grep amdgpu
amdgpu               4014080  11
chash                  16384  1 amdgpu
i2c_algo_bit           16384  1 amdgpu
amd_iommu_v2           20480  1 amdgpu
gpu_sched              36864  1 amdgpu
drm_kms_helper        229376  1 amdgpu
ttm                   122880  1 amdgpu
drm                   536576  7 gpu_sched,drm_kms_helper,amdgpu,ttm

==> /sys/class/drm/card0/device/hwmon/hwmon1/device <==
head: error reading '/sys/class/drm/card0/device/hwmon/hwmon1/device': Is a directory

==> /sys/class/drm/card0/device/hwmon/hwmon1/fan1_enable <==
1

==> /sys/class/drm/card0/device/hwmon/hwmon1/fan1_input <==
0

==> /sys/class/drm/card0/device/hwmon/hwmon1/fan1_target <==
0

==> /sys/class/drm/card0/device/hwmon/hwmon1/in0_input <==
head: error reading '/sys/class/drm/card0/device/hwmon/hwmon1/in0_input': Invalid argument

==> /sys/class/drm/card0/device/hwmon/hwmon1/in0_label <==
vddgfx

==> /sys/class/drm/card0/device/hwmon/hwmon1/name <==
amdgpu

==> /sys/class/drm/card0/device/hwmon/hwmon1/power <==
head: error reading '/sys/class/drm/card0/device/hwmon/hwmon1/power': Is a directory

==> /sys/class/drm/card0/device/hwmon/hwmon1/power1_average <==
head: error reading '/sys/class/drm/card0/device/hwmon/hwmon1/power1_average': Invalid argument

==> /sys/class/drm/card0/device/hwmon/hwmon1/power1_cap <==

==> /sys/class/drm/card0/device/hwmon/hwmon1/power1_cap_max <==

==> /sys/class/drm/card0/device/hwmon/hwmon1/power1_cap_min <==
0

==> /sys/class/drm/card0/device/hwmon/hwmon1/pwm1 <==
108

==> /sys/class/drm/card0/device/hwmon/hwmon1/pwm1_enable <==
1
[... 100 lines cut ...]
parm:           virtual_display:Enable virtual display feature (the virtual_display will be set like xxxx:xx:xx.x,x;xxxx:xx:xx.x,x) (charp)
parm:           ngg:Next Generation Graphics (1 = enable, 0 = disable(default depending on gfx)) (int)
parm:           prim_buf_per_se:the size of Primitive Buffer per Shader Engine (default depending on gfx) (int)
parm:           pos_buf_per_se:the size of Position Buffer per Shader Engine (default depending on gfx) (int)
parm:           cntl_sb_buf_per_se:the size of Control Sideband per Shader Engine (default depending on gfx) (int)
parm:           param_buf_per_se:the size of Off-Chip Pramater Cache per Shader Engine (default depending on gfx) (int)
parm:           job_hang_limit:how much time allow a job hang and not drop it (default 0) (int)
parm:           lbpw:Load Balancing Per Watt (LBPW) support (1 = enable, 0 = disable, -1 = auto) (int)
parm:           compute_multipipe:Force compute queues to be spread across pipes (1 = enable, 0 = disable, -1 = auto) (int)
parm:           gpu_recovery:Enable GPU recovery mechanism, (1 = enable, 0 = disable, -1 = auto) (int)
parm:           emu_mode:Emulation mode, (1 = enable, 0 = disable) (int)
parm:           si_support:SI support (1 = enabled, 0 = disabled (default)) (int)
parm:           cik_support:CIK support (1 = enabled, 0 = disabled (default)) (int)
parm:           smu_memory_pool_size:reserve gtt for smu debug usage, 0 = disable,0x1 = 256Mbyte, 0x2 = 512Mbyte, 0x4 = 1 Gbyte, 0x8 = 2GByte (uint)
parm:           sched_policy:Scheduling policy (0 = HWS (Default), 1 = HWS without over-subscription, 2 = Non-HWS (Used for debugging only) (int)
parm:           hws_max_conc_proc:Max # processes HWS can execute concurrently when sched_policy=0 (0 = no concurrency, #VMIDs for KFD = Maximum(default)) (int)
parm:           cwsr_enable:CWSR enable (0 = Off, 1 = On (Default)) (int)
parm:           max_num_of_queues_per_device:Maximum number of supported queues per device (1 = Minimum, 4096 = default) (int)
parm:           send_sigterm:Send sigterm to HSA process on unhandled exception (0 = disable, 1 = enable) (int)
parm:           debug_largebar:Debug large-bar flag used to simulate large-bar capability on non-large bar machine (0 = disable, 1 = enable) (int)
parm:           ignore_crat:Ignore CRAT table during KFD initialization (0 = use CRAT (default), 1 = ignore CRAT) (int)
parm:           noretry:Set sh_mem_config.retry_disable on Vega10 (0 = retry enabled (default), 1 = retry disabled) (int)
parm:           halt_if_hws_hang:Halt if HWS hang is detected (0 = off (default), 1 = on) (int)
parm:           dcfeaturemask:all stable DC features enabled (default)) (uint)
```

parkerlreed, 2019-03-04:
Setting a 10% speed should be "off" for most cards (at least with the fan curve screen)

marazmista, 2019-03-09:
RPM fan speed is read from sysfs file and is displayed in GUI without any conversions, so if file reports that speed is 0 but fan is actually spinning seems like something may be wrong in kernel or bios parts that put value into that file.